sustainable-computing-io/kepler #906: Find Azure Virtual Machine(s) to install Kepler on AKS

AntonioDiTuri, 2023-09-01:
### What would you like to be added?

**Is your feature request related to a problem? Please describe.**

This is the Azure twin from the [issue 707](https://github.com/sustainable-computing-io/kepler/issues/707)

Currently there is no documented way to deploy Kepler on Azure AKS.

This can be challenging given that there are specific [requirements](https://sustainable-computing.io/installation/kepler/#requirements) for Kepler to work with eBPF and access RAPL.

Describe the solution you'd like

Find and list Azure virtual machines that:

satisfy Kepler's requirements for eBPF and access to RAPL
can be used in an AKS cluster
Then, it would be great to open a PR to the website to document these findings.

Additional context

Most likely the instances will have to be bare metal instances to have direct access to hardware. 
The tradeoff is that these are usually more expensive, but we will work with this to unblock AKS support.

### Why is this needed?

This would help the reachability of the projects to the Azure Cloud Users

AntonioDiTuri, 2023-09-01:
@sanjayprak would like to work on this

sanjayprak, 2023-09-01:
glad to work on this!

closed 2023-11-07 by stale[bot]