sustainable-computing-io/kepler #907: Metrics error when kepler is installed on eks with single node c5.metal

AntonioDiTuri, 2023-09-01:
### What happened?

@rossf7 were investigating the possibility to use kepler on AWS EKS.

Part of the story can be followed on this issue: https://github.com/sustainable-computing-io/kepler/issues/707

In this comment @rootfs asked to create a new ticket with the following logs:

```
k exec -ti -n kepler daemonset/kepler-exporter -- bash -c "curl localhost:9102/metrics" | grep 'kepler_'
```

result: https://pastebin.com/yYHeFs5H 

### What did you expect to happen?

kepler_container_package_joules_total different from 0

### How can we reproduce it (as minimally and precisely as possible)?

Create an eks cluster with a single node

```
eksctl create cluster --name rapl-test --region us-east-2 --nodes 1 --instance-types c5.metal --spot --ssh-access --node-ami-family Ubuntu2004
```

Load the kernel module for rapl
```
sudo apt-get install linux-modules-extra-$(uname -r)
sudo modprobe intel_rapl_common
```

Used the makefile with the "BM_DEPLOY" option on manifests 0.5.4 as requested [here](https://github.com/sustainable-computing-io/kepler/issues/707#issuecomment-1684438854)

```
git checkout release-0.5.4
make build-manifest OPTS="BM_DEPLOY"
```

This deployment.yaml was generated, I just changed the tag to match the one requested [here] (latest-libbpf)(https://github.com/sustainable-computing-io/kepler/issues/707#issuecomment-1684438854)

```
apiVersion: v1
kind: Namespace
metadata:
  labels:
    pod-security.kubernetes.io/audit: privileged
    pod-security.kubernetes.io/enforce: privileged
    pod-security.kubernetes.io/warn: privileged
    security.openshift.io/scc.podSecurityLabelSync: "false"
    sustainable-computing.io/app: kepler
  name: kepler
---
apiVersion: v1
kind: ServiceAccount
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: kepler-sa
  namespace: kepler
---
apiVersion: rbac.authorization.k8s.io/v1
kind: ClusterRole
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: kepler-clusterrole
rules:
- apiGroups:
  - ""
[... 282 lines cut ...]

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2023-09-01:
```
kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="491136917149570c8e880c5eaa0583f454759e0c38010fa6c5361a96b93d676c",container_name="coredns",container_namespace="kube-system",pod_name="coredns-cbbbbb9cb-22dd4"} 0
kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="4ed8d09abda59733fe460dfb9c53de5c815d653cd9b89c2a75a73dcdaf257835",container_name="kube-proxy",container_namespace="kube-system",pod_name="kube-proxy-trjh8"} 0
kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="4fa3525f7c80f97f068d4980795cb28e0b8a52b09e1393f5bda19d191a1b5fba",container_name="coredns",container_namespace="kube-system",pod_name="coredns-cbbbbb9cb-dvnmt"} 0
kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="6ff21a22f38baf413aa49b1dbb69c423cd9f194db29f8bf397d0f2af133684da",container_name="aws-node",container_namespace="kube-system",pod_name="aws-node-lxhsn"} 0
kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="a27afe2c351fb137fe9aa1069402d14f5f6ff7c3e25dcb416646247f881c9615",container_name="kepler-exporter",container_namespace="kepler",pod_name="kepler-exporter-r2xgx"} 0
kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="d98b90b7199e5affa57dcbd1d106cad57705b1d23e8c2c66b2882425c03d0949",container_name="aws-vpc-cni-init",container_namespace="kube-system",pod_name="aws-node-lxhsn"} 0
kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="system_processes",container_name="system_processes",container_namespace="system",pod_name="system_processes"} 0
```

it looks kepler cannot get the cgroup info from the node, probably EKS uses a different container runtime than kepler expects.

@AntonioDiTuri can you help get the container's cgroup path info? In the following example, I `exec` to kepler pod, get a `cgroup` path from a container (in this case the `kepler` pod), and get the full path of the cgroup. 

```console
# kubectl exec -ti -n kepler daemonset/kepler-exporter "bash"
kubectl exec [POD] [COMMAND] is DEPRECATED and will be removed in a future version. Use kubectl exec [POD] -- [COMMAND] instead.
basename: missing operand
Try 'basename --help' for more information.
basename: missing operand
Try 'basename --help' for more information.
[root@kepler-exporter-p6bt8 /]# 
[root@kepler-exporter-p6bt8 /]# ps -ef |grep "/usr/bin/kepler" |head -1 |awk '{print $2}' |xargs -I {} cat  /proc/{}/cgroup
0::/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-pode6fb4163_f63d_44de_a75e_e57713818738.slice/crio-1cf2f25eae689748d798634dba6cc43f247119a861876fdcd605699cf37bbc2f.scope
[root@kepler-exporter-p6bt8 /]# find /sys/ -iname "crio-1cf2f25eae689748d798634dba6cc43f247119a861876fdcd605699cf37bbc2f.scope"
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-pode6fb4163_f63d_44de_a75e_e57713818738.slice/crio-1cf2f25eae689748d798634dba6cc43f247119a861876fdcd605699cf37bbc2f.scope
```

AntonioDiTuri, 2023-09-08:
Sure! Here you go: 

```
[root@kepler-exporter-sxqkk /]# ps -ef |grep "/usr/bin/kepler" |head -1 |awk '{print $2}' |xargs -I {} cat  /proc/{}/cgroup
13:cpuset:/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
12:pids:/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
11:hugetlb:/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
10:misc:/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
9:blkio:/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
8:devices:/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
7:net_cls,net_prio:/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
6:perf_event:/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
5:rdma:/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
4:freezer:/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
3:cpu,cpuacct:/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
2:memory:/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
1:name=systemd:/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
```

I then tried: 
```
# find /sys/ -iname "pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa"

find: warning: ‘-iname’ matches against basenames only, but the given pattern contains a directory separator (‘/’), thus the expression will evaluate to false all the time.  Did you mean ‘-iwholename’?
```
I then tried the same command with the suggested option '-iwholename' but nothing was returned in output.
I am not sure why I don't have a similar .scope file. 
I also don't know if getting many lines is expected. 
I am not familiar with the concept of cgroup, let me read a bit more :)

rootfs, 2023-09-08:
can you try just the filename?
```
find /sys/ -iname "d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa"
```

AntonioDiTuri, 2023-09-08:
```
[root@kepler-exporter-sxqkk 24720]# find /sys/ -iname "d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa"

/sys/fs/cgroup/cpuset/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
/sys/fs/cgroup/pids/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
/sys/fs/cgroup/hugetlb/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
/sys/fs/cgroup/misc/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
/sys/fs/cgroup/blkio/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
/sys/fs/cgroup/devices/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
/sys/fs/cgroup/net_cls,net_prio/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
/sys/fs/cgroup/perf_event/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
/sys/fs/cgroup/rdma/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
/sys/fs/cgroup/freezer/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
/sys/fs/cgroup/cpu,cpuacct/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
/sys/fs/cgroup/memory/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
/sys/fs/cgroup/systemd/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
/sys/fs/cgroup/unified/kubepods/burstable/pod69ec2724-f389-498a-9a04-ed9b2065005d/d2bd18310b0f469c8d77b914762d0744aa60963364497fe53887b1f8ef5a1dfa
```

rootfs, 2023-09-08:
@AntonioDiTuri looks this is a cgroup v1 setup. What is the container runtime you are using, containerd or something else?

rootfs, 2023-09-08:
@AntonioDiTuri i created an eks cluster using your command and applied the latest manfiest (from kepler main branch). The numbers are non-zero 
```console
work % kubectl apply -f https://gist.githubusercontent.com/rootfs/24f30eec07d955df9da1b10f8d403d8d/raw/98ab2d2ab93e85a1ed0d3f921f40a2ef013babcd/kepler-eks-bm.yaml
namespace/kepler created
serviceaccount/kepler-sa created
clusterrole.rbac.authorization.k8s.io/kepler-clusterrole created
clusterrolebinding.rbac.authorization.k8s.io/kepler-clusterrole-binding created
configmap/kepler-cfm created
secret/redfish-4kh9d7bc7m created
service/kepler-exporter created
daemonset.apps/kepler-exporter created
work % kubectl get pods -A -w
NAMESPACE     NAME                      READY   STATUS              RESTARTS   AGE
kepler        kepler-exporter-csqcl     0/1     ContainerCreating   0          14s
kube-system   aws-node-8h6wt            1/1     Running             0          69m
kube-system   coredns-8fd4db68f-9gbht   1/1     Running             0          79m
kube-system   coredns-8fd4db68f-kj78t   1/1     Running             0          79m
kube-system   kube-proxy-xncwm          1/1     Running             0          69m
kepler        kepler-exporter-csqcl     1/1     Running             0          31s
^C%
work % kubectl exec -ti -n kepler daemonset/kepler-exporter bash
kubectl exec [POD] [COMMAND] is DEPRECATED and will be removed in a future version. Use kubectl exec [POD] -- [COMMAND] instead.
basename: missing operand
Try 'basename --help' for more information.
basename: missing operand
Try 'basename --help' for more information.
[root@kepler-exporter-csqcl /]# curl localhost:9102/metrics |kepler_container_joules
bash: kepler_container_joules: command not found
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (23) Failure writing output to destination
[root@kepler-exporter-csqcl /]# curl localhost:9102/metrics |grep  kepler_container_joules
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0# HELP kepler_container_joules_total Aggregated RAPL Package + Uncore + DRAM + GPU + other host components (platform - package - dram) in joules
# TYPE kepler_container_joules_total counter
kepler_container_joules_total{command="",container_id="0527479c710b3e527a3fb6a262f7fb108837426e56e566927c3ae0428061ca3e",container_name="kube-proxy",container_namespace="kube-system",mode="dynamic",pod_name="kube-proxy-xncwm"} 803.55
kepler_container_joules_total{command="",container_id="99129acaa15662c74435d781a25ae9b4407b3ffa3416b0fbfe28029378420fca",container_name="aws-node",container_namespace="kube-system",mode="dynamic",pod_name="aws-node-8h6wt"} 803.55
kepler_container_joules_total{command="",container_id="9b7ef25b7ec0a6e401f1e969f32f2d388f8f22270e0490cc3003944b6cc883fb",container_name="coredns",container_namespace="kube-system",mode="dynamic",pod_name="coredns-8fd4db68f-kj78t"} 803.55
kepler_container_joules_total{command="",container_id="ab5007969f777946a649f9ba5b8ca5d92d39b0e8fc4f3d6bd7134eba3f1bba34",container_name="kepler-exporter",container_namespace="kepler",mode="dynamic",pod_name="kepler-exporter-csqcl"} 803.55
kepler_container_joules_total{command="",container_id="bb377e2f35d642f78034df50fe78a5e01b856ac2b21e40fd8b6305a01ee0b496",container_name="aws-vpc-cni-init",container_namespace="kube-system",mode="dynamic",pod_name="aws-node-8h6wt"} 803.55
kepler_container_joules_total{command="",container_id="d5d1d9746f24d1b561d7c8a7ba66d6527a6b0702624376e0d725b283d5d10b71",container_name="coredns",container_namespace="kube-system",mode="dynamic",pod_name="coredns-8fd4db68f-9gbht"} 803.55
kepler_container_joules_total{command="",container_id="system_processes",container_name="system_processes",container_namespace="system",mode="dynamic",pod_name="system_processes"} 803.55
100 46039    0 46039    0     0  10.9M      0 --:--:-- --:--:-- --:--:-- 10.9M
```

rootfs, 2023-09-08:
the cgroupfs stat is 0
```console
[root@kepler-exporter-csqcl /]# curl localhost:9102/metrics |grep  kepler_container_cgroupfs_user_cpu_usage_us_total
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0# HELP kepler_container_cgroupfs_user_cpu_usage_us_total Aggregated user cpu usage obtained from cGroups
# TYPE kepler_container_cgroupfs_user_cpu_usage_us_total counter
100 46082    0 46082    0     kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="0527479c710b3e527a3fb6a262f7fb108837426e56e566927c3ae0428061ca3e",container_name="kube-proxy",container_namespace="kube-system",pod_name="kube-proxy-xncwm"} 0
0  kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="99129acaa15662c74435d781a25ae9b4407b3ffa3416b0fbfe28029378420fca",container_name="aws-node",container_namespace="kube-system",pod_name="aws-node-8h6wt"} 0
7kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="9b7ef25b7ec0a6e401f1e969f32f2d388f8f22270e0490cc3003944b6cc883fb",container_name="coredns",container_namespace="kube-system",pod_name="coredns-8fd4db68f-kj78t"} 0
50kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="ab5007969f777946a649f9ba5b8ca5d92d39b0e8fc4f3d6bd7134eba3f1bba34",container_name="kepler-exporter",container_namespace="kepler",pod_name="kepler-exporter-csqcl"} 0
0kkepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="bb377e2f35d642f78034df50fe78a5e01b856ac2b21e40fd8b6305a01ee0b496",container_name="aws-vpc-cni-init",container_namespace="kube-system",pod_name="aws-node-8h6wt"} 0
  kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="d5d1d9746f24d1b561d7c8a7ba66d6527a6b0702624376e0d725b283d5d10b71",container_name="coredns",container_namespace="kube-system",pod_name="coredns-8fd4db68f-9gbht"} 0
  kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="system_processes",container_name="system_processes",container_namespace="system",pod_name="system_processes"} 0
```

the kepler indicates node power is available
```console
work % kubectl logs -n kepler daemonset/kepler-exporter
I0908 16:05:41.483590       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0908 16:05:41.519253       1 qat.go:35] Failed to init qat-telemtry err: could not get qat status exit status 127
I0908 16:05:41.533749       1 exporter.go:158] Kepler running on version: 2ec465f
I0908 16:05:41.533768       1 config.go:270] using gCgroup ID in the BPF program: true
I0908 16:05:41.533804       1 config.go:272] kernel version: 5.15
I0908 16:05:41.533837       1 exporter.go:170] LibbpfBuilt: false, BccBuilt: true
I0908 16:05:41.533887       1 config.go:203] kernel source dir is set to /usr/share/kepler/kernel_sources
I0908 16:05:41.533957       1 exporter.go:189] EnabledBPFBatchDelete: true
I0908 16:05:41.533988       1 rapl_msr_util.go:129] failed to open path /dev/cpu/0/msr: no such file or directory
I0908 16:05:41.534067       1 power.go:71] Unable to obtain power, use estimate method
I0908 16:05:41.534129       1 redfish.go:173] failed to initialize node credential: no supported node credential implementation
I0908 16:05:41.541691       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0908 16:05:41.567640       1 exporter.go:204] Initializing the GPU collector
I0908 16:05:47.570895       1 watcher.go:66] Using in cluster k8s config
modprobe: FATAL: Module kheaders not found in directory /lib/modules/5.15.0-1044-aws
chdir(/lib/modules/5.15.0-1044-aws/build): No such file or directory
I0908 16:05:47.704089       1 bcc_attacher.go:80] failed to attach the bpf program: <nil>
I0908 16:05:47.704115       1 bcc_attacher.go:158] failed to attach perf module with options [-DMAP_SIZE=10240 -DNUM_CPUS=96]: failed to attach the bpf program: <nil>, from default kernel source.
I0908 16:05:47.704141       1 bcc_attacher.go:161] trying to load eBPF module with kernel source dir /usr/share/kepler/kernel_sources/4.18.0-477.13.1.el8_8.x86_64
I0908 16:05:48.429588       1 bcc_attacher.go:129] Successfully load eBPF module from using bcc
I0908 16:05:48.429616       1 bcc_attacher.go:167] Successfully loaded eBPF module with options: [-DMAP_SIZE=10240 -DNUM_CPUS=96] from kernel source "/usr/share/kepler/kernel_sources/4.18.0-477.13.1.el8_8.x86_64"
I0908 16:05:48.429641       1 bcc_attacher.go:186] Successfully load eBPF module from bcc with option: [-DMAP_SIZE=10240 -DNUM_CPUS=96]
I0908 16:05:48.493276       1 container_energy.go:109] Using the Ratio/DynPower Power Model to estimate Container Platform Power
I0908 16:05:48.493304       1 container_energy.go:118] Using the Ratio/DynPower Power Model to estimate Container Component Power
I0908 16:05:48.493327       1 process_power.go:108] Using the Ratio/DynPower Power Model to estimate Process Platform Power
I0908 16:05:48.493349       1 process_power.go:117] Using the Ratio/DynPower Power Model to estimate Process Component Power
I0908 16:05:48.493628       1 node_platform_energy.go:53] Using the LinearRegressor/AbsPower Power Model to estimate Node Platform Power
I0908 16:05:48.493749       1 node_component_energy.go:54] Using the LinearRegressor/AbsPower Power Model to estimate Node Component Power
I0908 16:05:48.493856       1 exporter.go:276] Started Kepler in 6.960132643s
```

rootfs, 2023-09-12:
> Is it ok that the numbers are all the same even if the containers are different?

@marceloamaral can you share your thought? thanks

AntonioDiTuri, 2023-09-12:
The version you installed is working for me as well: 
```
curl localhost:9102/metrics |grep  kepler_container_joules
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
# HELP kepler_container_joules_total Aggregated RAPL Package + Uncore + DRAM + GPU + other host components (platform - package - dram) in joules
# TYPE kepler_container_joules_total counter
kepler_container_joules_total{command="",container_id="08d61e8454454ecf25c0897440333a3f0d68b6a6c184ac0326f849b33fa1aafb",container_name="coredns",container_namespace="kube-system",mode="dynamic",pod_name="coredns-cbbbbb9cb-jvjth"} 486.63599999999997
kepler_container_joules_total{command="",container_id="26d774a6f8ed2a38b88487e4f911067fd8a7418dbbd1405977c381ad15d6ceea",container_name="aws-vpc-cni-init",container_namespace="kube-system",mode="dynamic",pod_name="aws-node-6ggpn"} 486.63599999999997
kepler_container_joules_total{command="",container_id="979218e0cf59e157384a6bfddc2aa7c4266667b767411592fff1cc3767fba7fb",container_name="kepler-exporter",container_namespace="kepler",mode="dynamic",pod_name="kepler-exporter-k5bkd"} 486.63599999999997
kepler_container_joules_total{command="",container_id="9be06d7cb3f9b8b5d753330e3280abee51491bd5d1468206204cd19ac328209d",container_name="coredns",container_namespace="kube-system",mode="dynamic",pod_name="coredns-cbbbbb9cb-4g8db"} 486.63599999999997
kepler_container_joules_total{command="",container_id="a780a3dab921d319a56b73a3e0b8ae25c509ecc5c2b1169a7b3e7890ea7271d3",container_name="aws-node",container_namespace="kube-system",mode="dynamic",pod_name="aws-node-6ggpn"} 486.63599999999997
kepler_container_joules_total{command="",container_id="e2af45f4fe185f7c7a904944265de287fc75a2d5be0db43f541478a17cc282f6",container_name="kube-proxy",container_namespace="kube-system",mode="dynamic",pod_name="kube-proxy-rsn5l"} 486.63599999999997
kepler_container_joules_total{command="",container_id="system_processes",container_name="system_processes",container_namespace="system",mode="dynamic",pod_name="system_processes"} 486.63599999999997
100 46512    0 46512    0     0  11.0M      0 --:--:-- --:--:-- --:--:-- 11.0M
```
But I get always only one number

cfgroupfs is also 0

```
curl localhost:9102/metrics |grep  kepler_container_cgroupfs_user_cpu_usage_us_total
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0# 
# HELP kepler_container_cgroupfs_user_cpu_usage_us_total Aggregated user cpu usage obtained from cGroups
# TYPE kepler_container_cgroupfs_user_cpu_usage_us_total counter
kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="08d61e8454454ecf25c0897440333a3f0d68b6a6c184ac0326f849b33fa1aafb",container_name="coredns",container_namespace="kube-system",pod_name="coredns-cbbbbb9cb-jvjth"} 0
1kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="26d774a6f8ed2a38b88487e4f911067fd8a7418dbbd1405977c381ad15d6ceea",container_name="aws-vpc-cni-init",container_namespace="kube-system",pod_name="aws-node-6ggpn"} 0
00 4kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="979218e0cf59e157384a6bfddc2aa7c4266667b767411592fff1cc3767fba7fb",container_name="kepler-exporter",container_namespace="kepler",pod_name="kepler-exporter-k5bkd"} 0
647kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="9be06d7cb3f9b8b5d753330e3280abee51491bd5d1468206204cd19ac328209d",container_name="coredns",container_namespace="kube-system",pod_name="coredns-cbbbbb9cb-4g8db"} 0
6  kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="a780a3dab921d319a56b73a3e0b8ae25c509ecc5c2b1169a7b3e7890ea7271d3",container_name="aws-node",container_namespace="kube-system",pod_name="aws-node-6ggpn"} 0
  0kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="e2af45f4fe185f7c7a904944265de287fc75a2d5be0db43f541478a17cc282f6",container_name="kube-proxy",container_namespace="kube-system",pod_name="kube-proxy-rsn5l"} 0
 kepler_container_cgroupfs_user_cpu_usage_us_total{command="",container_id="system_processes",container_name="system_processes",container_namespace="system",pod_name="system_processes"} 0

```

And those are the kepler logs: 
```
I0912 14:37:12.798396       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No │
│  such file or directory                                                                                                                                                        │
│ I0912 14:37:12.849331       1 qat.go:35] Failed to init qat-telemtry err: could not get qat status exit status 127                                                             │
│ I0912 14:37:12.864778       1 exporter.go:158] Kepler running on version: 2ec465f                                                                                              │
│ I0912 14:37:12.864800       1 config.go:270] using gCgroup ID in the BPF program: true                                                                                         │
[... cut ...]
│ I0912 14:37:19.742092       1 node_platform_energy.go:53] Using the LinearRegressor/AbsPower Power Model to estimate Node Platform Power                                       │
│ I0912 14:37:19.742199       1 node_component_energy.go:54] Using the LinearRegressor/AbsPower Power Model to estimate Node Component Power                                     │
│ I0912 14:37:19.742304       1 exporter.go:276] Started Kepler in 6.877546749s
```
It seems that also my setting is unable to get the power metrics and using the estimation model, I wonder why...

marceloamaral, 2023-09-21:
@sunya-ch which Kepler metric the estimation model is using now?
The ebpf cpu time? @AntonioDiTuri can you also check the values of this metric?

sunya-ch, 2023-09-21:
> @sunya-ch which Kepler metric the estimation model is using now?
> The ebpf cpu time? @AntonioDiTuri can you also check the values of this metric?

Yes, it is bpf cpu time: https://github.com/sustainable-computing-io/kepler/blob/main/data/model_weight/acpi_AbsPowerModel.json.

marceloamaral, 2023-09-21:
Right, it is missing the kernel headers.

I thought the libbpf was already the default approach.

@AntonioDiTuri can you update the Kepler image to:
`quay.io/sustainable_computing_io/kepler:latest-libbpf`

AntonioDiTuri, 2023-09-21:
I don't know if you noticed but [in the first message:](https://github.com/sustainable-computing-io/kepler/issues/907#issue-1877540435) I already used the latest-libbpf without success. 

I don't know if in the latest-libbpf there were relevant changes in the last 3 weeks. 
I was not able to find that out from the github repo.

Anyway when I tried that image I only had 0 for: 
- kepler_container_cgroupfs_user_cpu_usage_us_total
- kepler_container_bpf_cpu_time_us_total

You can check the past logs in this pasebin: https://pastebin.com/yYHeFs5H

@rootfs [mentioned](https://github.com/sustainable-computing-io/kepler/issues/907#issuecomment-1711770319) it might have been a problem with cgroupv1 setup.

I can try that again to double check.

marceloamaral, 2023-09-27:
@AntonioDiTuri the problem might be related to translate the PID to containerID. Kepler uses the `/proc/PID/cgroup` file to find the container ID of a process, if Kepler cannot access this file, the container ID resolution will fail.

How did you create your cluster? Can you run a workload, enter in the node (that Kepler is running), find the workload PID and get the data from the `/proc/PID/cgroup` ?

Also, can you run Kepler with log verbosity 5 and share it?