sustainable-computing-io/kepler #908: Sidecar estimator

Lai-Kenny, 2023-09-03:
### What happened?

My VM Environment

Ubuntu | Linux Kernel
-- | --
22.04.3 LTS | 5.15.0-78-generic x86_64

Kubernetes | k8s_cni_version | Docker | Helm
-- | -- | -- | --
1.21.0 | 0.7.5 | 20.10.12-0ubuntu1~22.04.3 | 3.5.4

I make build-manifest OPTS="ESTIMATOR_SIDECAR_DEPLOY"
And change the model server image to the quay.io/sustainable_computing_io/kepler_model_server:v0.6 and exporter image to quay.io/sustainable_computing_io/kepler:latest-libbpf in the daemonset.

The following is my estimator log:
```
2023-08-29 03:43:00.127466: I tensorflow/core/util/util.cc:169] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2023-08-29 03:43:00.136698: W tensorflow/stream_executor/platform/default/dso_loader.cc:64] Could not load dynamic library 'libcudart.so.11.0'; dlerror: libcudart.so.11.0: cannot open shared object file: No such file or directory
2023-08-29 03:43:00.136736: I tensorflow/stream_executor/cuda/cudart_stub.cc:29] Ignore above cudart dlerror if you do not have a GPU set up on your machine.
set NODE_COMPONENTS_ESTIMATOR to true.
set NODE_COMPONENTS_INIT_URL to https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-db/main/models/Linux-4.15.0-213-generic-x86_64_v0.6/rapl/AbsPower/KubeletOnly/GradientBoostingRegressorTrainer_1.zip.
clean socket
get archived model
get init url https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-db/main/models/Linux-4.15.0-213-generic-x86_64_v0.6/rapl/AbsPower/KubeletOnly/GradientBoostingRegressorTrainer_1.zip
try getting archieved model from URL: https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-db/main/models/Linux-4.15.0-213-generic-x86_64_v0.6/rapl/AbsPower/KubeletOnly/GradientBoostingRegressorTrainer_1.zip for AbsPower
<Response [200]>
load model from config:  /usr/local/src/estimate/../util/download/rapl/AbsPower
```

### What did you expect to happen?

I want to know this log is normal?

### How can we reproduce it (as minimally and precisely as possible)?

.

### Anything else we need to know?

.

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

sunya-ch, 2023-09-26:
@Lai-Kenny Shall we close this issue?

rootfs, 2023-09-26:
closing now. Reopen if new info comes

closed 2023-09-26 by rootfs