sustainable-computing-io/kepler #910: libbpf not working

jichenjc, 2023-09-05:
### What happened?

```
I0905 09:53:14.751132       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0905 09:53:14.796442       1 qat.go:35] Failed to init qat-telemtry err: could not get qat status exit status 127
I0905 09:53:14.808681       1 exporter.go:158] Kepler running on version: v0.5.4-47-g02c16c1-dirty
I0905 09:53:14.808747       1 config.go:271] using gCgroup ID in the BPF program: true
I0905 09:53:14.808798       1 config.go:273] kernel version: 4.18
I0905 09:53:14.808835       1 exporter.go:170] LibbpfBuilt: false, BccBuilt: true

```

even though I set the LIBBPF_ATTACH in configmap to true, still reporting above ..

### What did you expect to happen?

libbpf default enable, try it and if it failed. go bcc

### How can we reproduce it (as minimally and precisely as possible)?

always

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2023-09-05:
did you use the `quay.io/sustainable_computing_io/kepler:latest-libbpf` image?

jichenjc, 2023-09-06:
no, it's just a flag .. I will use the `quay.io/sustainable_computing_io/kepler:latest-libbpf` give a try

sunya-ch, 2023-09-06:
```
LibbpfBuilt: false, BccBuilt: true
```
is showing the build tag of the image. We may remove the config `LIBBPF_ATTACH` to avoid this confusion because it is exclusion between bcc and libbpf. It is set by the image tag.

jichenjc, 2023-09-06:
ok, I will read more about this ,I thought it's `config` item which seems to be wrong
let me adjust my previous PR

closed 2023-09-12 by jichenjc