sustainable-computing-io/kepler #913: single build too long

jichenjc, 2023-09-05:
### What would you like to be added?

try to run make , every time it takes around 4+ min which is way more slower than before (I remember it's only about 1-2 min) before, it's ok to run this once slow but later on should be faster 

full log below
```
[root@jitest48 kepler]# date;make;date
Tue Sep  5 10:08:26 UTC 2023
GO111MODULE=off go get -u github.com/go-bindata/go-bindata/...
./hack/bindata.sh
go mod tidy -v
go mod vendor
./automation/presubmit-tests/gofmt.sh
Checking go format
echo BIN_TIMESTAMP==2023-09-05T10:08:29Z
BIN_TIMESTAMP==2023-09-05T10:08:29Z
/bin/podman build -t quay.io/sustainable_computing_io/kepler:v0.5.4-47-g02c16c1-linux-amd64-bcc \
        -f /root/kepler/build/Dockerfile.bcc.kepler \
        --network host \
        --build-arg SOURCE_GIT_TAG=v0.5.4-47-g02c16c1 \
        --build-arg BIN_TIMESTAMP=2023-09-05T10:08:29Z \
        --platform="linux/amd64" \
        .
[1/2] STEP 1/11: FROM quay.io/sustainable_computing_io/kepler_builder:ubi-9-bcc-0.26-go1.18 AS builder
[1/2] STEP 2/11: ARG BIN_TIMESTAMP
--> Using cache 646807b3cf5425d33d785a82b01215fae199b001cdd868c03a1defb0398e452f
--> 646807b3cf54
[1/2] STEP 3/11: ARG SOURCE_GIT_TAG
--> Using cache 136dcf519b6a8b957df5d0d9b5e7b42d74b96e7c15f057f058f570ddf365e373
--> 136dcf519b6a
[1/2] STEP 4/11: LABEL name=kepler-builder
--> Using cache 97607d165afdf8716b3f1d69a6e3087dacee10a7d7470e5e360ba6856d4640e4
--> 97607d165afd
[1/2] STEP 5/11: ENV GOPATH=/opt/app-root GO111MODULE=on GOROOT=/usr/local/go
--> Using cache a688e80219f3095633ec0717ab56cfb772f85e9a94dc538b2415ab5d7b68115b
--> a688e80219f3
[1/2] STEP 6/11: ENV PATH=$GOPATH/bin:$GOROOT/bin:$PATH
--> Using cache 3f3d1d28976b1ea72d311bb439f438a3bca8abe6e0a11632ee059838cd868405
--> 3f3d1d28976b
[1/2] STEP 7/11: WORKDIR $GOPATH/src/github.com/sustainable-computing-io/kepler
--> Using cache 72b65256614a75df51210d63d7108ba4074ec5988e82e77dd525a03052b4def6
--> 72b65256614a
[1/2] STEP 8/11: COPY . .
--> 3a758fbc9e45
[1/2] STEP 9/11: RUN mkdir -p data
--> 4b595b0179dc
[1/2] STEP 10/11: RUN make build SOURCE_GIT_TAG=$SOURCE_GIT_TAG BIN_TIMESTAMP=$BIN_TIMESTAMP
rm -rf _output/bin
go mod tidy -v
go: downloading github.com/aquasecurity/libbpfgo v0.4.9-libbpf-1.2.0
go: downloading github.com/iovisor/gobpf v0.2.1-0.20221005153822-16120a1bf4d4
go: downloading github.com/jaypipes/ghw v0.10.0
go: downloading golang.org/x/sys v0.6.0
go: downloading k8s.io/klog/v2 v2.80.0
go: downloading github.com/onsi/ginkgo/v2 v2.4.0
go: downloading github.com/onsi/gomega v1.22.1
go: downloading github.com/containerd/cgroups v1.1.0
go: downloading github.com/opencontainers/runc v1.1.8
go: downloading k8s.io/api v0.25.3
go: downloading github.com/containerd/cgroups/v3 v3.0.1
go: downloading github.com/opencontainers/runtime-spec v1.0.3-0.20210326190908-1c3f411f0417
go: downloading github.com/prometheus/client_golang v1.13.1
go: downloading github.com/jszwec/csvutil v1.7.1
go: downloading github.com/prometheus/client_model v0.3.0
go: downloading github.com/prometheus/common v0.37.0
go: downloading k8s.io/apimachinery v0.25.3
go: downloading k8s.io/client-go v0.25.3
go: downloading github.com/prometheus/prometheus v0.40.3
go: downloading github.com/NVIDIA/go-nvml v0.11.6-0
go: downloading github.com/go-logr/logr v1.2.3
[... 527 lines cut ...]
--> 01a1f51b60f6
[2/2] STEP 9/12: RUN mkdir -p /usr/share/kepler/kernel_sources
--> Using cache 7098fbc722895dfd79064859ba20a5e63fdd47908ce727ea6f314ebb23e53fec
--> 7098fbc72289
[2/2] STEP 10/12: COPY --from=quay.io/sustainable_computing_io/kepler_kernel_source_images:ubi8 /usr/src/kernels /usr/share/kepler/kernel_sources
--> Using cache 8ac5475cf6416ea6dee3c705670df5585101b56abdd9af3fc0d2f05dd50496a6
--> 8ac5475cf641
[2/2] STEP 11/12: COPY --from=quay.io/sustainable_computing_io/kepler_kernel_source_images:ubi9 /usr/src/kernels /usr/share/kepler/kernel_sources
--> Using cache 97bf7f6e02ec087e01e443152c9a2a9c9190e453025ee8ad26b917442c9d17a1
--> 97bf7f6e02ec
[2/2] STEP 12/12: ENTRYPOINT ["/usr/bin/kepler"]
--> Using cache a8bf2f59cbbc520eb8b7fe2932c5c36f50057c1931528d087fade3da9d5c153e
[2/2] COMMIT quay.io/sustainable_computing_io/kepler:v0.5.4-47-g02c16c1-linux-amd64-bcc
--> a8bf2f59cbbc
Successfully tagged quay.io/sustainable_computing_io/kepler:v0.5.4-47-g02c16c1-linux-amd64-bcc
a8bf2f59cbbc520eb8b7fe2932c5c36f50057c1931528d087fade3da9d5c153e
/bin/podman tag quay.io/sustainable_computing_io/kepler:v0.5.4-47-g02c16c1-linux-amd64-bcc quay.io/sustainable_computing_io/kepler:latest
Tue Sep  5 10:12:52 UTC 2023
[root@jitest48 kepler]#

```

### Why is this needed?

it's ok to run this once slow but later on should be faster

marceloamaral, 2023-09-06:
The build process should be fast, the slow part is probably downloading the go libs.

jichenjc, 2023-09-06:
why download every time? maybe need some way to speed up..

SamYuan1990, 2023-09-18:
ref https://github.com/sustainable-computing-io/kepler/blob/main/vendor/modules.txt#L281
hence I suppose those packages are out of vendor folder and it download every time as for CI as an clean env, there no cache for those package.

Saurabhkr952, 2023-11-10:
After implementing cache mounts, I've observed a significant improvement in build times. The details are summarized below:

**Environment:**
EC2 AMI with 2 vCPU and 8 GB RAM

**Build Performance Summary:**
- **Initial Build without changes:**
  Duration: `128.6 seconds`

- **Build with Makefile Changes:**
  Duration: `89 seconds` (few files are cached)

- **Now I have implemented cache mounts:**
  Makefile further modified
  Duration: `8.7 seconds` (No library downloads during the build)

Considering these improvements, do you think it would be beneficial to create a pull request for these changes? I'm happy to contribute back to the project if these modifications align with the community's goals.

Looking forward to your guidance on this. Thanks!

rootfs, 2023-11-10:
@Saurabhkr952 this stat looks very promising!

How about we go this way in the Makefile targets? 

In most cases the vendor pkgs are not changed, we can use current vendor folder without having to run `go mod tidy`. The build can done in a fast path. If a PR changes `go.mod` `go.sum`, then it is time for `go mod tidy` to download pkgs to ensure the vendor pkgs are consistent, the build goes the long running way. This could save much CI time without breaking the vendor pkgs.

Saurabhkr952, 2023-11-11:
@rootfs  Yes, this approach looks more efficient.

The current build process runs 'go mod tidy' & 'go mod vendor' in every build, which can be time-consuming. Running go mod tidy unnecessarily in such cases can lead to longer CI build times.

We can implement a conditional check in the **Makefile** to determine whether there are changes in `go.mod` or `go.sum` files. 
If no changes are detected, we can skip the  'go mod tidy' & 'go mod vendor' step.

rootfs, 2023-11-11:
sounds good @Saurabhkr952 ! Look forward to your PR

Saurabhkr952, 2023-11-11:
I made in the Makefile by implementing a conditional check to optimize the build process. Here's the code snippet:

```bash
tidy-vendor:
    @if [ -n "$$(git status --porcelain go.mod go.sum)" ]; then \
        echo "Changes detected in go.mod or go.sum"; \
        go mod tidy -v; \
        go mod vendor; \
    else \
        echo "No changes in go.mod or go.sum"; \
    fi
```

Even after skipping go mod tidy -v and go mod vendor, few files are cached as well but the build is still taking about 82 seconds due to the `format` target defined in the Makefile, which runs a script. 
 

To further optimize the build time, I'm considering  cache mount for the format target. Aiming to reduce the overall build time.

In addition, I've proposed a small adjustment in the Dockerfile. By copying over go.mod and go.sum files and using a cache, we can avoid downloading dependencies repeatedly, making our builds even faster. 

```Dockerfile
COPY go.mod go.sum ./
RUN --mount=type=cache,target=/go/cache \
go mod download -x
```

so it will take time 8-10 seconds to build the image as mentioned yesterday .

should I go ahead and submit a pull request?

rootfs, 2023-11-11:
@Saurabhkr952 that's cool! Can you make each of the improvements their own PR so we can track the improvement. Thanks.

SamYuan1990, 2023-11-16:
@jichenjc , I suppose by switch to go vendor folder we will not download the go mod's during container build, I will close this issue. Please let me know if further questions for this issue and reopen then....

closed 2023-11-16 by SamYuan1990