sustainable-computing-io/kepler #916: bcc/libbpf different behavior in attach event

jichenjc, 2023-09-06:
### What happened?

compare with same machine but with libbpf and bcc image logs:

libbpf version:
```
I0906 03:44:15.849230       1 libbpf_attacher.go:149] failed to attach perf event cpu_instructions_hc_reader: failed to open bpf perf event on cpu 0: no such file or directory
I0906 03:44:15.849254       1 libbpf_attacher.go:149] failed to attach perf event cache_miss_hc_reader: failed to open bpf perf event on cpu 0: no such file or directory
I0906 03:44:15.849262       1 libbpf_attacher.go:149] failed to attach perf event cpu_cycles_hc_reader: failed to open bpf perf event on cpu 0: no such file or directory
I0906 03:44:15.849270       1 libbpf_attacher.go:149] failed to attach perf event cpu_ref_cycles_hc_reader: failed to open bpf perf event on cpu 0: no such file or directory
```

I saw this in libbpf version (libbpf-latest) only but not on bcc version
which supposed to report ? 
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/bpfassets/attacher/bcc_attacher.go#L120 

### What did you expect to happen?

libbpf/bcc should behavior same in identical env? 

### How can we reproduce it (as minimally and precisely as possible)?

compare latest version

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

marceloamaral, 2023-09-06:
Is it a VM or a BM?

jichenjc, 2023-09-06:
it's a VM, but doesn't matter, I think they should be identical, do they?

marceloamaral, 2023-09-06:
Where is the bcc logs? I can only see the libbpf.

So, both bcc and libbpf should show warnings that the hardware counters cannot be collected because it is a VM.

jichenjc, 2023-09-06:
I didn't paste it as I can't see in my log output..

marceloamaral, 2023-09-06:
In the BCC code we return if a hardware counter cannot be attached: https://github.com/sustainable-computing-io/kepler/blob/main/pkg/bpfassets/attacher/bcc_attacher.go#L115

The libbpf try to attach all hardware counters regardless if there was an error: https://github.com/sustainable-computing-io/kepler/blob/main/pkg/bpfassets/attacher/libbpf_attacher.go#L144

jichenjc, 2023-09-06:
ok, will read more about the logic and see whether something can be improved

jichenjc, 2023-09-07:
ok, I think above logic make sense so this should not be a problem

closed 2023-09-07 by jichenjc