sustainable-computing-io/kepler #917: process can't be found in logs

jichenjc, 2023-09-06:
### What happened?

both libbpf/bcc version in a VM env

seems I didn't see this eror before ..
```
I0906 04:00:36.936517       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294967629 (command=containerd-shim): failed to open cgroup description file for pid 244889: open /proc/244889/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.936614       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294967297 (command=kauditd): failed to open cgroup description file for pid 41: open /proc/41/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.936673       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294967297 (command=ksoftirqd/3): failed to open cgroup description file for pid 32: open /proc/32/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.936744       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294968214 (command=kube-apiserver): failed to open cgroup description file for pid 147639: open /proc/147639/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.936792       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294967297 (command=kworker/1:1H): failed to open cgroup description file for pid 317: open /proc/317/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.936829       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294967297 (command=auditd): failed to open cgroup description file for pid 726: open /proc/726/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.936916       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294967297 (command=vhost-7793): failed to open cgroup description file for pid 7808: open /proc/7808/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.936989       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294967297 (command=worker): failed to open cgroup description file for pid 213922: open /proc/213922/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.937043       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294967297 (command=worker): failed to open cgroup description file for pid 218719: open /proc/218719/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.937081       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294967297 (command=kworker/2:1H): failed to open cgroup description file for pid 288: open /proc/288/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.937111       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294968679 (command=local-path-prov): failed to open cgroup description file for pid 148905: open /proc/148905/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.937137       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294967297 (command=sssd_nss): failed to open cgroup description file for pid 826: open /proc/826/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.937186       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294967297 (command=worker): failed to open cgroup description file for pid 224279: open /proc/224279/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.937263       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294967297 (command=sedispatch): failed to open cgroup description file for pid 728: open /proc/728/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.937836       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294968214 (command=kube-apiserver): failed to open cgroup description file for pid 147637: open /proc/147637/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.937891       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294967297 (command=worker): failed to open cgroup description file for pid 224262: open /proc/224262/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.937927       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294968454 (command=kubelet): failed to open cgroup description file for pid 147762: open /proc/147762/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.937957       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294967297 (command=ksoftirqd/1): failed to open cgroup description file for pid 20: open /proc/20/cgroup: no such file or directory, set containerID=system_processes
I0906 04:00:36.937979       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 4294967297 (command=worker): failed to open cgroup description file for pid 213918: open /proc/213918/cgroup: no such file or directory, set containerID=system_processes

```

### What did you expect to happen?

no such error report

### How can we reproduce it (as minimally and precisely as possible)?

latest code 

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

marceloamaral, 2023-09-06:
Maybe something has changed in the log settings, but it's important to note that the warnings are correct.

When we create a cgroup handler, we check if we can use any of the PIDs inside the container. This is because a container might have some short-lived processes, and then the path `proc/$PID/` won't exist for them. However, this path will exist for long-running processes. Therefore, we need to test all PIDs until we find one that can be used.

Note that OS processes don't have cgroup metrics, so they will always trigger a warning. To avoid repeatedly trying to create a cgroup handler for these OS processes, we could implement a cache that keeps track of them.

Considering this, it might be also a good idea to adjust the log verbosity for this warning.

jichenjc, 2023-09-07:
I set the verbostiy to 5 and start to see those logs , but I didn't see this kind of info before even set to 5
need read more about the comments on the reason, thanks for the detailed explanation

sunya-ch, 2023-09-07:
Is there any container reporting value to Prometheus? If no other container than just system_process (such as kepler itself), it might be the issue of proc path is not in the correct pid namespace.

sunya-ch, 2023-11-08:
@jichenjc Any update on this issue?

jichenjc, 2023-11-08:
>>>Is there any container reporting value to Prometheus? If no other container than just system_process (such as kepler itself), it might be the issue of proc path is not in the correct pid namespace.

sorry, I missed this ,and checked my env seems I can get metrics from the container 
so it should not be the container not report and /proc path is not incorrect

given cgroup (and kubelet ) is not used by default now ,will this not be a problem ?

sunya-ch, 2023-11-09:
Yes, the cgroup metric is not used by default anymore. However, we still use the cgroup to map the pid to the container name. If you can get the metric of Kepler-exporter container, it should be okay.

jichenjc, 2023-11-09:
ok, given above comments ,let me close this issue

closed 2023-11-09 by jichenjc