sustainable-computing-io/kepler #919: Installing Kepler on EKS | kepler-exporter stuck at "ContainerCreating"

husky-parul, 2023-09-07:
### What happened?

When trying to deploy Kepler on EKS using the [documentation]( https://sustainable-computing.io/installation/kepler/ ) 
```
git clone --depth 1 git@github.com:sustainable-computing-io/kepler.git
cd ./kepler
make build-manifest OPTS="CI_DEPLOY PROMETHEUS_DEPLOY"
kubectl apply -f _output/generated-manifest/deployment.yaml
```
the kepler-exporter pods are stuck at `ContainerCreating` with the error

`MountVolume.SetUp failed for volume "proc" : hostPath type check failed: /proc-host is not a directory`

cc: @knarayan

### What did you expect to happen?

We expect Kepler exporter pod to be deployed

### How can we reproduce it (as minimally and precisely as possible)?

```
git clone --depth 1 git@github.com:sustainable-computing-io/kepler.git
cd ./kepler
make build-manifest OPTS="CI_DEPLOY PROMETHEUS_DEPLOY"
kubectl apply -f _output/generated-manifest/deployment.yaml
```

From describe po/kepler-exporter
```
Events:
  Type     Reason       Age                   From     Message
  ----     ------       ----                  ----     -------
  Warning  FailedMount  50m (x66 over 18h)    kubelet  Unable to attach or mount volumes: unmounted volumes=[proc], unattached volumes=[proc cfm redfish kube-api-access-5xgvr usr-src lib-modules tracing]: timed out waiting for the condition
  Warning  FailedMount  30m (x69 over 17h)    kubelet  Unable to attach or mount volumes: unmounted volumes=[proc], unattached volumes=[kube-api-access-5xgvr usr-src lib-modules tracing proc cfm redfish]: timed out waiting for the condition
  Warning  FailedMount  19m (x96 over 17h)    kubelet  Unable to attach or mount volumes: unmounted volumes=[proc], unattached volumes=[usr-src lib-modules tracing proc cfm redfish kube-api-access-5xgvr]: timed out waiting for the condition
  Warning  FailedMount  5m26s (x65 over 18h)  kubelet  Unable to attach or mount volumes: unmounted volumes=[proc], unattached volumes=[cfm redfish kube-api-access-5xgvr usr-src lib-modules tracing proc]: timed out waiting for the condition
  Warning  FailedMount  29s (x544 over 18h)   kubelet  MountVolume.SetUp failed for volume "proc" : hostPath type check failed: /proc-host is not a directory

````

### Kepler image tag

<details>
    Image:         quay.io/sustainable_computing_io/kepler:latest
</details>

### Kubernetes version

<details>

```console
$ kubectl version
Client Version: v1.28.1
Kustomize Version: v5.0.4-0.20230601165947-6ce0bf390ce3
Server Version: v1.26.7+IKS
WARNING: version difference between client (1.28) and server (1.26) exceeds the supported minor version skew of +/-1
```

</details>

### Cloud provider or bare metal

<details>
IBM Cloud
</details>

### OS version

Since it is a managed Kubernetes cluster on IBM cloud

````
kubectl get nodes -o wide
NAME          STATUS   ROLES    AGE   VERSION       INTERNAL-IP   EXTERNAL-IP   OS-IMAGE             KERNEL-VERSION      CONTAINER-RUNTIME
10.242.64.4   Ready    <none>   7d    v1.26.7+IKS   10.242.64.4   10.242.64.4   Ubuntu 20.04.6 LTS   5.4.0-156-generic   containerd://1.7.4
10.242.64.5   Ready    <none>   7d    v1.26.7+IKS   10.242.64.5   10.242.64.5   Ubuntu 20.04.6 LTS   5.4.0-156-generic   containerd://1.7.4
[parsingh@parsingh-thinkpadx1nanogen2 kube-prometheus]$ 

```

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

knarayan, 2023-09-11:
Thanks @sunya-ch, excluding `CI_DEPLOY` seems working. Below are the steps followed:
```
git clone --depth 1 git@github.com:sustainable-computing-io/kepler.git
cd ./kepler
make build-manifest OPTS="PROMETHEUS_DEPLOY"
kubectl apply -f _output/generated-manifest/deployment.yaml
```
The kepler-exporter pods are now in running state.
```
% kubectl get pods -n kepler
NAME                    READY   STATUS    RESTARTS   AGE
kepler-exporter-bvtlr   1/1     Running   0          20m
kepler-exporter-pj9cj   1/1     Running   0          20m
```

sunya-ch, 2023-09-11:
> Thanks @sunya-ch, excluding `CI_DEPLOY` seems working. Below are the steps followed:
>
> ```
> git clone --depth 1 git@github.com:sustainable-computing-io/kepler.git
> cd ./kepler
> make build-manifest [...]

@knarayan Nice. Would you mind me closing the issue?

closed 2023-09-11 by husky-parul