sustainable-computing-io/kepler #923: GPU power utilization showing up in system instead of pod

rahul-shiv, 2023-09-10:
### What happened?

I tried out using kepler to monitor GPU power utilization of a pod using kind and microk8s deployments and the gpu power usage showed up in system instead of the pod where the workload was running.

For the kind use case, I used a custom version of kind to support GPUs : https://jacobtomlinson.dev/posts/2022/quick-hack-adding-gpu-support-to-kind/.
Kind kepler build command used: make build-manifest OPTS="CI_DEPLOY PROMETHEUS_DEPLOY"

For the microk8s use case, I used the bare-metal version of kepler.
microk8s kepler build command used: make build-manifest OPTS="BM_DEPLOY PROMETHEUS_DEPLOY"

Are either of these platforms supported and if so, is there something I am doing wrong? If not, what would be the right way to deploy kepler and kubernetes on a single node setup to try out kepler's GPU power monitoring capability?

Please let me know if I can provide any relevant logs.

### What did you expect to happen?

Expected the power utilization to show up in the respective pod where the workload was running instead of the "SYSTEM" namespace.

### How can we reproduce it (as minimally and precisely as possible)?

For kind case, run a gpu workload using custom version of kind to support GPUs : https://jacobtomlinson.dev/posts/2022/quick-hack-adding-gpu-support-to-kind/

No specific steps for microk8s case.

### Kepler image tag

<details>
Built from github main branch, pulled on Sep 8th.
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```
Client Version: v1.28.1
Kustomize Version: v5.0.4-0.20230601165947-6ce0bf390ce3
Server Version: v1.21.2
</details>

### Cloud provider or bare metal

<details>
Bare Metal
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

NAME="Ubuntu"
VERSION="20.04.4 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.4 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

Linux hongpeng-System-Product-Name 5.13.0-27-generic #29~20.04.1-Ubuntu SMP Fri Jan 14 00:32:30 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

marceloamaral, 2023-09-11:
To associate the node GPU power to the pod, Kepler collects the GPU utilization per process.
We need to double check  in logs if Kepler could collect this information. Do you see any of the following errors?
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/power/accelerator/gpu/power.go#L86
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/power/accelerator/gpu/power.go#L89
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/power/accelerator/gpu/source/gpu_nvml.go#L121

rahul-shiv, 2023-09-11:
Below are the logs I see for kubectl logs -n kepler daemonset/kepler-exporter:

```
I0910 23:02:44.232327       1 gpu_nvml.go:62] found 2 gpu devices
I0910 23:02:44.237729       1 gpu_nvml.go:73] GPU 0 NVIDIA GeForce RTX 2070 SUPER
I0910 23:02:44.243132       1 gpu_nvml.go:73] GPU 1 NVIDIA GeForce RTX 2080 Ti
I0910 23:02:44.243148       1 gpu.go:40] Using nvml to obtain gpu power
I0910 23:02:44.278218       1 qat.go:35] Failed to init qat-telemtry err: could not get qat status exit status 127
I0910 23:02:44.286626       1 exporter.go:158] Kepler running on version: 2ec465f
I0910 23:02:44.286650       1 config.go:270] using gCgroup ID in the BPF program: true
I0910 23:02:44.286695       1 config.go:272] kernel version: 5.13
I0910 23:02:44.286806       1 config.go:297] The Idle power will be exposed. Are you running on Baremetal or using single VM per node?
I0910 23:02:44.286819       1 exporter.go:170] LibbpfBuilt: false, BccBuilt: true
I0910 23:02:44.286873       1 config.go:203] kernel source dir is set to /usr/share/kepler/kernel_sources
I0910 23:02:44.286941       1 exporter.go:189] EnabledBPFBatchDelete: true
I0910 23:02:44.287006       1 power.go:54] use sysfs to obtain power
I0910 23:02:44.287052       1 redfish.go:173] failed to initialize node credential: no supported node credential implementation
I0910 23:02:44.288800       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0910 23:02:45.312054       1 exporter.go:204] Initializing the GPU collector
I0910 23:02:51.314279       1 watcher.go:66] Using in cluster k8s config
I0910 23:02:52.194939       1 bcc_attacher.go:129] Successfully load eBPF module from using bcc
I0910 23:02:52.194997       1 bcc_attacher.go:186] Successfully load eBPF module from bcc with option: [-DMAP_SIZE=10240 -DNUM_CPUS=24]
I0910 23:02:52.241582       1 container_energy.go:109] Using the Ratio/DynPower Power Model to estimate Container Platform Power
I0910 23:02:52.241598       1 container_energy.go:118] Using the Ratio/DynPower Power Model to estimate Container Component Power
I0910 23:02:52.241605       1 process_power.go:108] Using the Ratio/DynPower Power Model to estimate Process Platform Power
I0910 23:02:52.241611       1 process_power.go:117] Using the Ratio/DynPower Power Model to estimate Process Component Power
I0910 23:02:52.241714       1 node_platform_energy.go:53] Using the LinearRegressor/AbsPower Power Model to estimate Node Platform Power
I0910 23:02:52.241806       1 exporter.go:276] Started Kepler in 7.955211761s
```

From this it looks ok. Please let me know if this is the wrong place to look for logs or if you need any other info. Thanks!

rootfs, 2023-09-11:
@rahul-shiv can you get all the kepler metrics and let's take a look?

rahul-shiv, 2023-09-11:
Kepler metrics using command : kubectl exec -ti -n kepler kepler-exporter -- bash  -c "curl localhost:9102/metrics"

attached in pastebin : https://pastebin.com/RLmsiJsB

Please let me know if these were not the metrics requested.

marceloamaral, 2023-09-12:
@rahul-shiv which workload are oy trying to see? `gpu-workload`?
If yes, I can see the values in the logs that you shared.

```
kepler_container_gpu_joules_total{command="",container_id="7f44c190a296ce466bf4e9ecda5fd238213c54affe6f15202563cd060d2c7e19",container_name="gpu-workload-ontainer",container_namespace="workload",mode="dynamic",pod_name="gpu-workload"} 8.637
kepler_container_gpu_joules_total{command="",container_id="7f44c190a296ce466bf4e9ecda5fd238213c54affe6f15202563cd060d2c7e19",container_name="gpu-workload-container",container_namespace="workload",mode="idle",pod_name="gpu-workload"} 34.269
```

rahul-shiv, 2023-09-12:
Yes, I'm interested in the gpu-workload pod. And am expecting roughly 100 - 120W of power to be used in this pod. Considering this is the only gpu workload running on my system. However, this is being aggregated as part of "system_processes" which shouldn't have such high values if kepler is working as expected.

kepler_container_joules_total{command="",container_id="system_processes",container_name="system_processes",container_namespace="system",mode="dynamic",pod_name="system_processes"} 1.515297999e+06
kepler_container_joules_total{command="",container_id="system_processes",container_name="system_processes",container_namespace="system",mode="idle",pod_name="system_processes"} 2.5718398740000003e+06

This observation is also evident in the grafana dashboard

1 for System:

marceloamaral, 2023-09-13:
@rahul-shiv Definitely there is something wrong, actually I found the bug and submitted the PR #927 to fix it.

It is expected the idle power be divided between all process, but the dynamic power should be only associated to the process that is running in the GPU....

After the PR get merged, can you please verify it again?

rahul-shiv, 2023-09-13:
Definitely! Thanks for the quick turn around!

Can you also please take a look at why there is no cpu energy utilization per pod in the metrics? I'm happy to raise a separate issue for that if required.

The total energy seems to be captured:

```
# TYPE kepler_node_package_joules_total counter
kepler_node_package_joules_total{instance="rahul-cluster-control-plane",mode="dynamic",package="0",source="rapl"} 1.507904837e+06
kepler_node_package_joules_total{instance="rahul-cluster-control-plane",mode="idle",package="0",source="rapl"} 2.559987638e+06
```

but kepler_container_package_joules_total seems to be 0 for all the pods/containers. Am I missing something?

marceloamaral, 2023-09-13:
@rahul-shiv actually, I think the previous PR will not fix the problem. This was for estimating process power not container/pod power.... There was not bug for the container.

So, the problem might be related to something else...

The package power is divided based on the CPU instructions and the kepler metrics `kepler_container_cpu_instructions_total` has no information about the `gpu-workload`....

Do you see CPU utilization for the `gpu-workload`? If yes, there might be a problem converting the PID to ContainerID for this process....
If the CPU utilization is very small, maybe BPF are not collecting metrics  and this can create all the problems...

Also, for some reason, part of the GPU utilization is being associated to the `system_process`.... Probably related to the problem of converting the PID to ContainerID....

Could you please get the `/proc/$PID/cgroup` from the PID of the `gpu-workload`?

Also, can you increase the Kepler log verbosity to 5 and share the logs? It might help to debug it.

rahul-shiv, 2023-09-14:
Oh ok, that makes sense. Here are the kepler logs for 1 collection cycle with verbosity 5 : https://pastebin.com/pbN5eqZM

I definitely see cpu utilization as well for the workload (using top). As you mentioned, I guess the issue is common between the gpu and cpu where kepler isn't able to map the PID to the Container.

Output of /proc/$PID/cgroup from the host and the same is present in the /proc-host path in the container:

```
rshivapr@hongpeng-System-Product-Name:~$ cat /proc/28899/cgroup
13:freezer:/system.slice/docker-701c443334f086aa29e06aba3e42dc979a9605f794b0b19af1f8304bbe5abe7a.scope/kubelet/kubepods/besteffort/pode4b97838-73cc-4b94-9f3a-88153d3c8e31/c664156b4a3cd7658b484ba00f1cb35c037edd2a427931ed8dd9fe20f35f84a2
12:rdma:/system.slice/docker-701c443334f086aa29e06aba3e42dc979a9605f794b0b19af1f8304bbe5abe7a.scope
11:blkio:/system.slice/docker-701c443334f086aa29e06aba3e42dc979a9605f794b0b19af1f8304bbe5abe7a.scope/kubelet/kubepods/besteffort/pode4b97838-73cc-4b94-9f3a-88153d3c8e31/c664156b4a3cd7658b484ba00f1cb35c037edd2a427931ed8dd9fe20f35f84a2
10:cpu,cpuacct:/system.slice/docker-701c443334f086aa29e06aba3e42dc979a9605f794b0b19af1f8304bbe5abe7a.scope/kubelet/kubepods/besteffort/pode4b97838-73cc-4b94-9f3a-88153d3c8e31/c664156b4a3cd7658b484ba00f1cb35c037edd2a427931ed8dd9fe20f35f84a2
9:hugetlb:/system.slice/docker-701c443334f086aa29e06aba3e42dc979a9605f794b0b19af1f8304bbe5abe7a.scope/kubelet/kubepods/besteffort/pode4b97838-73cc-4b94-9f3a-88153d3c8e31/c664156b4a3cd7658b484ba00f1cb35c037edd2a427931ed8dd9fe20f35f84a2
8:cpuset:/system.slice/docker-701c443334f086aa29e06aba3e42dc979a9605f794b0b19af1f8304bbe5abe7a.scope/kubelet/kubepods/besteffort/pode4b97838-73cc-4b94-9f3a-88153d3c8e31/c664156b4a3cd7658b484ba00f1cb35c037edd2a427931ed8dd9fe20f35f84a2
7:memory:/system.slice/docker-701c443334f086aa29e06aba3e42dc979a9605f794b0b19af1f8304bbe5abe7a.scope/kubelet/kubepods/besteffort/pode4b97838-73cc-4b94-9f3a-88153d3c8e31/c664156b4a3cd7658b484ba00f1cb35c037edd2a427931ed8dd9fe20f35f84a2
6:devices:/system.slice/docker-701c443334f086aa29e06aba3e42dc979a9605f794b0b19af1f8304bbe5abe7a.scope/kubelet/kubepods/besteffort/pode4b97838-73cc-4b94-9f3a-88153d3c8e31/c664156b4a3cd7658b484ba00f1cb35c037edd2a427931ed8dd9fe20f35f84a2
5:misc:/system.slice/docker-701c443334f086aa29e06aba3e42dc979a9605f794b0b19af1f8304bbe5abe7a.scope
4:perf_event:/system.slice/docker-701c443334f086aa29e06aba3e42dc979a9605f794b0b19af1f8304bbe5abe7a.scope/kubelet/kubepods/besteffort/pode4b97838-73cc-4b94-9f3a-88153d3c8e31/c664156b4a3cd7658b484ba00f1cb35c037edd2a427931ed8dd9fe20f35f84a2
3:net_cls,net_prio:/system.slice/docker-701c443334f086aa29e06aba3e42dc979a9605f794b0b19af1f8304bbe5abe7a.scope/kubelet/kubepods/besteffort/pode4b97838-73cc-4b94-9f3a-88153d3c8e31/c664156b4a3cd7658b484ba00f1cb35c037edd2a427931ed8dd9fe20f35f84a2
2:pids:/system.slice/docker-701c443334f086aa29e06aba3e42dc979a9605f794b0b19af1f8304bbe5abe7a.scope/kubelet/kubepods/besteffort/pode4b97838-73cc-4b94-9f3a-88153d3c8e31/c664156b4a3cd7658b484ba00f1cb35c037edd2a427931ed8dd9fe20f35f84a2
1:name=systemd:/system.slice/docker-701c443334f086aa29e06aba3e42dc979a9605f794b0b19af1f8304bbe5abe7a.scope/kubelet/kubepods/besteffort/pode4b97838-73cc-4b94-9f3a-88153d3c8e31/c664156b4a3cd7658b484ba00f1cb35c037edd2a427931ed8dd9fe20f35f84a2
0::/system.slice/docker-701c443334f086aa29e06aba3e42dc979a9605f794b0b19af1f8304bbe5abe7a.scope/system.slice/containerd.service

```
Oddly running nvidia-smi doesn't seem to be recognising the workload as a process running inside the node (docker container) and this pid isn't present in the node. 

Nvidia_smi Output from the host:

```
rshivapr@hongpeng-System-Product-Name:~$ nvidia-smi
Thu Sep 14 13:17:10 2023
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 470.94       Driver Version: 470.94       CUDA Version: 11.4     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  On   | 00000000:17:00.0 Off |                  N/A |
| 43%   62C    P2    72W / 215W |   1171MiB /  7982MiB |     51%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   1  NVIDIA GeForce ...  On   | 00000000:65:00.0 Off |                  N/A |
| 27%   42C    P8    21W / 250W |    161MiB / 11011MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1500      G   /usr/lib/xorg/Xorg                  4MiB |
|    0   N/A  N/A     76742      C   python                           1163MiB |
[... cut ...]
          containerPath: /usr/bin/nvidia-container-runtime
        - hostPath: /proc
          containerPath: /proc-host
        - hostPath: /usr/src
          containerPath: /usr/src
```

yaml file for gpu-workload:

```
apiVersion: v1
kind: Pod
metadata:
  name: gpu-workload
  namespace: workload
spec:
  containers:
  - name: gpu-workload-container
    image: rahulshiv/deep-learning-container:latest
    resources:
      limits:
        nvidia.com/gpu: 1
    command: ["python", "/app/model.py"]
```

rahul-shiv, 2023-09-20:
Hi @marceloamaral / @rootfs,

Is there any update or can I provide any more info?

marceloamaral, 2023-09-21:
Right, as we were suspecting, there is a PID resolution problem
```
I0914 17:59:49.104258       1 metric_collector.go:135] energy from pod/container (0 active processes): name: nvidia-device-plugin-daemonset-dhdxx/nvidia-device-plugin namespace: gpu-operator
	cgrouppid: 0 pid: [] comm:  containerid:52151bfe267a884580b1182c8076838c496b5dd14cf0f7cdfa9cab55260cbf0a
	Dyn ePkg (mJ): 0 (0) (eCore: 0 (0) eDram: 0 (0) eUncore: 0 (0)) eGPU (mJ): 0 (88995) eOther (mJ): 0 (0)
	Idle ePkg (mJ): 0 (0) (eCore: 0 (0) eDram: 0 (0) eUncore: 0 (0)) eGPU (mJ): 0 (1129431) eOther (mJ): 0 (0)
	CPUTime:  0 (0)
	NetTX IRQ: 0 (0)
	NetRX IRQ: 0 (0)
	Block IRQ: 0 (0)
	counters: map[cache_miss:0 (0) cpu_cycles:0 (0) cpu_instructions:0 (0) cpu_ref_cycles:0 (0) gpu_mem_util:0 (0) gpu_sm_util:0 (0)]
	cgroupfs: map[block_devices_used:0 (0) cgroupfs_cpu_usage_us:0 (0) cgroupfs_ioread_bytes:0 (0) cgroupfs_iowrite_bytes:0 (0) cgroupfs_kernel_memory_usage_bytes:0 (0) cgroupfs_memory_usage_bytes:0 (0) cgroupfs_system_cpu_usage_us:0 (0) cgroupfs_tcp_memory_usage_bytes:0 (0) cgroupfs_user_cpu_usage_us:0 (0)]
	kubelets: map[kubelet_cpu_usage:0 (15) kubelet_memory_bytes:0 (8728576)]
```

We can see that there is no listed PID in the container.

The cat `/proc/PID/cgroup` is from the host or container? Can you verify in the container as well as?

We might need to print more information, can you include a debug message in: https://github.com/sustainable-computing-io/kepler/blob/main/pkg/collector/container_hc_collector.go#L121
```
klog.Infoln("DEBUG:", containerID, ct.PID, comm, ct)
```
This new line will show how all PIDs are being classified and if the gpu workload PID is appearing in the eBPF code.

rahul-shiv, 2023-09-28:
Hi,

Sorry for the delayed response, I had an issue with my server.

@rootfs yes, but doesn't kepler require mounting of /proc to /proc-host as per : https://sustainable-computing.io/installation/local-cluster/?

@marceloamaral 

`
I0928 03:57:56.290340       1 container_hc_collector.go:118] failed to resolve container for cGroup ID 0 (command=apiservice-kick): failed to open cgroup description file for pid 1794616: open /proc/1794616/cgroup: no such file or directory, set containerID=system_processes
I0928 03:57:56.290344       1 container_hc_collector.go:121] DEBUG: system_processes 1794616 apiservice-kick {0 1794616 0 950489 908349 779 [0 0 0 0 0 0 0 0 0 0] [97 112 105 115 101 114 118 105 99 101 45 107 105 99 107 0]}
I0928 03:57:56.290349       1 container_hc_collector.go:121] DEBUG: system_processes 12425 containerd-shim {0 12425 0 4770078 3540661 18105 [0 0 0 0 0 0 0 0 0 0] [99 111 110 116 97 105 110 101 114 100 45 115 104 105 109 0]}
I0928 03:57:56.290355       1 container_hc_collector.go:121] DEBUG: system_processes 1775122 containerd-shim {0 1775122 0 1020247 1364571 1181 [0 0 0 0 0 0 0 0 0 0] [99 111 110 116 97 105 110 101 114 100 45 115 104 105 109 0]}
I0928 03:57:56.290381       1 container_hc_collector.go:118] failed to resolve container for cGroup ID 0 (command=bash): could not find cgroup description entry for pid 913103, set containerID=system_processes
I0928 03:57:56.290385       1 container_hc_collector.go:121] DEBUG: system_processes 913103 bash {0 913103 0 1468909 1172116 15530 [0 0 0 0 0 0 0 0 0 0] [98 97 115 104 0 0 0 0 0 0 0 0 0 0 0 0]}
I0928 03:57:56.290391       1 container_hc_collector.go:121] DEBUG: system_processes 1773073 containerd-shim {0 1773073 0 2110775 2773897 5117 [0 0 0 0 0 0 0 0 0 0] [99 111 110 116 97 105 110 101 114 100 45 115 104 105 109 0]}
I0928 03:57:56.290397       1 container_hc_collector.go:121] DEBUG: system_processes 1791237 local-path-prov {0 1791237 0 183272 108897 1531 [0 0 0 0 0 0 0 0 0 0] [108 111 99 97 108 45 112 97 116 104 45 112 114 111 118 0]}
I0928 03:57:56.290440       1 container_hc_collector.go:121] DEBUG: system_processes 1794455 prometheus {0 1794455 3 33037565 32876924 100064 [0 0 0 8 0 0 0 5 0 4] [112 114 111 109 101 116 104 101 117 115 0 0 0 0 0 0]}
I0928 03:57:56.290447       1 container_hc_collector.go:121] DEBUG: system_processes 1791797 python {0 1791797 1231 6710338156 8027065183 14014515 [0 35 0 1 0 0 0 509 0 614] [112 121 116 104 111 110 0 0 0 0 0 0 0 0 0 0]}
I0928 03:57:56.290453       1 container_hc_collector.go:121] DEBUG: system_processes 1762984 kube-apiserver {0 1762984 20 74015217 108666480 196845 [0 0 0 42 0 0 0 6 0 7] [107 117 98 101 45 97 112 105 115 101 114 118 101 114 0 0]}
I0928 03:57:56.290462       1 container_hc_collector.go:121] DEBUG: system_processes 1770219 containerd {0 1770219 2 24079374 22853356 118725 [0 1 0 0 0 0 0 6 0 3] [99 111 110 116 97 105 110 101 114 100 0 0 0 0 0 0]}
`

the pid of the gpu process (pid: 1791797) does end up in this debug.

The cat of /proc/$PID/cgroup is from the host. The same is available in the /proc-host path in the container. On finding the equivalent pid mapping to the container from /proc/$PID/status -> NSpid field, I was able to find the same cgroup output in the container's PID path as well.

rootfs, 2023-09-28:
> `container_hc_collector.go:121] DEBUG: system_processes 1791797 python`

this message is printed after process `1791797` has already been assigned to `system process`, i.e. not associated with any containers. In fact the log show all the processes are system processses, that indicate kepler cannot find their `cgroup` or match their `cgroup` to container paths.

rahul-shiv, 2023-09-28:
Yes, do you have any suggestions as to what could be missing or do you require any other info? I've also set /proc/sys/kernel/perf_event_paranoid to -1.

rootfs, 2023-09-28:
this is the pid issue, as seen the pastebin result:
```
I0914 17:59:48.988850       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 0 (command=python): failed to open cgroup description file for pid 4162253: open /proc/4162253/cgroup: no such file or directory, set containerID=system_processes
```

the pid seen inside of the kind cluster is different from seeing it on the host.

Depending how your kind cluster is configured, this could be a tricky issue. You can find similar things [here](https://github.com/kubernetes-sigs/kind/issues/3182)

rootfs, 2023-09-29:
@mcalman do you have a chance to share your kind setup? thanks

mcalman, 2023-09-29:
```
kind: Cluster
apiVersion: kind.x-k8s.io/v1alpha4
name: kind-cluster
nodes:
- role: control-plane
  image: kindest/node:v1.27.3@sha256:3966ac761ae0136263ffdb6cfd4db23ef8a83cba8a463690e98317add2c9ba72
  extraMounts:
  - hostPath: /proc
    containerPath: /proc-host
  - hostPath: /usr/src
    containerPath: /usr/src
```
@rootfs This is the setup I've been using for a single node cluster.

rahul-shiv, 2023-10-03:
@rootfs 

I checked out the link you shared. But seems like they aren't going forward with the feature proposed in the thread. 

I fail to understand why the pid of the host and container needs to be shared (I thought the whole point of the mounting hostpath /proc as /proc-host was to work around this).

Further, I tried to setup a cluster(with a vanilla version of kind) with @mcalman's config. But this seems to have the same issue with the processes not being resolved correctly.  At this point, I'm unable to get kepler working with even just cpu based processes when following the setup instructions in https://sustainable-computing.io/installation/local-cluster/ and I'm not sure what I'm missing.

marceloamaral, 2023-10-04:
Can I see your Kepler daemonset?
`kubectl describe daemonset -n kepler kepler-exporter`
I should have the:
```
   Mounts:
      /proc from proc (rw)
 Volumes:
   proc:
    Type:          HostPath (bare host directory volume)
    Path:          /proc-host
    HostPathType:  Directory
```

rahul-shiv, 2023-10-04:
Yes, looks like that's present.

`rshivapr@hongpeng-System-Product-Name:~/kepler/kepler$ kubectl describe daemonset -n kepler kepler-exporter
Name:           kepler-exporter
Selector:       app.kubernetes.io/component=exporter,app.kubernetes.io/name=kepler-exporter,sustainable-computing.io/app=kepler
Node-Selector:  <none>
Labels:         sustainable-computing.io/app=kepler
Annotations:    deprecated.daemonset.template.generation: 1
Desired Number of Nodes Scheduled: 1
Current Number of Nodes Scheduled: 1
Number of Nodes Scheduled with Up-to-date Pods: 1
Number of Nodes Scheduled with Available Pods: 1
Number of Nodes Misscheduled: 0
Pods Status:  1 Running / 0 Waiting / 0 Succeeded / 0 Failed
Pod Template:
  Labels:           app.kubernetes.io/component=exporter
                    app.kubernetes.io/name=kepler-exporter
                    sustainable-computing.io/app=kepler
  Service Account:  kepler-sa
  Containers:
   kepler-exporter:
    Image:      quay.io/sustainable_computing_io/kepler:latest
    Port:       9102/TCP
    Host Port:  0/TCP
    Command:
      /bin/sh
      -c
    Args:
      /usr/bin/kepler -v=5 -kernel-source-dir=/usr/share/kepler/kernel_sources -redfish-cred-file-path=/etc/redfish/redfish.csv
    Requests:
      cpu:     100m
      memory:  400Mi
    Liveness:  http-get http://:9102/healthz delay=10s timeout=10s period=60s #success=1 #failure=5
    Environment:
      NODE_IP:     (v1:status.hostIP)
      NODE_NAME:   (v1:spec.nodeName)
    Mounts:
      /etc/kepler/kepler.config from cfm (ro)
      /etc/redfish from redfish (ro)
      /lib/modules from lib-modules (rw)
      /proc from proc (rw)
      /sys from tracing (rw)
      /usr/src from usr-src (rw)
  Volumes:
   proc:
    Type:          HostPath (bare host directory volume)
    Path:          /proc-host
    HostPathType:  Directory
   usr-src:`

marceloamaral, 2023-10-05:
@rahul-shiv the PID seems to be wrong

Although you can see the PID `1791797` in `container_hc_collector.go:121`, the PID to container ID resolution seems to be using a wrong PID `4162197`......

```
I0928 03:57:56.290447       1 container_hc_collector.go:121] DEBUG: system_processes 1791797 python
```

```
I0914 17:59:48.992657       1 container_hc_collector.go:104] failed to resolve container for cGroup ID 0 (command=python): failed to open cgroup description file for pid 4162197: open /proc/4162197/cgroup: no such file or directory, set containerID=system_processes
```

Can you please change the following logs and verify the `/proc/PID/cgroup` again?

https://github.com/sustainable-computing-io/kepler/blob/main/pkg/collector/container_hc_collector.go#L118
```
klog.V(5).Infof("failed to resolve container for cGroup ID: %v, PID: %v, command=%s: %v, setting containerID=%s", ct.CGroupID, ct.PID, comm, err, c.systemProcessName)
```

https://github.com/sustainable-computing-io/kepler/blob/main/pkg/collector/container_hc_collector.go#L121
```
klog.Infoln("DEBUG:", containerID, ct.PID, comm, ct.ProcessRunTime, ct.CPUCycles)
```
After that, can you please also try to edit the kepler Daemonset to include in the command the option `enable-cgroup-id=false`?

rahul-shiv, 2023-10-05:
The discrepancy in the PID was actually because the gpu-workload runs for 100 epochs and dies. There was a gap between the logs produced previously and hence a new process/PID was spawned.

Here are the logs requested with no discrepancy wrt to the PID. We can see that the PID to be examined is 2729382 as per nvidia-smi output and I've attached the requested logs for the same. I think the mapping seems to be happening correctly, considering there are no logs for "failed to resolve container" for PID:2729382.

https://pastebin.com/pbN5eqZM

For enable-cgroup-id=false, **NOTE PID is now 2806951** 

https://pastebin.com/K27Z1HCd

marceloamaral, 2023-10-12:
@rahul-shiv can you see power metrics after using `enable-cgroup-id=false`?

I am not sure if it is getting the right container ID from the cgroup.
```
13:hugetlb:/system.slice/docker-2a2e129f1f337a56dfc04361faa684b041ab586453020e8dcaffa20f56c42322.scope/kubelet/kubepods/besteffort/pod528cf6a4-41eb-4e0f-a3ee-2d5ab261ba54/7e5eb81613289b20c592cbbf4215df1e06fb8f7e75ce73f0e2495834bb6d0f75
```
The container id resolution is getting the docker container ID `2a2e129f1f337a56dfc04361faa684b041ab586453020e8dcaffa20f56c42322`
But we need to verify if the container ID in your system is the last part in the string: `7e5eb81613289b20c592cbbf4215df1e06fb8f7e75ce73f0e2495834bb6d0f75`

Can you enable again the v=5 and also include the container name in the debug log?
```
klog.Infoln("DEBUG:", containerID, c.ContainersMetrics[containerID].ContainerName, ct.PID, comm, ct.ProcessRunTime, ct.CPUCycles)
```

rahul-shiv, 2023-10-12:
@marceloamaral  I wasn't able to see the power metrics on using enable-cgroup-id=false. Your conclusion is consistent with what I've found recently:

I ended up doing some digging and found a couple of possible bugs explaining why kepler is failing in my use cases.

Debugs used in  pkg/cgroup/resolve_container.go::  https://pastebin.com/nc4zW2Xu

I tried combinations of cgroup v1, v2 and kubernetes cluster using microk8s and kind.

I was unable to use kind for cgroup v2 as I wasn't able to get docker gpu working with v2.

Case 1 : Kind cluster with cgroup v1

I saw the below log for the pid associated with the workload:

`
I1012 05:30:24.033546       1 resolve_container.go:150] extractPodContainerIDfromPath for id 75741, path 13:hugetlb:/system.slice/docker-2fa3e04b676df750842faf017052dd37ea0cc5bc7259a3487a1718c7fe100c94.scope/kubelet/kubepods/besteffort/podbdd4097d-6795-404e-9bd8-6a1383386198/c79788e0da15a6597263eb2b9c51d05dd1a9a1d08c53c1161dc8c45d2dac6b38, containerId 2fa3e04b676df750842faf017052dd37ea0cc5bc7259a3487a1718c7fe100c94
I1012 05:30:24.033550       1 container_hc_collector.go:121] DEBUG: 2fa3e04b676df750842faf017052dd37ea0cc5bc7259a3487a1718c7fe100c94 75741 python 1328 6774507761
`
and the pod:
`
I1012 05:30:24.124687       1 metric_collector.go:141] energy from pod/container (0 active processes): name: gpu-workload/gpu-workload-container namespace: monitoring
	cgrouppid: 0 pid: [] comm:  containerid:**c79788e0da15a6597263eb2b9c51d05dd1a9a1d08c53c1161dc8c45d2dac6b38**
	Dyn ePkg (mJ): 0 (0) (eCore: 0 (0) eDram: 0 (0) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
	Idle ePkg (mJ): 0 (0) (eCore: 0 (0) eDram: 0 (0) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
	CPUTime:  0 (0)
	NetTX IRQ: 0 (0)
	NetRX IRQ: 0 (0)
	Block IRQ: 0 (0)
	counters: map[cache_miss:0 (0) cpu_cycles:0 (0) cpu_instructions:0 (0) cpu_ref_cycles:0 (0) gpu_mem_util:0 (0) gpu_sm_util:0 (0)]
	cgroupfs: map[block_devices_used:0 (0) cgroupfs_cpu_usage_us:0 (0) cgroupfs_ioread_bytes:0 (0) cgroupfs_iowrite_bytes:0 (0) cgroupfs_kernel_memory_usage_bytes:0 (0) cgroupfs_memory_usage_bytes:0 (0) cgroupfs_system_cpu_usage_us:0 (0) cgroupfs_tcp_memory_usage_bytes:0 (0) cgroupfs_user_cpu_usage_us:0 (0)]
	kubelets: map[kubelet_cpu_usage:0 (1108) kubelet_memory_bytes:0 (2397872128)]

`

In this case, we can see that kepler is extracting the wrong part of the path as the containerId and hence is unable to map it correctly. I haven't tried a fix for this but this could be a potential issue.

Case 2: Microk8s (BM_DEPLOY) and cgroup v1

`
I1012 07:14:08.033170       1 resolve_container.go:240] cgroup version 1
I1012 07:14:08.033182       1 resolve_container.go:254] extractPodContainerIDfromPath 1
I1012 07:14:08.033196       1 resolve_container.go:260] regexFindContainerIDPath2 /kubepods/besteffort/pod1f612ee7-054c-430a-b17b-dba1f6fbc2e5/1e600e5294ad7d6cf13dfb30b978e77fdd882672f892b5b36efed1eb5b9c74fc [/kubepods/besteffort/pod1f612ee7-054c-430a-b17b-dba1f6fbc2e5/1e600e5294ad7d6cf13dfb30b978e77fdd882672f892b5b36efed1eb5b9c74fc] system_processes
I1012 07:14:08.033203       1 resolve_container.go:150] extractPodContainerIDfromPath for id 127493, path 13:devices:/kubepods/besteffort/pod1f612ee7-054c-430a-b17b-dba1f6fbc2e5/1e600e5294ad7d6cf13dfb30b978e77fdd882672f892b5b36efed1eb5b9c74fc, containerId system_processes
I1012 07:14:08.033207       1 container_hc_collector.go:121] DEBUG: system_processes 127493 python 1329 6072916336
`

and the pod:

`
I1012 07:14:17.068441       1 metric_collector.go:141] energy from pod/container (0 active processes): name: gpu-workload/gpu-workload-container namespace: monitoring
	cgrouppid: 0 pid: [] comm:  containerid:1e600e5294ad7d6cf13dfb30b978e77fdd882672f892b5b36efed1eb5b9c74fc
	Dyn ePkg (mJ): 0 (0) (eCore: 0 (0) eDram: 0 (0) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
	Idle ePkg (mJ): 0 (0) (eCore: 0 (0) eDram: 0 (0) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
	CPUTime:  0 (0)
	NetTX IRQ: 0 (0)
	NetRX IRQ: 0 (0)
	Block IRQ: 0 (0)
	counters: map[cache_miss:0 (0) cpu_cycles:0 (0) cpu_instructions:0 (0) cpu_ref_cycles:0 (0) gpu_mem_util:0 (0) gpu_sm_util:0 (0)]
	cgroupfs: map[block_devices_used:0 (0) cgroupfs_cpu_usage_us:0 (0) cgroupfs_ioread_bytes:0 (0) cgroupfs_iowrite_bytes:0 (0) cgroupfs_kernel_memory_usage_bytes:0 (0) cgroupfs_memory_usage_bytes:0 (0) cgroupfs_system_cpu_usage_us:0 (0) cgroupfs_tcp_memory_usage_bytes:0 (0) cgroupfs_user_cpu_usage_us:0 (0)]
	kubelets: map[kubelet_cpu_usage:0 (701) kubelet_memory_bytes:0 (2361532416)]
`

Case 3: Microk8s (BM_DEPLOY) and cgroup v2

`
I1012 07:28:30.890754       1 resolve_container.go:254] extractPodContainerIDfromPath 1
I1012 07:28:30.890801       1 resolve_container.go:260] regexFindContainerIDPath2 /kubepods/besteffort/pod2b28534b-557f-40b3-a271-4d49dfaf1687/332b319919473066b04804eca446ec2667136d4b4b056f52231be39d48044c31 [/kubepods/besteffort/pod2b28534b-557f-40b3-a271-4d49dfaf1687/332b319919473066b04804eca446ec2667136d4b4b056f52231be39d48044c31] system_processes
I1012 07:28:30.890823       1 resolve_container.go:150] extractPodContainerIDfromPath for id 25694, path 0::/kubepods/besteffort/pod2b28534b-557f-40b3-a271-4d49dfaf1687/332b319919473066b04804eca446ec2667136d4b4b056f52231be39d48044c31, containerId system_processes
`

and the pod:

`
I1012 07:28:48.897585       1 metric_collector.go:141] energy from pod/container (0 active processes): name: gpu-workload/gpu-workload-container namespace: monitoring
[... cut ...]
	cgroupfs: map[block_devices_used:0 (0) cgroupfs_cpu_usage_us:0 (0) cgroupfs_ioread_bytes:0 (0) cgroupfs_iowrite_bytes:0 (0) cgroupfs_kernel_memory_usage_bytes:0 (0) cgroupfs_memory_usage_bytes:0 (0) cgroupfs_system_cpu_usage_us:0 (0) cgroupfs_tcp_memory_usage_bytes:0 (0) cgroupfs_user_cpu_usage_us:0 (0)]
	kubelets: map[kubelet_cpu_usage:0 (17) kubelet_memory_bytes:0 (2601816064)]
`

For cases 2 and 3, there seems to be an error in pkg/cgroup/resolve_container.go:extractPodContainerIDfromPath where kepler works as expected on commenting lines : https://github.com/sustainable-computing-io/kepler/blob/main/pkg/cgroup/resolve_container.go#L249 - 254

marceloamaral, 2023-10-12:
@rahul-shiv I created the PR #982 to fix this problem.

As we can see, the container ID used by Kubernetes is always in the end of the cgroup path. 
Then, the PR split the string and only extract the container id from the end of the string.

```
0::/system.slice/
docker-13ae6ef476d7a6c4fa8f55c50434258845724ea9a0a1de97b8865d6cfb31752e.scope/
kubelet.slice/kubelet-kubepods.slice/
kubelet-kubepods-burstable.slice/
kubelet-kubepods-burstable-pod8eb60a52_9a53_45cb_8b4a_f8c3acd19053.slice/
cri-containerd-0c2884d6b96d7e6fe8e9d99845e1fe61562258bf57c652942be9c17ed036e134.scope
```

```
0::/system.slice/
docker-ebabb403a3bf7e7ffe93cc48d5ba1b1e98b603bf7ab2efaa2194ed38f70dd1a6.scope/
kubelet.slice/kubelet-kubepods.slice/kubelet-kubepods-burstable.slice/
kubelet-kubepods-burstable-pod099e571e_f064_48a9_925d_c03a3b7e5e9f.slice/
cri-containerd-68d99510b2a029c2156ec2e2455121088b7f97c26404061dbb31863b45a723c5.scope
```

```
12:cpu,cpuacct:/kubepods.slice/kubepods-besteffort.slice/
kubepods-besteffort-pod8b958c57_8aa0_4360_ab06_ea17d3b680bd.slice/
crio-20d6610a9e6feb9e62f677400b10b0621c462ca6b267387e460aec974cec275d.scope
```

and your system
```
13:hugetlb:/system.slice/
docker-2fa3e04b676df750842faf017052dd37ea0cc5bc7259a3487a1718c7fe100c94.scope/
kubelet/kubepods/besteffort/
podbdd4097d-6795-404e-9bd8-6a1383386198/
c79788e0da15a6597263eb2b9c51d05dd1a9a1d08c53c1161dc8c45d2dac6b38
```

rahul-shiv, 2023-10-12:
@marceloamaral 

Thanks for the quick turnaround! The patch you provided works when I applied it to my local image which is built at https://github.com/sustainable-computing-io/kepler/commit/d87adfa659982fb994a37b539709cc07f171b1fa.

However when I tried it with the latest build pulled from quay I see very large values:

[image]

Not sure if this is a known issue. Maybe this should be a different bug?

Thanks for the fix with this issue.

marceloamaral, 2023-10-13:
@rahul-shiv this power is way too high....

The OTHER power is actually the `platform - PKG - DRAM - GPU`....
It might have something wrong with this calculation.... Or maybe something wrong with the platform power....

Can you check in Prometheus the Node and container power?
`irate(kepler_node_platform_joules_total[1m])`
`irate(kepler_node_package_joules_total[1m])`
`irate(kepler_node_dram_joules_total[1m])`
`irate(kepler_node_gpu_joules_total[1m])`

`irate(kepler_container_platform_joules_total[1m])`
`irate(kepler_container_package_joules_total[1m])`
`irate(kepler_container_dram_joules_total[1m])`
`irate(kepler_container_gpu_joules_total[1m])`

Also, can you run the command `sensors` in your BM?

closed 2023-12-19 by stale[bot]