sustainable-computing-io/kepler #943: Improving DRAM power model

rootfs, 2023-09-21:
### What would you like to be added?

As discussed in the 09/19 community meeting, modeling DRAM power with current bpf features is a challenge. More memory usage features are needed to improve the model accuracy.

Potential features are:
- process memory usage.  Kepler can read the /proc/ stats and find the memory usage. This requires additional processing and does not likely scalable very well with many processes.
- page fault. Track the page fault tracepoint and capture how many page faults a process has. This method is used in [this](https://arxiv.org/pdf/2303.05919.pdf) study that estimate the working set size. 

### Why is this needed?

Improving DRAM power model

rootfs, 2023-09-25:
I experimented with getting the `total_vm` of the process with this script
```python
import time
from bcc import BPF

b = BPF(text="""
#include <uapi/linux/ptrace.h>
#include <linux/sched.h>
#include <linux/mm.h>

BPF_HASH(total_vm_map, u32, u64);

TRACEPOINT_PROBE(sched, sched_switch) {
    u32 pid = bpf_get_current_pid_tgid();
    struct task_struct *task = (struct task_struct *)bpf_get_current_task();
    
    // Accessing task->mm->total_vm
    if (task->mm) {
        u64 total_vm = task->mm->total_vm;
        total_vm_map.update(&pid, &total_vm);
    }

    return 0;
}
""")

# Sleep for a while to collect data
try:
    time.sleep(5)
finally:
    # Iterate over BPF map and print total_vm for each PID
    for k, v in b["total_vm_map"].items():
        print(f"PID: {k.value}, Total VM: {v.value}")
```

And got this result:
```console
PID: 2647, Total VM: 713766
PID: 586890, Total VM: 46493
PID: 6195, Total VM: 188620
PID: 586865, Total VM: 46493
PID: 586769, Total VM: 46493
....
```

Since this is quick to implement, we can try it for fast prototyping and see if the total VM feature can be useful.

marceloamaral, 2023-09-27:
This makes sense. I'll work on creating a pull request (PR) for this.

By the way, I believe page faults are the only metric we can rely on. It will show memory operations that comes from I/O and CPU time will reflect the CPU cache operations.

Nonetheless, memory allocation doesn't accurately reflect memory usage, such as read and write operations. So that, we probably don't need to use this metric.

marceloamaral, 2023-09-29:
In my experiments, I found that the [task->mm->total_vm](https://docs.huihoo.com/doxygen/linux/kernel/3.7/mm__types_8h_source.html#l00342) metric is not a reliable measure for tracking memory operations. I also looked at other metrics, such as [task->mm->exec_vm](https://docs.huihoo.com/doxygen/linux/kernel/3.7/mm__types_8h_source.html#l00346) and [page cache hit](https://github.com/iovisor/bcc/blob/master/tools/cachetop.py).

During the experiments, I ran two different stress-ng workloads: `stream` and `vm-rw`. Both these workloads involves both memory reading and writing and consumes DRAM energy. But they work differently, `stream` don't frequently access the same data and thus don't trigger page cache hits.

When I run `stress-ng --c 1 --vm-bytes 20G`:
| Timestamp | Command | total_vm | exec_vm | page_cache_hit | DRAM_Power(W) |
| --- | --- | --- | --- | --- | --- |
|0| idle | | | |27.4|
|1| kepler | 1404797 | 4576 | 0 | |
|1| stress-ng | 31344 | 6556 | 964172 | 30.0 |
|2| kepler | 1404797 | 4576 | 0 | |
|2| stress-ng | 637552 | 6556 | 2045227 | 29.6 |
|3| kepler | 1404797 | 4576 | 0 | |
|3| stress-ng | 637552 | 6556 | 3108028 | 30.18 |

For the `vm-rw` workload, the `page_cache_hit` metric proved to be a useful indicator of memory usage. It increased as the workload executed, reflecting the workload's utilization of memory. \o/

When I run `stress-ng --stream 1 --metrics-brief`:
>_Note that the `stream` workload consumed more energy than the `vm-rw`..._

| Timestamp | Command | total_vm | exec_vm | page_cache_hit | DRAM_Power(W) |
| --- | --- | --- | --- | --- | --- |
|0| idle | | | |27.4|
|1| kepler | 1404797 | 4576 | 0 | |
|1| stress-ng | 637552 | 6556 | 0 | 32.18 |
|2| kepler | 1404797 | 4576 | 0 | |
|2| stress-ng | 637552 | 6556 | 0 | 32.18 |

As I previously mentioned, the metrics `total_vm` and `exec_vm` prove unreliable since they remained constant throughout the workload execution, but both workloads are constantly doing memory operations. Furthermore, when the Kepler execution has practically no DRAM energy consumption despite these metrics are showing non-zero values (actually they have high values). This indicates that relying on `total_vm` and `exec_vm` metrics could lead to inaccurate power estimations, incorrectly attributing DRAM power consumption to workloads that have pages allocated by the OS but are not actively utilizing the pages/DRAM.

In fact, @sunya-ch has experience some similar problem but with CPU time. When attempting to predict DRAM power using CPU time, she saw in her results that the power model inaccurately associated DRAM power consumption to workloads utilizing CPU resources but that were actually not consuming DRAM power, since they were CPU intensive only workloads. **Consequently, I think we actually should not use CPU time as a basis for accounting for DRAM power consumption.** We should remove it from our power model because it will generate wrong results. For example show DRAM power consumption for `stress-ng cpu` or `coremarks`...

In summary, including the `page cache hit` metric can enhance the DRAM power model, but it only applies to workloads that utilize page cache operations. Other workloads, like `stream`, which don't frequently access the same data and thus don't trigger page cache hits, won't have their power consumption accurately estimated.

In reality, accurately tracking memory operations without hardware counters can be challenging because many operations bypass the operating system and are issued directly from the hardware, as we observed in the `stream` workload, where data moves from LL3 cache to main memory without OS involvement. So, maybe it will be impossible to have an accurate DRAM power model. Note that, inaccurate DRAM power estimation might also affect the `platform` power estimation. But this should be okay since the DRAM power consumption is much smaller compared to other components.

rootfs, 2023-09-29:
thank you @marceloamaral for this very comprehensive study! This looks without L3 cache counter, it is not reliable to predict memory activities under different scenarios.

marceloamaral, 2023-09-29:
In any case, I am going to create a PR counting page cache hits (read and write).

rootfs, 2023-09-29:
@marceloamaral would you like to see if TLB flush, total_vm, and paging together can get any approximation of L3 cache activity?
```console
# bpftrace -e 'tracepoint:tlb:tlb_flush { @[comm] = count(); }'
Attaching 1 probe...
^C

@[sssd]: 1
@[gmain]: 1
@[sshd]: 1
@[conmon]: 1
@[kworker/u8:5]: 2
@[sedispatch]: 2
@[kworker/0:1]: 2
@[kworker/3:1H]: 3
@[ksoftirqd/3]: 4
@[xfsaild/vda3]: 4
@[in:imjournal]: 4
@[auditd]: 5
@[kworker/1:6]: 6
@[kworker/2:9]: 6
@[kworker/3:5]: 6
@[ksoftirqd/2]: 6
@[kube-state-metr]: 7
@[ksoftirqd/0]: 8
@[kworker/2:1H]: 9
@[kworker/1:1H]: 9
@[ksoftirqd/1]: 9
@[kworker/0:1H]: 11
@[flanneld]: 13
@[rcu_sched]: 18
@[alertmanager]: 19
@[bpftrace]: 32
@[operator]: 41
@[grafana-server]: 81
@[tuned]: 144
@[prometheus-conf]: 154
@[coredns]: 213
@[kube-scheduler]: 255
@[adapter]: 346
@[crio]: 389
@[prometheus]: 672
@[kube-controller]: 1103
@[kubelet]: 1415
@[kepler]: 1552
@[etcd]: 1553
@[swapper/0]: 2172
@[swapper/3]: 2188
@[swapper/2]: 2530
@[swapper/1]: 2668
@[kube-apiserver]: 2891
```

rootfs, 2023-09-29:
TLB flush tracepoint works both on VM and BM.

rootfs, 2023-09-29:
i ran both TLB flush and cache misses, it looks to me the correlation between them is not very promising
```bash
bpftrace -e 'hardware:cache-misses:10 { @[comm, "Cache Misses"] = count(); } tracepoint:tlb:tlb_flush { 
@[comm, "TLB Flushes"] = count();} interval:s:10 { print(@); clear(@);}'
```
results are at https://pastebin.com/DrnnNpYP

marceloamaral, 2023-09-29:
Yes, the TBL flush is very promising:

`stress-ng --vm-rw 1 --vm-bytes 20G`
```
ID: 1031436, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 1484, Page Cahe Hit: 0
PID: 1031061, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 1557, Page Cahe Hit: 11409110
--------
PID: 1031436, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 1621, Page Cahe Hit: 0
PID: 1031061, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 1689, Page Cahe Hit: 12415214
--------
PID: 1031436, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 1739, Page Cahe Hit: 0
PID: 1031061, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 1811, Page Cahe Hit: 13316658
--------
PID: 1031436, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 1870, Page Cahe Hit: 0
PID: 1031061, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 1943, Page Cahe Hit: 14322762
--------
PID: 1031436, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 2000, Page Cahe Hit: 0
PID: 1031061, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 2076, Page Cahe Hit: 15319196
--------
PID: 1031436, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 2129, Page Cahe Hit: 0
PID: 1031061, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 2217, Page Cahe Hit: 16300671
```

`stress-ng --stream 1 --metrics-brief --stream-l3-size 200M`
```
PID: 3021156, Comm:kepler, Total VM: 1372184, TLB Flush: 302, Page Cahe Hit: 0
PID: 502139, Comm:stress-ng-strea, Total VM: 637552, TLB Flush: 0, Page Cahe Hit: 0
--------
PID: 3021156, Comm:kepler, Total VM: 1372184, TLB Flush: 313, Page Cahe Hit: 0
PID: 502139, Comm:stress-ng-strea, Total VM: 637552, TLB Flush: 48, Page Cahe Hit: 0
--------
PID: 3021156, Comm:kepler, Total VM: 1372184, TLB Flush: 313, Page Cahe Hit: 0
PID: 502139, Comm:stress-ng-strea, Total VM: 637552, TLB Flush: 177, Page Cahe Hit: 0
--------
PID: 3021156, Comm:kepler, Total VM: 1372184, TLB Flush: 322, Page Cahe Hit: 0
PID: 502139, Comm:stress-ng-strea, Total VM: 637552, TLB Flush: 306, Page Cahe Hit: 0
--------
PID: 3021156, Comm:kepler, Total VM: 1372184, TLB Flush: 352, Page Cahe Hit: 0
PID: 502139, Comm:stress-ng-strea, Total VM: 637552, TLB Flush: 436, Page Cahe Hit: 0
--------
PID: 3021156, Comm:kepler, Total VM: 1372184, TLB Flush: 352, Page Cahe Hit: 0
PID: 502139, Comm:stress-ng-strea, Total VM: 637552, TLB Flush: 564, Page Cahe Hit: 0
--------
PID: 3021156, Comm:kepler, Total VM: 1372184, TLB Flush: 356, Page Cahe Hit: 0
PID: 502139, Comm:stress-ng-strea, Total VM: 637552, TLB Flush: 693, Page Cahe Hit: 0
--------
PID: 3021156, Comm:kepler, Total VM: 1372184, TLB Flush: 389, Page Cahe Hit: 0
PID: 502139, Comm:stress-ng-strea, Total VM: 637552, TLB Flush: 823, Page Cahe Hit: 0
--------
PID: 3021156, Comm:kepler, Total VM: 1372184, TLB Flush: 389, Page Cahe Hit: 0
PID: 502139, Comm:stress-ng-strea, Total VM: 637552, TLB Flush: 953, Page Cahe Hit: 0
--------
PID: 3021156, Comm:kepler, Total VM: 1372184, TLB Flush: 392, Page Cahe Hit: 0
PID: 502139, Comm:stress-ng-strea, Total VM: 637552, TLB Flush: 1085, Page Cahe Hit: 0
--------
PID: 3021156, Comm:kepler, Total VM: 1372184, TLB Flush: 392, Page Cahe Hit: 0
PID: 502139, Comm:stress-ng-strea, Total VM: 637552, TLB Flush: 1215, Page Cahe Hit: 0
--------
PID: 3021156, Comm:kepler, Total VM: 1372184, TLB Flush: 392, Page Cahe Hit: 0
PID: 502139, Comm:stress-ng-strea, Total VM: 637552, TLB Flush: 1267, Page Cahe Hit: 0
--------
PID: 3021156, Comm:kepler, Total VM: 1372184, TLB Flush: 402, Page Cahe Hit: 0
PID: 502139, Comm:stress-ng-strea, Total VM: 637552, TLB Flush: 1397, Page Cahe Hit: 0
```

The TLB flush increased as the workload executed, reflecting the workload's utilization of memory.

rootfs, 2023-09-29:
That's great to hear @marceloamaral

marceloamaral, 2023-09-29:
The TBL flush also increase the CPU intensive workload, but in a much lower rate

`stress-ng --cpu 1`
```
PID: 997527, Comm:stress-ng-cpu, Total VM: 23152, TLB Flush: 3, Page Cahe Hit: 0
--------
PID: 997527, Comm:stress-ng-cpu, Total VM: 23152, TLB Flush: 6, Page Cahe Hit: 0
--------
PID: 997527, Comm:stress-ng-cpu, Total VM: 23152, TLB Flush: 8, Page Cahe Hit: 0
--------
PID: 997527, Comm:stress-ng-cpu, Total VM: 23152, TLB Flush: 11, Page Cahe Hit: 0
--------
PID: 997527, Comm:stress-ng-cpu, Total VM: 23152, TLB Flush: 15, Page Cahe Hit: 0
--------
PID: 997527, Comm:stress-ng-cpu, Total VM: 23152, TLB Flush: 21, Page Cahe Hit: 0
--------
PID: 997527, Comm:stress-ng-cpu, Total VM: 23152, TLB Flush: 24, Page Cahe Hit: 0
--------
PID: 997527, Comm:stress-ng-cpu, Total VM: 23152, TLB Flush: 25, Page Cahe Hit: 0
--------
PID: 997527, Comm:stress-ng-cpu, Total VM: 23152, TLB Flush: 27, Page Cahe Hit: 0
```

marceloamaral, 2023-09-29:
Looks very promissing

The TBL rate is:
| workload | TBL rate (1s) | DRAM_Power(W) |
| --- | --- | --- |
|stress-ng --cpu | ~3 | 27.4 |
|stress-ng --vm-rw | ~120 | 30.18 |
|stress-ng --stream | ~130 | 32.18 |

marceloamaral, 2023-09-29:
Not sure why the `stress-ng --vm-rw 1` has 2 PIDs, it is configured to have only process.
Can it be the same thing? If yes, we will double count the metrics....

marceloamaral, 2023-09-29:
CPU pinning fixed the 2 PIDs problem, but....
With CPU pinning the TLB flush is much smaller (2 per second), but the DRAM power consumption remains the same

`stress-ng --vm-rw 1 --vm-bytes 20G --taskset 35`
```
PID: 3885332, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 0, Page Cahe Hit: 0
PID: 3884955, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 52, Page Cahe Hit: 13817392
PID: 2931113, Comm:kepler, Total VM: 1372184, TLB Flush: 370, Page Cahe Hit: 1
--------
PID: 3885332, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 0, Page Cahe Hit: 0
PID: 3884955, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 68, Page Cahe Hit: 14956847
PID: 2931113, Comm:kepler, Total VM: 1372184, TLB Flush: 370, Page Cahe Hit: 1
--------
PID: 3885332, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 0, Page Cahe Hit: 0
PID: 3884955, Comm:stress-ng-vm-rw, Total VM: 31344, TLB Flush: 70, Page Cahe Hit: 16067051
PID: 2931113, Comm:kepler, Total VM: 1372184, TLB Flush: 379, Page Cahe Hit: 1
```

rootfs, 2023-09-29:
maybe the bpftrace program needs some changes too?

marceloamaral, 2023-09-30:
I improved the python code to flush the ebpf map to show the event rate.
I have also tested multiple different memory tracing events.

```python
import time
from bcc import BPF

b = BPF(text="""
#include <uapi/linux/ptrace.h>
#include <linux/sched.h>
#include <linux/mm.h>
        
typedef struct process_metrics_t {
    u64 totalvm;
    u64 tlbflush;
    u64 cachehit;
    char comm[32];
} process_metrics_t;

BPF_HASH(processes, u32, process_metrics_t);

int kprobe__finish_task_switch(struct pt_regs *ctx, struct task_struct *prev) {
    u32 pid = bpf_get_current_pid_tgid();
    struct task_struct *task = (struct task_struct *)bpf_get_current_task();
    
    struct process_metrics_t *process_metrics;
    process_metrics = processes.lookup(&pid);
    if (process_metrics)
    {
        if (task->mm) 
        {
            process_metrics->totalvm = task->mm->total_vm;
        }
        bpf_get_current_comm(&process_metrics->comm, sizeof(process_metrics->comm));
    } 
        else
    {
        struct process_metrics_t new_process = {};
        if (task->mm) 
        {
            new_process.totalvm = task->mm->total_vm;
        }
        processes.update(&pid, &new_process);
    }
        
    u32 prev_pid = prev->pid;
    process_metrics = processes.lookup(&prev_pid);
    if (process_metrics)
    {
        if (prev->mm) 
        {
            process_metrics->totalvm = prev->mm->total_vm;
        }
        bpf_probe_read(&process_metrics->comm, sizeof(process_metrics->comm), prev->comm);
    } 
        else
    {
        struct process_metrics_t new_process = {};
        if (prev->mm) 
        {
            new_process.totalvm = prev->mm->total_vm;
        }
        processes.update(&prev_pid, &new_process);
    }
        
    return 0;
}
        
int kprobe__mark_page_accessed(struct pt_regs *ctx) { // read page cache
    u32 pid = bpf_get_current_pid_tgid();
    struct process_metrics_t *process_metrics;
    process_metrics = processes.lookup(&pid);
    if (process_metrics)
    {
        process_metrics->cachehit += 1;
    } 
    return 0;
}
        
int kprobe__mark_buffer_dirty(struct pt_regs *ctx) { // write page cache
    u32 pid = bpf_get_current_pid_tgid();
    struct process_metrics_t *process_metrics;
    process_metrics = processes.lookup(&pid);
    if (process_metrics)
    {
        process_metrics->cachehit += 1;
    } 
    return 0;
}
        
//tracepoint:tlb:tlb_flush
int trace_tlb_flush(struct pt_regs *ctx)
{
    u32 pid = bpf_get_current_pid_tgid();
    struct process_metrics_t *process_metrics;
    process_metrics = processes.lookup(&pid);
    if (process_metrics)
    {
        process_metrics->tlbflush += 1;
    } 
    return 0;
}
""")

b.attach_tracepoint(tp="tlb:tlb_flush", fn_name="trace_tlb_flush")
# b.attach_tracepoint(tp="kmem:kmalloc", fn_name="trace_tlb_flush")
# b.attach_tracepoint(tp="kmem:mm_page_alloc", fn_name="trace_tlb_flush")
# b.attach_tracepoint(tp="syscalls:sys_enter_read", fn_name="trace_tlb_flush")
# b.attach_tracepoint(tp="syscalls:sys_enter_madvise", fn_name="trace_tlb_flush")
# b.attach_tracepoint(tp="syscalls:sys_enter_writev", fn_name="trace_tlb_flush")
# b.attach_tracepoint(tp="syscalls:sys_enter_pwritev", fn_name="trace_tlb_flush")
# b.attach_tracepoint(tp="syscalls:sys_enter_readv", fn_name="trace_tlb_flush")
# b.attach_tracepoint(tp="syscalls:sys_enter_mbind", fn_name="trace_tlb_flush")
# b.attach_tracepoint(tp="percpu:percpu_alloc_percpu", fn_name="trace_tlb_flush")
# b.attach_tracepoint(tp="vmscan:mm_vmscan_direct_reclaim_begin", fn_name="trace_tlb_flush")

while True:
    kadded = 0 # print only one kepler pid
    # Sleep for a while to collect data
    try:
        time.sleep(5)
    finally:
        # Iterate over BPF map and print total_vm for each PID
        for k, v in b["processes"].items():
            pid = k.value
            command = v.comm.decode('utf-8', 'replace')

            if  ("stress" in command) or ("kepler" in command and kadded == 0):
                print(f"PID: {pid}, Comm:{command}, Total VM: {v.totalvm}, TLB Flush: {v.tlbflush }, Page Cahe Hit: {v.cachehit}")
            # print the kepler command only one time (there are several PIDs)
            if  "kepler" in command and kadded == 0:
                kadded = 1

        print("--------")
        # Clear the table
        b["processes"].clear()
```

marceloamaral, 2023-09-30:
@rootfs With the newer code _(that resets the ebpf map and aggregates PIDs)_, it was easier to verify the TLB flush events for a longer period. And, **unfortunately, the TLB flush events are not consistent to show memory utilization.**
We can see in the logs below that Kepler has always higher TLB flush events than the `stream` and `vm-rw` workloads. This will make the regression wrongly learn a pattern that Kepler consumes more DRAM than the `stream` and `vm-rw` workloads, which we know that is false.

`stress-ng --stream 1`
```
Comm:kepler, Total VM: 3852, TLB Flush: 3852, Page Cahe Hit: 0.002
Comm:stress-ng-strea, Total VM: 130, TLB Flush: 130, Page Cahe Hit: 0.0
--------
Comm:kepler, Total VM: 2841, TLB Flush: 2841, Page Cahe Hit: 0.002
--------
Comm:kepler, Total VM: 3189, TLB Flush: 3189, Page Cahe Hit: 0.002
Comm:stress-ng-strea, Total VM: 128, TLB Flush: 128, Page Cahe Hit: 0.0
--------
Comm:kepler, Total VM: 2996, TLB Flush: 2996, Page Cahe Hit: 0.002
Comm:stress-ng-strea, Total VM: 0, TLB Flush: 0, Page Cahe Hit: 0.0
--------
Comm:kepler, Total VM: 1441, TLB Flush: 1441, Page Cahe Hit: 0.001
Comm:stress-ng-strea, Total VM: 140, TLB Flush: 140, Page Cahe Hit: 0.0
--------
Comm:stress-ng-strea, Total VM: 20, TLB Flush: 20, Page Cahe Hit: 0.0
Comm:kepler, Total VM: 585, TLB Flush: 585, Page Cahe Hit: 0.001
--------
Comm:kepler, Total VM: 3180, TLB Flush: 3180, Page Cahe Hit: 0.002
Comm:stress-ng-strea, Total VM: 12, TLB Flush: 12, Page Cahe Hit: 0.0
--------
Comm:kepler, Total VM: 1190, TLB Flush: 1190, Page Cahe Hit: 0.001
Comm:stress-ng-strea, Total VM: 3, TLB Flush: 3, Page Cahe Hit: 0.0
```

However, as I commented before, the `page cache hit` metric can be used for workload that use virtual memory.
`stress-ng --vm-rw 1 --vm-bytes 20G`
```
Comm:kepler, Total VM: 3240, TLB Flush: 3240, Page Cahe Hit: 0.002
Comm:stress-ng-vm-rw, Total VM: 1731, TLB Flush: 1731, Page Cahe Hit: 6103.149
--------
Comm:kepler, Total VM: 4342, TLB Flush: 4342, Page Cahe Hit: 0.002
Comm:stress-ng-vm-rw, Total VM: 1778, TLB Flush: 1778, Page Cahe Hit: 6323.232
--------
Comm:kepler, Total VM: 1412, TLB Flush: 1412, Page Cahe Hit: 0.001
Comm:stress-ng-vm-rw, Total VM: 1820, TLB Flush: 1820, Page Cahe Hit: 6469.439
--------
Comm:kepler, Total VM: 4526, TLB Flush: 4526, Page Cahe Hit: 0.002
Comm:stress-ng-vm-rw, Total VM: 1767, TLB Flush: 1767, Page Cahe Hit: 6280.659
--------
Comm:kepler, Total VM: 3782, TLB Flush: 3782, Page Cahe Hit: 0.002
Comm:stress-ng-vm-rw, Total VM: 1703, TLB Flush: 1703, Page Cahe Hit: 6087.55
--------
Comm:kepler, Total VM: 2625, TLB Flush: 2625, Page Cahe Hit: 0.001
Comm:stress-ng-vm-rw, Total VM: 1738, TLB Flush: 1738, Page Cahe Hit: 6178.568
```

In conclusion, unfortunately, tracking the memory utilization of workloads that do not make syscalls, such as the stream workload, is very challenging, if not impossible. However, `page cache hit` metric can assist us in estimating DRAM power consumption for certain types of workloads. While having a perfect model may be difficult, it is crucial that our model should be highly consistent and explainable.

rootfs, 2023-10-01:
thanks! that's a very good study!

marceloamaral, 2023-10-12:
@rootfs Huge pages access can be measured with page faults

`stress-ng --mmaphuge 1`
```bash
Comm:kepler, Page Fault: 385, Page Cahe Hit: 0.002
Comm:stress-ng-mmaph, Page Fault: 55267, Page Cahe Hit: 49.908
Comm:stress-ng, Page Fault: 611, Page Cahe Hit: 0.283
--------
Comm:kepler, Page Fault: 614, Page Cahe Hit: 0.002
Comm:stress-ng-mmaph, Page Fault: 81566, Page Cahe Hit: 71.218
--------
Comm:stress-ng-mmaph, Page Fault: 76953, Page Cahe Hit: 66.736
Comm:kepler, Page Fault: 188, Page Cahe Hit: 0.001
--------
Comm:kepler, Page Fault: 426, Page Cahe Hit: 0.002
Comm:stress-ng-mmaph, Page Fault: 60326, Page Cahe Hit: 50.24
--------
^CComm:kepler, Page Fault: 117, Page Cahe Hit: 0.001
Comm:stress-ng-mmaph, Page Fault: 22457, Page Cahe Hit: 582.364
Comm:stress-ng, Page Fault: 78, Page Cahe Hit: 2.524
```

But page fault cannot be used for virtual memory intensive application.
`stress-ng --vm-rw 1 --vm-bytes 20G`
```
Comm:kepler, Page Fault: 364, Page Cahe Hit: 0.002
Comm:stress-ng-vm-rw, Page Fault: 0, Page Cahe Hit: 7762.29
--------
Comm:kepler, Page Fault: 231, Page Cahe Hit: 0.001
Comm:stress-ng-vm-rw, Page Fault: 0, Page Cahe Hit: 7705.658
--------
Comm:kepler, Page Fault: 371, Page Cahe Hit: 0.002
Comm:stress-ng-vm-rw, Page Fault: 4, Page Cahe Hit: 7696.895
--------
Comm:kepler, Page Fault: 417, Page Cahe Hit: 0.002
Comm:stress-ng-vm-rw, Page Fault: 0, Page Cahe Hit: 7635.478
--------
Comm:stress-ng, Page Fault: 80, Page Cahe Hit: 2.553
Comm:kepler, Page Fault: 273, Page Cahe Hit: 0.001
Comm:stress-ng-vm-rw, Page Fault: 15, Page Cahe Hit: 7257.999
```

After the PR #953 is merged, we can add page fault too.

Additionally, we can also create a PR to disable those metrics for BMs. Only enabling it to collect metrics for model training.

sunya-ch, 2023-11-09:
I collected and trained the model using the newly-introduced Page Cahe Hit metric.
I do see an improvement on dram power estimation. 
- less overestimate dram power on non-dram workload
- more accurate dram power estimation on some dram-related workload (vm-rw)

**v0.6 (before)**
[image]

**v0.7 with Page Cahe Hit:**
[image]

**metrics:**
[image]

- `cache` and `stream` are still missing. However, `vm-rw` can be detected.

rootfs, 2023-11-09:
@sunya-ch thank you for the update! The DRAM plot is a bit hard to read, do you happen to share the 0.6 and 0.7 MAE?

sunya-ch, 2023-11-10:
> @sunya-ch thank you for the update! The DRAM plot is a bit hard to read, do you happen to share the 0.6 and 0.7 MAE?

Sure. 

**v0.6:**

model_name |   |   |   |   |   |   |   |   |   | dram_mae
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
KNeighborsRegressorTrainer_1 |   |   |   |   |   |   |   |   |   | 2.434813976890930
GradientBoostingRegressorTrainer_1 |   |   |   |   |   |   |   |   |   | 2.59612418872896
PolynomialRegressionTrainer_1 |   |   |   |   |   |   |   |   |   | 6.498511224454870
SVRRegressorTrainer_1 |   |   |   |   |   |   |   |   |   | 5.309787682048190
SGDRegressorTrainer_1 |   |   |   |   |   |   |   |   |   | 7.177116337667640
LinearRegressionTrainer_1 |   |   |   |   |   |   |   |   |   | 7.258152636342270

**v0.7:**

model_name |   |   |   |   |   |   |   |   |   | dram_mae
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
KNeighborsRegressorTrainer_1 |   |   |   |   |   |   |   |   |   | 1.376094801223570
GradientBoostingRegressorTrainer_1 |   |   |   |   |   |   |   |   |   | 1.6790692030277900
XgboostFitTrainer_1 |   |   |   |   |   |   |   |   |   | 1.7114801533552500
LinearRegressionTrainer_1 |   |   |   |   |   |   |   |   |   | 4.666680640008090
SVRRegressorTrainer_1 |   |   |   |   |   |   |   |   |   | 4.023135250577460
PolynomialRegressionTrainer_1 |   |   |   |   |   |   |   |   |   | 5.188883814498100
SGDRegressorTrainer_1 |   |   |   |   |   |   |   |   |   | 4.516702814602950

rootfs, 2023-11-10:
Thank you @sunya-ch! The improvement is really promising

marceloamaral, 2023-11-10:
@sunya-ch which metrics did you use for the DRAM power model?

sunya-ch, 2023-11-13:
> @sunya-ch which metrics did you use for the DRAM power model?

Now it is both BPF CPU time and Page cache hit.
https://github.com/sustainable-computing-io/kepler-model-server/blob/c02be1c627cfbbf16822f171a55e485678acfa65/src/util/train_types.py#L19.

marceloamaral, 2023-11-13:
> Note that, the page cache hit still does not reflect the DRAM power for the iomix as well.

Right, we need to add the page faults.

The problem to use the CPU time, is that other workloads that are only CPU-bound might have wrongly associated DRAM power as we saw before.

closed 2024-01-19 by stale[bot]