sustainable-computing-io/kepler #944: unexpected behavior when no power meter, dynamic != 0 when CPU time = 0

sunya-ch, 2023-09-26:
### What happened?

Here are my offline analysis results of container power reported by Kepler on VM where no power meter.

When the total usage = 0, the estimated container energy is not zero.
[image]

Plot over timestamp:
[image]

### What did you expect to happen?

From kepler design and also from the code, we expected the dynamic energy = totalE - idleE

https://github.com/sustainable-computing-io/kepler/blob/23519cec1db92c6c06fe46587383503cdaab6112/pkg/collector/metric/node_metric.go#L413-L418

If usage = 0, both totalE and idleE should be equal because it uses deterministic model predicting value of zeros.

### How can we reproduce it (as minimally and precisely as possible)?

Run kepler on VM with no resource usage.

### Anything else we need to know?

The problem is totalE is multiplied by sampling period while the idleE is overridden with the value without multiplication.

When collecting node energy, the following lines are called. 
https://github.com/sustainable-computing-io/kepler/blob/23519cec1db92c6c06fe46587383503cdaab6112/pkg/collector/node_energy_collector.go#L116-L125

When no power metric, `model.UpdateNodeComponentEnergy` and `model.UpdateNodePlatformEnergy` will be executed first. As seen for node platform case, the estimated power will be multiplied by sampling period as the return is in watt unit.
Both totalE and idleE are good here.
https://github.com/sustainable-computing-io/kepler/blob/23519cec1db92c6c06fe46587383503cdaab6112/pkg/model/node_platform_energy.go#L93-L106

However, when `updateNodeIdleEnergy` is called, the idleE will be replaced by the value without multiplication.

https://github.com/sustainable-computing-io/kepler/blob/23519cec1db92c6c06fe46587383503cdaab6112/pkg/collector/node_energy_collector.go#L93-L108

I think the simplest way to fix is to remove the overriding part. 

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

sunya-ch, 2023-09-27:
Even if the PR https://github.com/sustainable-computing-io/kepler/pull/945 fixes overridden issue, there is still some issue observed when the CPU usage delta is low (e.g., equals to zero). 

Here are results on VM cluster using BPFOnly power model. (Only dynamic energy reported)
[image]

When we zoom in the low CPU usage, Kepler reported energy consumption of package more than zero (>150 J/3s). 
[image]

When I forcefully set to return zero dynamic energy for the container that has zero CPU usage, the result seems more make sense. 
[image]

I'm thinking the issue might come from the logic to update IdlePower here.

https://github.com/sustainable-computing-io/kepler/blob/d87adfa659982fb994a37b539709cc07f171b1fa/pkg/collector/metric/node_metric.go#L364-L381.

@marceloamaral Could you explain the logic here (delta vs deltaIdle)? 
I think it is mainly for BM. Could you confirm that shall we change something for VM case?

marceloamaral, 2023-09-27:
The idea here is to determine the lowest power consumption level, which serves as the idle power baseline.

We calculate the Delta power by subtracting the current aggregated counter value from the previous one. 

The IdleDelta represents the smallest Delta value we find, and it only gets updated when the system's resource utilization decreases. This ensures consistency in our approach.

marceloamaral, 2023-09-27:
> When I forcefully set to return zero dynamic energy for the container that has zero CPU usage, the result seems more make sense.

The dynamic energy should be 0 for zero CPU usage, but not 0 for idle power. Note that, we will not export this idle power for VMs.

If the container dynamic power for zero CPU usage is higher than zero, the idle power is not been updated for the VM, right?

Did you check the idle power?

sunya-ch, 2023-09-27:
> The dynamic energy should be 0 for zero CPU usage, but not 0 for idle power. Note that, we will not export this idle power for VMs.
>
> If the container dynamic power for zero CPU usage is higher than [...]

The idle power is zero. There are only dynamic power but it is not zero when the container usage delta is zero

sunya-ch, 2023-09-27:
> The idea here is to determine the lowest power consumption level, which serves as the idle power baseline.
>
> We calculate the Delta power by subtracting the current aggregated counter value from the [...]

Got it. Thank you for your explanation. 

After seeing the code again, for VM this function should not be called. 
Do you have any idea why the dynamic energy on VM is not zero when the CPU usage is zero?
totalE and idleE should be all set to AbsPower(0). calcDynEnergy should return zero.

marceloamaral, 2023-10-04:
@sunya-ch do we still have this problem?

closed 2024-02-09 by stale[bot]