sustainable-computing-io/kepler #952: Align Kepler local estimator with updated models

rootfs, 2023-10-02:
### What happened?

The updated models use `scale` and `weight`, while current Kepler still uses `mean` and `variance`. 

### What did you expect to happen?

Kepler local estimator should use `scale` and `weight` to align with latest models.

### How can we reproduce it (as minimally and precisely as possible)?

Running kepler post 0.5.5 image doesn't emit the right energy estimate.

### Kepler image tag

<details>
latest
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2023-10-02:
Will pick fix into 0.6 release, cc @husky-parul

marceloamaral, 2023-10-02:
@rootfs the PR #952 fix this.

closed 2023-10-03 by rootfs