sustainable-computing-io/kepler #956: Double counting GPU energy for the multiple processes in a container

rinana, 2023-10-03:
### What happened?

I ran the same python script w/ torchrun command and estimated GPU energy w/ Kepler for several times by changing the number of GPUs (1,2,4,8 GPUs). 
Prometheus seems to double-count Kepler's GPU energy inside a container when multiple commands were executed although the GPU energyper node did not increase with the same number of GPUs.

GPU energy per node result from Prometheus
[image]

GPU energy per container result from Prometheus (sum by pod/container)
The pink one should be similar to the orange one and the last green one should be similar to the blue one, but they seems to consume more.
[image]

GPU energy per container result from Prometheus (sum by pod/container/command)
[image]

### What did you expect to happen?

I expected the Kepler's exported GPU energy per container result in almost the same with the same number of GPUs.

### How can we reproduce it (as minimally and precisely as possible)?

Run torchrun workload in a container or multiple commands that consumes GPU in turns inside a single container?

### Kepler image tag

<details>
quay.io/sustainable_computing_io/kepler:latest-libbpf
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rinana, 2023-10-03:
@marceloamaral

marceloamaral, 2023-10-03:
Thanks, @rinana, for investigating this issue.

As we discussed internally, the problem may be related to how we expose the container command. Since a container comprises multiple processes, Kepler randomly associates a command from one of these processes to the container. In other words, if a container has more than one process, it selects the command of the most recently executed process.

The GPU power estimation pipeline calculates the power consumption of processes that use the GPU, each with its own command. However, Kepler updates the GPU energy for the entire container, regardless of which command was associated with it. Consequently, the GPU power is intermittently exposed with different commands.

An immediate solution is to disable the `command` for containers.

In the future, I plan to address the code duplication issue we have (#663). We will retain only per-process metrics, eliminating duplicated container and VM maps and code. With this change, power consumption will be correctly attributed to the respective PID/command, and we can then consider enabling the optional exposure of the command.

Please note that removing the command from the Prometheus metric might also reduce metric cardinality, thus minimizing Kepler's overhead in Prometheus

marceloamaral, 2023-10-03:
@rootfs could you please review the PR #957?

closed 2023-12-09 by stale[bot]