sustainable-computing-io/kepler #960: Adding bpf tools to Kepler base image to debug runtime issues

rootfs, 2023-10-04:
### What would you like to be added?

When bpf programs run into issues, it is hard to debug without tools such as `bpftool perf bpftrace`. We need a debug image to have these tools to help us debug what is going on the system

### Why is this needed?

Debugging bpf issues without these tools is very challenging.

marceloamaral, 2023-10-04:
It is possible to enter in the container and install the tool?

SamYuan1990, 2023-10-04:
> It is possible to enter in the container and install the tool?

Same question as the base image is from docker.io/nvidia/cuda:12.1.1-base-ubi9 is it possible?

rootfs, 2023-10-04:
no, once in the container, the filesystem is read only, no pkg can be installed.

jichenjc, 2023-10-11:
>>>once in the container, the filesystem is read only, no pkg can be installed.

this is different to my thought, you are saying something like CoreOS ? otherwise general container can install software

jichenjc, 2023-10-20:
seems I can do install on the pod (even though there is no repo) 
or we need follow here to do a new image?
https://kubernetes.io/docs/tasks/debug/debug-application/debug-running-pod/#ephemeral-container

```
[root@kepler-exporter-kvf82 /]# yum install bpftool
Updating Subscription Management repositories.
Unable to read consumer identity

This system is not registered with an entitlement server. You can use subscription-manager to register.

cuda                                                                                                                                                          11 kB/s | 3.5 kB     00:00
cuda                                                                                                                                                         387 kB/s | 982 kB     00:02
Extra Packages for Enterprise Linux 9 - x86_64                                                                                                                12 kB/s |  30 kB     00:02
Extra Packages for Enterprise Linux 9 - x86_64                                                                                                               6.7 MB/s |  19 MB     00:02
Red Hat Universal Base Image 9 (RPMs) - BaseOS                                                                                                                11 kB/s | 3.8 kB     00:00
Red Hat Universal Base Image 9 (RPMs) - BaseOS                                                                                                               752 kB/s | 510 kB     00:00
Red Hat Universal Base Image 9 (RPMs) - AppStream                                                                                                             19 kB/s | 4.2 kB     00:00
Red Hat Universal Base Image 9 (RPMs) - AppStream                                                                                                            2.5 MB/s | 1.7 MB     00:00
Red Hat Universal Base Image 9 (RPMs) - CodeReady Builder                                                                                                     14 kB/s | 4.2 kB     00:00
Red Hat Universal Base Image 9 (RPMs) - CodeReady Builder                                                                                                    276 kB/s | 187 kB     00:00
No match for argument: bpftool
Error: Unable to find a match: bpftool
```

Saurabhkr952, 2023-11-13:
@rootfs 

### Dockerfile Update

I've made the following changes to the Dockerfile:

- Added CentOS Stream 9 repository.
- Installed `bpftool`.
- Installed `perf`.
- Installed `bpftrace`.

[image]

---

### Build Error

I encountered an error while building the `dockerfile.base` image from the repository. The issue arises during the installation of `bcc`, and here is the error log:

```
11.91 Error:
11.91  Problem: conflicting requests
11.91   - nothing provides libLLVM-15.so()(64bit) needed by bcc-0.26.0-2.el9.x86_64 from @commandline
11.91   - nothing provides libLLVM-15.so(LLVM_15)(64bit) needed by bcc-0.26.0-2.el9.x86_64 from @commandline
```

Please take a look and verify on your end. If the error persists, I'll fix it. Additionally, I plan to address the issue #1047 related to this. 

Thanks!

rootfs, 2023-11-13:
@Saurabhkr952 Yes, this is the issue mentioned in #1047. We are going to depreciate bcc images, so if you hit issues in bcc, just ignore it and focus on libbpf dockerfiles.

Saurabhkr952, 2023-11-13:
## Dockerfile Modification Summary 

### Additions:
- Enabled the CentOS Stream 9 repository.
- Installed `bpftool`, `perf`, and `bpftrace`.

### Removals:
- Removed `bcc` and `bcc-devel`.

Before proceeding, please thoroughly review the changes and kindly confirm the following:

- Are there any other modifications that need to be made?
- Are there any additional packages that should be installed or removed?

If everything looks good, I'll proceed with creating the pull request.
 Thanks

rootfs, 2023-11-13:
@Saurabhkr952 please update https://github.com/sustainable-computing-io/kepler/blob/main/build/Dockerfile.libbpf.base first . This is the libbpf image base, it doesn't have bcc rpms.

closed 2024-01-19 by stale[bot]