sustainable-computing-io/kepler #965: Help needed in CI for rpm build

SamYuan1990, 2023-10-05:
### What happened?

see details at
https://github.com/sustainable-computing-io/kepler/actions/runs/6404025217/job/17383739761#step:4:51

### What did you expect to happen?

the RPM package can be build.

### How can we reproduce it (as minimally and precisely as possible)?

daily CI

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

SamYuan1990, 2023-10-05:
hi @rootfs , could you please help check?

rootfs, 2023-10-05:
the `BuildArch` can be hard coded to `x86_64` for now (since we don't have aarch64 support yet)

PR #969 (2023-10-09): CI: update RPM build with arch info
Try to fix #965