sustainable-computing-io/kepler #971: Only "dynamic" mode power metrics exported in latest Kepler image.

jiere, 2023-10-08:
### What happened?

When running "quay.io/sustainable_computing_io/kepler:latest" on BM with RAPL supported, only "dynamic" mode power metrics exported, no "idle" mode metrics found in Prometheus.

### What did you expect to happen?

Both "dynamic" and "idle" mode metrics exported for both container level and node level, so that we could get the total power for specific container or specific node.

### How can we reproduce it (as minimally and precisely as possible)?

Deploy kepler:latest on BM, check on Prometheus side or Grafana side.
See snapshot as follows:
[image]
[image]

### Kepler image tag

<details>
latest or release-0.6
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>
Bare Metal
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

sunya-ch, 2023-10-10:
Need fix:

Sorry it comes from this PR https://github.com/sustainable-computing-io/kepler/pull/955.
The default is false.
https://github.com/marceloamaral/kepler/blob/2ab0fb0dba35fa76a30485f6e2fc7cfc403f8eb9/pkg/config/config.go#L80

In prometheus_collector, we need to add `or` logic for the system that has power meter support.

https://github.com/sustainable-computing-io/kepler/blob/a4b2077d8bda8fbff9ba3f3cd22926fa710a423c/pkg/collector/prometheus_collector.go#L518

marceloamaral, 2023-10-10:
This is strange, because the following line should do this.
https://github.com/marceloamaral/kepler/blob/2ab0fb0dba35fa76a30485f6e2fc7cfc403f8eb9/cmd/exporter/exporter.go#L165

marceloamaral, 2023-10-10:
Found the bug, the PR #955 only fixed the `IsEstimatedIdlePowerEnabled` function, but we need to also fix the `SetEnabledEstimatedIdlePower`...

marceloamaral, 2023-10-10:
I created the PR #975 to fix this.

jichenjc, 2023-10-11:
PR merged @jiere maybe you can validate then we can close this issue

jiere, 2023-10-12:
verified on kepler:latest.