sustainable-computing-io/kepler #980: add integration test compare between previous/newer version to ensure metrics consistency

jichenjc, 2023-10-11:
### What would you like to be added?

https://github.com/sustainable-computing-io/kepler/pull/974#issuecomment-1754228690

### Why is this needed?

https://github.com/sustainable-computing-io/kepler/pull/974#issuecomment-1754228690

sthaha, 2023-10-11:
Thank you for filing this. I was just about to it 🙇

sthaha, 2023-10-11:
We should define how regressions can be prevented. 
One idea that @vprashar2929 tried was to setup a Bare Metal and a Virtual Machine, deploy kepler on to both and compare that the metrics given the same workload isn't too different (delta <= 0.05) .

marceloamaral, 2023-10-12:
@rootfs Can we have BMs in CI?

jichenjc, 2023-10-12:
>>>tried was to setup a Bare Metal and a Virtual Machine

unless we have 3rd party CI it might be hard to run in github CI? @SamYuan1990

rootfs, 2023-10-12:
@marceloamaral i requested the BM setup from CNCF
https://github.com/cncf/cluster/issues/253

Some related links:
- https://www.youtube.com/watch?v=O7GZBI6WBj0
- https://keda.sh/docs/2.12/scalers/github-runner/

closed 2023-12-18 by stale[bot]