sustainable-computing-io/kepler #985: Running `make cluster-up` cleared kubeconfig

Spazzy757, 2023-10-13:
### What happened?

Firstly, would like to say thank you for this project, it is an incredible idea

When looking into running the the local environment, I ran the `make cluster-up` command. Before running this I had my Kube contexts (found at `~/.kube/config`) setup with various clusters I work with day to day, after running the make command the only context set was the `kind-kind` context. 

### What did you expect to happen?

The `kind-kind` context should be appended to my list of contexts so that I will not have to recreate all my credentials everytime I setup the local environment

### How can we reproduce it (as minimally and precisely as possible)?

You can create a kind cluster:

```bash
kind create cluster test
```

You will see the context:
```bash
$ kubectl config get-contexts
CURRENT   NAME        CLUSTER     AUTHINFO    NAMESPACE
*         kind-test   kind-test   kind-test
```

then run the `make cluster-up` and check the contexts again

```bash
$ kubectl config get-contexts
CURRENT   NAME        CLUSTER     AUTHINFO    NAMESPACE
*         kind-kind   kind-kind   kind-kind
```

### Anything else we need to know?

I know its a nitpick for the development environment and not urgent 😅 but for people unaware who want to try contribute its a little annoying

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

Spazzy757, 2023-10-13:
Maybe this is better to move to [local-dev-cluster repo](https://github.com/sustainable-computing-io/local-dev-cluster) let me know 😄

SamYuan1990, 2023-10-13:
Hi @Spazzy757 ,

May I know if you already has a k8s cluster, so... will you have to use local-dev-cluster to set up your k8s cluster?
IMO, if you already has a k8s cluster, then you can use it for kepler development if possible.

If we have to have 2 k8s cluster, one for your daily job and one for kepler...
do we have any option(existing project?) to maintain k8s cluster setting or any existing approach help us with context switch?

Spazzy757, 2023-10-13:
Hi, I work on a lot of different clusters, but I would prefer not to use any of them to develop on keplar. I think the main issue is how the make script overwrites the config

```bash
    ✅ kind cluster kind is up and running

09:32:20 🔔 INFO : Coping kind kubeconfig to /Users/spazzy757/.kube/config

 ❯ cp /Users/spazzy757/repos/spazzy/kepler/local-dev-cluster/tmp/kind/kubeconfig /Users/spazzy757/.kube/config
```

You shouldnt need to do this with kind, you can just use the `${HOME}/.kube/config` or alternative you can set the `$KUBECONFIG` variable and use a completley seperate file for the config for instance in the above instead of copying just set:

```bash
export KUBECONFIG=/Users/spazzy757/repos/spazzy/kepler/local-dev-cluster/tmp/kind/kubeconfig
```
and that will leave my local config alone

SamYuan1990, 2023-10-13:
> Hi, I work on a lot of different clusters, but I would prefer not to use any of them to develop on keplar. I think the main issue is how the make script overwrites the config
> 
> ```shell
>     ✅ kind cluster kind is up and running
> 
> 09:32:20 🔔 INFO : Coping kind kubeconfig to /Users/spazzy757/.kube/config
> 
>  ❯ cp /Users/spazzy757/repos/spazzy/kepler/local-dev-cluster/tmp/kind/kubeconfig /Users/spazzy757/.kube/config
> ```
> 
> You shouldnt need to do this with kind, you can just use the `${HOME}/.kube/config` or alternative you can set the `$KUBECONFIG` variable and use a completley seperate file for the config for instance in the above instead of copying just set:
> 
> ```shell
> export KUBECONFIG=/Users/spazzy757/repos/spazzy/kepler/local-dev-cluster/tmp/kind/kubeconfig
> ```
> 
> and that will leave my local config alone
> 
> I beleive the main issue can be found [here](https://github.com/sustainable-computing-io/local-dev-cluster/blob/dc8807260a8a3d1b11e848b83bf214a546e6582c/main.sh#L64)

well, I do agree that if necessary we need to make k8s context management to user themself to handle it.
but meanwhile we need to provide convenient to create a k8s cluster, and setting context as the local-dev-cluster is also used in our CI.
as you said, you works on different clusters. Any suggestion or tooling to help you management k8s cluster?
if you just use `KUBECONFIG ` to switch ... between cluster. it's one of the option.

hence for UX, maybe we will provide a hint and some environment to control the logic, but considering with CI usage. The default value need further test and further discussion.

if it's good for you, please open an issue/pr at local-dev-cluster.