sustainable-computing-io/kepler #1008: Bugs about VM metrics

LeiZhou-97, 2023-10-17:
### What happened?

I saw kepler support to capture vm process metrics. I tried it and met so many bugs, I don't know how it can the pass the CI successfully....

1. not mount the /var/run/libvirt, which means it is impossible for kepler to obtain vm information on the host. 
2. path join error https://github.com/sustainable-computing-io/kepler/blob/8c8dab40d82fcc952f2c00651c87470c7d868dfa/pkg/libvirt/resolve_vm.go#L35
```
I1017 02:22:25.436020       1 resolve_vm.go:41] fullPath: /proc/%s/task/proc/%s/task
```
3. not exclude libvirt `driver.pid`
https://github.com/sustainable-computing-io/kepler/blob/8c8dab40d82fcc952f2c00651c87470c7d868dfa/pkg/libvirt/resolve_vm.go#L70

.....

I can help to fix these bugs, but I found this part of the code will be refactored in the near feature. 

@aharivel @marceloamaral Any update https://github.com/sustainable-computing-io/kepler/pull/931?

### What did you expect to happen?

None

### How can we reproduce it (as minimally and precisely as possible)?

None

### Kepler image tag

<details>

</details>
 

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>
 None

### Cloud provider or bare metal

<details>

</details>
 None

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

aharivel, 2023-10-17:
Hi @LeiZhou-97 ,
Did you try it by installing Kepler via the RPM ?
If not you may encounter problem with path that are absolute to the Host compute. 

Is driver.pid process of Libvirt not consuming any power ? if so, I guess it has to be reported. 

The refactoring code will happen as soon as possible but is not yet started.

LeiZhou-97, 2023-10-17:
Hi @aharivel,
Thanks for the response.

1. No, I deploy kepler by manifests. I tried to mount /var/run/libvirt and /proc manually, it still has a bug. This seems to be caused by a problem with your code logic.

```go
        if procPath != "" {
                // extraPath=/proc/%s/task; procPath=/proc/%s/task
		fullPath = filepath.Join(extraPath, procPath)
	} else {
		fullPath = procPath
	}
```

2. I agree with you that the power of driver process should be reported, but I just mean `driver.pid` should not be included in the vm metrics.

I can fix these bugs or fix them yourself, but I don't know if the code will be refactored soon, so these fixes will be meaningless.

aharivel, 2023-10-17:
If you have an easy / quick fix for the problem you describe in 1. then you can submitted a bug fix. Otherwise I'll try to fix it next week. 

About 2. I imagined that if a user want to know how the consumption of everything related to VM is, it has to include the VM-1+ VM-2+ ... + driver. Then to do so it is easier to already have the driver process in the VM metrics. It is just a question of filtering the name. Maybe the naming is bad then ? "driver" should be "libivrt" ? 

The refactorisation will happen not before next month and maybe more end of next month. If there are any easy/quick fix that prevent people not to fall into your bug during that time I think that worth doing it.

marceloamaral, 2023-10-17:
@LeiZhou-97, I am currently in the process of refactoring the code. However, as pointed out by @aharivel, it might be good to consider implementing some quick fixes.

In relation to the VM libivrt process, the refactoring will aggregate all processes within the VM cgroup, including not only the ibvirt but also any other processes associated with the VM itself.

aharivel, 2023-10-17:
@marceloamaral looking forward to your re-factorization !
I planned to do so next month after my current work on QEMU but I will just wait on your patch and see.  
I might propose a patch to add a vCPU field in the VM metrics for better visibility of the workload threads.
Some fields like Telco uses CPU pinning / isolation for specific heavy task and it could be interesting to be able to filter it easily in Grafana to monitor the power consumption.

closed 2023-10-24 by LeiZhou-97