sustainable-computing-io/kepler #1021: Explore 'CPUID' alternative

rootfs, 2023-10-18:
### What would you like to be added?

To reduce the kepler dependency, we check how to minimize other binaries used by kepler. One of those is `cpuid`. On most platforms `lscpu` is available as default. We can try to use `lscpu` if `cpuid` is missing.

### Why is this needed?

Provide a fall back plan if `cpuid` is missing

rootfs, 2023-10-18:
cc @vimalk78

jiere, 2023-10-29:
PR raised already, lack of non-Intel platform, especially AMD/ARM platform to verify.
I would be appreciated to accept any help for "lscpu" output info on different AMD/ARM architecture platforms.

vimalk78, 2023-10-29:
currently we use `cpuid -1 | grep uarch` to identify cpu vendor and _exact cpu micro architecture_ . 
```
 $ cpuid -1 | grep uarch
   (uarch synth) = Intel Willow Cove {optim of Sunny Cove}, 10nm++
```

@rootfs  `cpuid` uses [`CPUID` instruction](https://www.felixcloutier.com/x86/cpuid) to obtain this information. Using `CPUID` instruction is a reliable source of this information.

whereas `lscpu` or `cat /proc/cpuinfo` can only give 
```
Vendor ID:               GenuineIntel
  Model name:            11th Gen Intel(R) Core(TM) i7-11850H @ 2.50GHz
```
It does not give micro architecture of cpu.

So it depends on what we need in our power model?, if we need exact micro architecture, we need to find alternative which can get this information.

vimalk78, 2023-10-29:
we should be able to map processor model number from `lscpu` and obtain its micro architecture from intel ark site

also, there is some information available in wikichip
list of micro-architectures https://en.wikichip.org/wiki/Category:all_microarchitectures including intel architectures. 
e.g. this page https://en.wikichip.org/wiki/intel/core_i7 has links for all intel processors, and each processor's page has info about its micro-architecture. but not sure how reliable is this info.

jiere, 2023-10-30:
> So it depends on what we need in our power model?, if we need exact micro architecture, we need to find alternative which can get this information.

IIUC, we use the micro architecture as "index" for power modeling, however, not all the vendors have such microarchitecture definition list, and even for Intel, there is product codename and microarchitecture mapping, sometimes it's tricky to understand the logic of synthesize in CPUID output, see example 1:
```
$ cpuid --version
cpuid version 20230505

$ lscpu |grep Model
Model name:                      Intel(R) Xeon(R) Platinum 8487C
Model:                           143

$ cpuid -1 |grep uarch
   (uarch synth) = Intel Sapphire Rapids {Golden Cove}, Intel 7

$ cpuid -1 |grep synth
      (family synth)  = 0x6 (6)
      (model synth)   = 0x8f (143)
      (simple synth)  = Intel Xeon (unknown type) (Sapphire Rapids E5/B3) {Golden Cove}, Intel 7
      (size synth)                       = 49152 (48 KB)
      (size synth)                       = 32768 (32 KB)
      (size synth)                       = 2097152 (2 MB)
      (size synth)                       = 110100480 (105 MB)
      (QoS monitoring counter size synth)         = 32
      (vuln to branch type confusion synth)    = true
   (multi-processing synth) = multi-core (c=56), hyper-threaded (t=2)
   (APIC widths synth): CORE_width=6 SMT_width=1
   (APIC synth): PKG_ID=1 CORE_ID=39 SMT_ID=0
   (uarch synth) = Intel Sapphire Rapids {Golden Cove}, Intel 7
   (synth) = Intel Scalable (4th Gen) Bronze/Silver/Gold/Platinum (Sapphire Rapids E5/B3) {Golden Cove}, Intel 7
```
AFAIK, "Sapphire Rapids" is product codename, "Golden Cove" is microarchitecture, so whether use the former or latter to identify the power modeling index?

See example 2:
```
$ cpuid --version
cpuid version 20211210

$ lscpu |grep Model
Model name:                      11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
Model:                           140

$ cpuid -1 |grep uarch
   (uarch synth) = Intel Willow Cove {Sunny Cove}, 10nm++

$ cpuid -1 |grep synth
      (family synth)  = 0x6 (6)
      (model synth)   = 0x8c (140)
      (simple synth)  = Intel (unknown type) (Tiger Lake-U B0) [Willow Cove] {Sunny Cove}, 10nm++
      (size synth)                       = 49152 (48 KB)
      (size synth)                       = 32768 (32 KB)
      (size synth)                       = 1310720 (1.2 MB)
      (size synth)                       = 12582912 (12 MB)
   (multi-processing synth) = multi-core (c=4), hyper-threaded (t=2)
   (APIC widths synth): CORE_width=4 SMT_width=1
   (APIC synth): PKG_ID=0 CORE_ID=3 SMT_ID=1
   (uarch synth) = Intel Willow Cove {Sunny Cove}, 10nm++
   (synth) = Intel Core i*-11000 / i*-1100G* (Tiger Lake-U B0) [Willow Cove] {Sunny Cove}, 10nm++
```
In example 2, it seems that the synthesize logic in "(synth)" is that using "{}" to identify the predecessor microarchitecture, using "[]" to identify the current microarchitecture, using "()" to identify product model codename.

But in example 1, the logic is different: using "{}" to identify current microarchitecture, no "[]" there...

So due to Kelper's current code, we are parsing the "(uarch synth)" section, 
For Sapphire Rapids, we get `Sapphire Rapids`(product codename) instead of `Golden Cove` ;
For Tiger Lake, we get `Willow Cove`

Not sure if such difference is due to the `cpuid` binary version difference, but this is really an issue.

Back to this ticket feature itself, we try to find CPUID alternative, it seems that the model number is the specific one to identify the CPU model(Not CPU product, since i3/i5/i7/i9/... etc, might be the same model) 

Bad thing is that I cannot find the specific model No. allocation mechanism and definition for all of the CPU models among different vendors, so in current PR, I only lists the Intel CPU models number which could be found in Linux Kernel  code and be aligned with continuously.

jiere, 2023-10-30:
> Using `CPUID` instruction is a reliable source of this information.

I 100% agree with this comment, the issue is that not all Linux Distros have built-in `cpuid` binary.
For Kepler's containerized release image, we have already bumped the latest cpuid binary in EPEL, so we are safe there;
For Kepler's binary/rpm release, we have to figure out a fall back plan as @rootfs mentioned in the feature requirement.

vimalk78, 2023-10-30:
yes availability of `cpuid` is an issue. i am doing downstream kepler build internal to Red Hat, and `cpuid` availability is an issue, and so raised it to @rootfs.

vimalk78, 2023-10-30:
looking to find some go packages or C programs which make use of `CPUID` instruction and gather cpu specific information.

one such is `github.com/intel-go/cpuid`. Info it prints is
```
VendorString:           GenuineIntel
ProcessorBrandString:   11th Gen Intel(R) Core(TM) i7-11850H @ 2.50GHz
SteppingId:     1
ProcessorType:  0
DisplayFamily:  6
DisplayModel:   141
CacheLineSize:  64
MaxLogicalCPUId:16
InitialAPICId:  3
Smallest monitor-line size in bytes:  0
Largest monitor-line size in bytes:   63
Monitor Interrupt break-event is supported:  true
MONITOR/MWAIT extensions are supported:      true
AVX state:     true
AVX-512 state: true
Interrupt thresholds in digital thermal sensor: 2
Features: SSE3 PCLMULQDQ DTES64 MONITOR DSI_CPL VMX SMX EST TM2 SSSE3 SDBG FMA CX16 XTPR PDCM PCID SSE4_1 SSE4_2 X2APIC MOVBE POPCNT TSC_DEADLINE AES XSAVE OSXSAVE AVX F16C RDRND FPU VME DE PSE TSC MSR PAE MCE CX8 APIC SEP MTRR PGE MCA CMOV PAT PSE_36 CLFSH DS ACPI MMX FXSR SSE SSE2 SS HTT TM PBE 
ExtendedFeatures: FSGSBASE IA32_TSC_ADJUST BMI1 AVX2  SMEP BMI2 ERMS INVPCID DFPUCDS PQE AVX512F AVX512DQ RDSEED ADX SMAP AVX512IFMA CLFLUSHOPT CLWB INTEL_PROCESSOR_TRACE AVX512CD SHA AVX512BW AVX512VL AVX512VBMI    AVX512VBMI2             
ExtraFeatures: LAHF_LM ABM PREFETCHW SYSCALL NX PDPE1GB RDTSCP LM 
ThermalAndPowerFeatures: TEMPERATURE_SENSOR TURBO_BOOST ARAT PLN ECMD PTM HWP HWP_NOTIF HWP_ACTIVITY_WINDOW HWP_ENERGY_PERFORMANCE HWP_PACKAGE_LEVEL_REQUEST HDC TURBO_BOOST_MAX HCFC PERFORMANCE_ENERGY_BIAS 
CacheDescriptor: {1 1  48 12 64 64 1}
CacheDescriptor: {1 2  32 8 64 64 1}
CacheDescriptor: {2 3  1280 20 64 1024 1}
CacheDescriptor: {3 3  24576 12 64 32768 1}
CacheDescriptor: {2 0 Cache info from leaf 0x80000006 for Intel 256 0 64 -1 0}
```

jiere, 2023-10-30:
> looking to find some go packages or C programs which make use of `CPUID` instruction and gather cpu specific information.
> 
> one such is `github.com/intel-go/cpuid`

Cool! It looks more friendly than "lscpu":)

However, still could not resolve concerns and issues we talked above. Actually even the `CPUID` instruction itself could not...
Maybe we need to focus on the usage of the so-called `cpu model` in Kepler, which content is enough for Kepler to differentiate platforms and power models.

vimalk78, 2023-10-30:
added a discussion post https://github.com/sustainable-computing-io/kepler/discussions/1031

marceloamaral, 2023-11-01:
While the microarchitecture can affect the CPU power consumption, because of different instructions, branch prediction, and power states. As we are discussing, getting the specific microcode version of a CPU is very hard, specially because it can vary due to BIOS or firmware updates. Static mapping will not solve this problem...

But note that CPU power consumption can also vary in the same CPU model and microarchitecture, because of hardware manufacturing errors, environment temperature, etc. Some research studies have shown up to a 30% difference in CPU power consumption among nodes with the same hardware in a datacenter.

Giving that, we don't need to make our use cases overly complicated. We can be more general, the CPU model should be enough to classify the power models for Kepler.

jiere, 2023-12-10:
close this issue since PR merged.

closed 2023-12-10 by jiere