sustainable-computing-io/kepler #1022: Support running the Kepler in a non-Kube environment

skeeey, 2023-10-19:
### What would you like to be added?

We want to run the Kepler in a non-Kube platform (e.g. RH Device Edge) to help us collecting the power metrics

### Why is this needed?

Now Kepler can expose the power level metrics in Kubernetes efficiently,  we want to leverage the Kepler ability collecting the power metrics in a non-Kube platform  (e.g. RH Device Edge).

husky-parul, 2023-10-19:
You can use the kepler rpm https://github.com/sustainable-computing-io/kepler/releases/download/v0.6.1/kepler.rpm.tar.gz to run Kepler on non Kubernetes envirnoment.

skeeey, 2023-10-19:
@husky-parul thanks, I will have a try...

skeeey, 2023-10-23:
I tried with Kepler with 0.6.1, it can work on non Kubernetes environment, thanks teams for the help

closed 2023-10-23 by skeeey