sustainable-computing-io/kepler #1033: Difference in metric value produced by Kepler when deployed using bcc and libbpf

vprashar2929, 2023-10-31:
### What happened?

When Kepler with `release-0.6.1` and `release-0.6.1-libbpf` is deployed on the OCP cluster(VM) there is a difference in the value the exporter produces. Kepler with `release-0.6.1` produces a value significantly on the higher side as compared to `release-0.6.1-libbpf`
Below attached is the screenshot for reference:
[image]

Here Kepler with `release-0.6.1` is deployed in `openshift-kepler-operator` namespace and `release-0.6.1-libbpf` in `openshift-kepler-operator-x` namespace

### What did you expect to happen?

No difference in metric value in both releases.

### How can we reproduce it (as minimally and precisely as possible)?

Deploy Kepler with `release-0.6.1` and `release-0.6.1-libbpf` on the same cluster(VM) preferably in a different namespace.

### Kepler image tag

<details>
```
release-0.6.1, release-0.6.1-libbpf
```
</details>

### Kubernetes version

<details>
```
OCP 4.13
```

</details>

### Cloud provider or bare metal

<details>
VM
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2023-11-01:
@vprashar2929 is the ratio between bcc and libbpf 1.031? I wonder if this is because libbpf has some missing ebpf readings as @marceloamaral discovered a while back.

vprashar2929, 2023-11-01:
Yes @rootfs the ratio between bcc and libbpf is 1.031

sthaha, 2023-11-01:
@marceloamaral @rootfs , We have seen the ratio differ by a factor of 2 or 3 .. I.E 2 times less reported by libbpf than bcc based image.

rootfs, 2023-11-02:
Another potential difference is that, in bcc attacher, kepler attaches to kprobe
```go
        ftswitch, err := m.LoadKprobe("kprobe__finish_task_switch")
	if err != nil {
		return nil, fmt.Errorf("failed to load kprobe__finish_task_switch: %s", err)
	}
	err = m.AttachKprobe("finish_task_switch", ftswitch, -1)
	if err != nil {
		klog.Infof("attaching kprobe to finish_task_switch failed, trying finish_task_switch.isra.0 instead")
		err = m.AttachKprobe("finish_task_switch.isra.0", ftswitch, -1)
		if err != nil {
			return nil, fmt.Errorf("failed to attach finish_task_switch: %s", err)
		}
	}
```

While in libbpf, kepler attaches to tracepoint:
```go
        // attach sched_switch tracepoint to kepler_trace function
	prog, err := libbpfModule.GetProgram("kepler_trace")
	if err != nil {
		return libbpfModule, fmt.Errorf("failed to get kepler_trace: %v", err)
	} else {
		_, err = prog.AttachTracepoint("sched", "sched_switch")
		if err != nil {
			return libbpfModule, fmt.Errorf("failed to attach sched/sched_switch: %v", err)
		}
	}
```

If the tracepoint and kprobe are not called at the same rate, then there could be a mismatch.

sthaha, 2023-11-02:
Does this explain why the variation is inconsistent ? I.E. only certain container metrics seems to be off while others are spot on.

rootfs, 2023-11-02:
libbpf and bcc have different retry strategies when reading map from kernel, this may cause some inconsistency
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/bpfassets/attacher/libbpf_attacher.go#L234-L261

closed 2024-01-08 by stale[bot]