sustainable-computing-io/kepler #1043: not able to start kepler standalone

jichenjc, 2023-11-10:
### What happened?

```
Nov 08 08:58:45 jitest51.novalocal kepler[36723]: I1108 08:58:45.988636   36723 bpf_perf.go:132] failed to attach bpf with libbpf: failed to load module: stat /bpfassets/libbpf/bpf.o/amd64_kepler.bpf.o: no such file or directory
Nov 08 08:58:45 jitest51.novalocal kepler[36723]: I1108 08:58:45.988660   36723 exporter.go:240] failed to start : failed to attach bpf assets: no bcc build tag
Nov 08 08:58:45 jitest51.novalocal kepler[36723]: I1108 08:58:45.988778   36723 exporter.go:267] Started Kepler in 1.363049ms

[root@jitest51 ~]# rpm -qa | grep kepler
kepler-0.6.1-0.6.1.x86_64

```

### What did you expect to happen?

kepler can start 

### How can we reproduce it (as minimally and precisely as possible)?

try kepler standalone 

I can do this to 
```
  144  mkdir -p /bpfassets/libbpf/bpf.o
  145  cp ./bpfassets/libbpf/bpf.o/amd64_kepler.bpf.o /bpfassets/libbpf/bpf.o

```

to fix this problem 

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

aharivel, 2023-11-29:
Hi, 
I think it would be better to install the eBPF binary here:
"/var/lib/kepler/bpfassets/amd64_kepler.bpf.o"
(via a RPM script? The eBPF is not part of the RPM at the moment..)
The code explicitly check the above path before checking on the local path.

Anthony

rootfs, 2023-11-29:
this is fixed in latest rpm build

closed 2023-11-29 by rootfs