sustainable-computing-io/kepler #1056: Unknown node feature in log made confusion.

sunya-ch, 2023-11-16:
### What happened?

When increase verbosity, we will see the following error log:

```
I1116 01:51:43.289034   11020 model.go:177] Model Config NODE_TOTAL: {ModelType:LinearRegressor ModelOutputType:AbsPower TrainerName: EnergySource:acpi SelectFilter: InitModelURL: InitModelFilepath: IsNodePowerModel:true ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataFeatureValues:[]}
I1116 01:51:43.289323   11020 lr.go:185] LR Model (AbsPower): loadWeightFromURLorLocal(): &map[platform:{{220.9079278650894 map[] map[bpf_cpu_time_us:{5911.969193263386 29.028228361462897}]}}] (error: <nil>)
I1116 01:51:43.289374   11020 model.go:120] Using Power Model AbsPower
I1116 01:51:43.289387   11020 node_platform_energy.go:53] Using the LinearRegressor/AbsPower Power Model to estimate Node Platform Power
I1116 01:51:43.289413   11020 model.go:177] Model Config NODE_COMPONENTS: {ModelType:LinearRegressor ModelOutputType:AbsPower TrainerName: EnergySource:rapl SelectFilter: InitModelURL: InitModelFilepath: IsNodePowerModel:true ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataFeatureValues:[]}
I1116 01:51:43.289583   11020 lr.go:185] LR Model (AbsPower): loadWeightFromURLorLocal(): &map[core:{{0 map[] map[bpf_cpu_time_us:{5911.969193263386 0}]}} dram:{{47.142633336743344 map[] map[bpf_cpu_time_us:{5911.969193263386 3.57348245077466}]}} package:{{69.91739430907396 map[] map[bpf_cpu_time_us:{5911.969193263386 22.16772409328642}]}} uncore:{{0 map[] map[bpf_cpu_time_us:{5911.969193263386 0}]}}] (error: <nil>)
I1116 01:51:43.289649   11020 model.go:120] Using Power Model AbsPower
I1116 01:51:43.289662   11020 node_component_energy.go:54] Using the LinearRegressor/AbsPower Power Model to estimate Node Component Power
I1116 01:51:43.289843   11020 node_metric.go:274] Unknown node feature: bpf_cpu_time_us, adding 0 value
I1116 01:51:43.289898   11020 node_metric.go:274] Unknown node feature: bpf_page_cache_hit, adding 0 value
I1116 01:51:43.289912   11020 node_metric.go:274] Unknown node feature: bpf_net_tx_irq, adding 0 value
I1116 01:51:43.289924   11020 node_metric.go:274] Unknown node feature: bpf_net_rx_irq, adding 0 value
...
```

same here in https://github.com/sustainable-computing-io/kepler/issues/860.

Log is from node_metric.go:

https://github.com/sustainable-computing-io/kepler/blob/5cb98b5dd3e7d1f448a5eee3775b4becc3c56c65/pkg/collector/metric/node_metric.go#L217

Actually, the node feature is not unknown but just does not have a value set by https://github.com/sustainable-computing-io/kepler/blob/5cb98b5dd3e7d1f448a5eee3775b4becc3c56c65/pkg/collector/metric/node_metric.go#L184.

### What did you expect to happen?

We should separate the case between unknown and unset to avoid confusion.  

### How can we reproduce it (as minimally and precisely as possible)?

Run with log verbosity 5

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

jichenjc, 2023-11-17:
yes, I am seeing similar error in my env as well

```
I1117 03:11:18.770403   52508 node_component_energy.go:54] Using the LinearRegressor/AbsPower Power Model to estimate Node Component Power
I1117 03:11:18.770511   52508 node_metric.go:274] Unknown node feature: bpf_cpu_time_us, adding 0 value
I1117 03:11:18.770544   52508 node_metric.go:274] Unknown node feature: bpf_net_tx_irq, adding 0 value
I1117 03:11:18.770553   52508 node_metric.go:274] Unknown node feature: bpf_net_rx_irq, adding 0 value
I1117 03:11:18.770567   52508 node_metric.go:274] Unknown node feature: bpf_block_irq, adding 0 value
I1117 03:11:18.770579   52508 node_metric.go:274] Unknown node feature: cpu_cycles, adding 0 value
I1117 03:11:18.770587   52508 node_metric.go:274] Unknown node feature: cpu_ref_cycles, adding 0 value
I1117 03:11:18.770595   52508 node_metric.go:274] Unknown node feature: cpu_instructions, adding 0 value
I1117 03:11:18.770603   52508 node_metric.go:274] Unknown node feature: cache_miss, adding 0 value
I1117 03:11:18.770610   52508 node_metric.go:274] Unknown node feature: cgroupfs_memory_usage_bytes, adding 0 value
I1117 03:11:18.770618   52508 node_metric.go:274] Unknown node feature: cgroupfs_kernel_memory_usage_bytes, adding 0 value
I1117 03:11:18.770626   52508 node_metric.go:274] Unknown node feature: cgroupfs_tcp_memory_usage_bytes, adding 0 value
I1117 03:11:18.770633   52508 node_metric.go:274] Unknown node feature: cgroupfs_cpu_usage_us, adding 0 value
I1117 03:11:18.770641   52508 node_metric.go:274] Unknown node feature: cgroupfs_system_cpu_usage_us, adding 0 value
I1117 03:11:18.770648   52508 node_metric.go:274] Unknown node feature: cgroupfs_user_cpu_usage_us, adding 0 value
I1117 03:11:18.770655   52508 node_metric.go:274] Unknown node feature: cgroupfs_ioread_bytes, adding 0 value
I1117 03:11:18.770662   52508 node_metric.go:274] Unknown node feature: cgroupfs_iowrite_bytes, adding 0 value
I1117 03:11:18.770670   52508 node_metric.go:274] Unknown node feature: block_devices_used, adding 0 value
I1117 03:11:18.770683   52508 node_metric.go:274] Unknown node feature: block_devices_used, adding 0 value
...
```

jichenjc, 2023-11-17:
um... seems all feature are with `"_DYN"` or `_IDLE` suffix
but my log shows those are something like block_devices_used , so either we didn't set suffix correctly 
or there are some logic error such cgroupfs_, bpf_ etc features are wrongly input?

marceloamaral, 2023-11-17:
The problem here is that before adding the container metrics, the node metric maps are empty. This will be fixed in the code refactoring that I am preparing...

The "_DYN" or _IDLE suffix are only for energy metrics.

sunya-ch, 2023-11-22:
Current version made confusion.
@jichenjc  It is not wrongly input. All of those  usage metrics is supposed to be return by this following loop. The less metrics are node metric including the energy metric as @marceloamaral explained.

https://github.com/sustainable-computing-io/kepler/blob/7ec92ba534e7d1da816a55deb05085eb46321537/pkg/collector/metric/node_metric.go#L219-L222

As I mentioned above, the current code required the function `AddNodeResUsageFromContainerResUsage` to set ne.ResourceUsage. At the first call of `ToEstimatorValues`, `AddNodeResUsageFromContainerResUsage` has not called yet. However, after that, you can see that no such a log coming up again.

closed 2024-01-28 by stale[bot]