sustainable-computing-io/kepler #1072: CI must validate that go.mod and `vendor` directories are up to date

sthaha, 2023-11-20:
### What happened?

Currently changes to vendor directory (which is used in build) isn't validated against `go mod vendor` to ensure that accidental modifications are prevented. 
We must validate that both `go mod tidy` and `go mod vendor` are executed before a PR is merged. 

### What did you expect to happen?

Expect CI to fail if `mod` or `vendor` is out of date. 

### How can we reproduce it (as minimally and precisely as possible)?

Submit modifications to `vendor` directory. 

### Kepler image tag

<details>

</details>

### Kubernetes version

NA

### Cloud provider or bare metal

NA

### OS version

NA

### Install tools

NA

### Kepler deployment config

NA

### Container runtime (CRI) and version (if applicable)

NA

### Related plugins (CNI, CSI, ...) and versions (if applicable)

NA

sthaha, 2023-11-20:
cc: @SamYuan1990

SamYuan1990, 2023-11-21:
I will create a PR and let's see.

SamYuan1990, 2023-11-21:
@sthaha , I just created a PR ref logic at https://github.com/sustainable-computing-io/kepler-action/blob/main/.github/workflows/check-dist.yml#L37

SamYuan1990, 2023-11-22:
close for now as #1073 merged.

closed 2023-11-22 by SamYuan1990