sustainable-computing-io/kepler #1074: Incorrect measurements on ARM (Ampere Altra Max)

simonarys, 2023-11-21:
### What happened?

We are interested in running Kepler on an ARM Ampere Altra Max machine (BM). We managed to successfully build the Kepler image from the Dockerfiles available in the build/ folder on the main GitHub branch (hash: 88c82f384f10ba4deb39675b2c88450bc28ee7b8). We then ran the image on a Kubernetes cluster, both on a x86 and the ARM machine. However on the ARM one, we've observed an anomaly in the Grafana dashboard, which is indicating unexpectedly low energy consumption metrics and the "system" namespace is showing unrealistically high power consumptions (more than 1 million W). Moreover, the DRAM energy measurements are always 0. See pictures below.

[image]
[image]

We would appreciate any insights or guidance on potential ARM-specific optimizations or configurations that might be necessary to ensure accurate energy consumption measurements.

To aid in troubleshooting, we attached logs and configuration details. Please let us know if further information is needed.

### What did you expect to happen?

We expected similar results to those obtained when running Kepler on a x86 Intel machine, since we followed the same steps on both architecture to build and deploy Kepler. On the x86 Intel machine we obtained plausible results, not so far from our PDU's power outlet metrics.

### How can we reproduce it (as minimally and precisely as possible)?

We had to change a few lines in the Dockerfiles to use the ARM architecture instead of x86 because only the Dockerfile.bcc.base has an ARM version available in the GitHub repo.

We built the following images using the Dockerfiles from the /build folder in this order: 
1) bcc.base 
2) bcc.builder
3) kernel-source-images
4) bcc.kepler
5) manifest

For bcc.base, we built the dockerfile with an arm64 extension that is already in the GitHub repository.

For bcc.builder, we replaced the FROM to use the bcc.base image we just built and replaced the `amd64` by `arm64` in the line 10:
```sh
RUN curl -LO https://go.dev/dl/go1.18.10.linux-arm64.tar.gz; mkdir -p /usr/local; tar -C /usr/local -xvzf go1.18.10.linux-arm64.tar.gz; rm -f go1.18.10.linux-arm64.tar.gz
```

For kernel-source-image, we replaced the whole file by this and do not use the `build-kernel-source-images.sh` script:
```sh
FROM registry.access.redhat.com/ubi8/ubi

ARG ARCH=aarch64

RUN yum install -y http://mirror.centos.org/centos/8-stream/BaseOS/aarch64/os/Packages/centos-gpg-keys-8-6.el8.noarch.rpm && \
    yum install -y http://mirror.centos.org/centos/8-stream/BaseOS/aarch64/os/Packages/centos-stream-repos-8-6.el8.noarch.rpm

RUN yum install -y kernel-devel
```

For bcc.kepler, we changed the FROMs of line 1 and 25 to use our previously built images (builder then base) and moved the file to the root of the repository before building it using docker.

For the manifest, firstly, we built the manifest using:
```sh
make build-manifest OPTS="CI_DEPLOY PROMETHEUS_DEPLOY"
```

Then, we replaced the image source at line 152 in `_output/generated_manifest/deployment.yaml` with our kepler image built in the previous step and uploaded on DockerHub.

Lastly, we deployed the manifest to our empty Kubernetes (Kind) cluster.

### Anything else we need to know?

We are using a kind cluster

Kepler pod logs : 
<details>

```
I1116 09:52:00.944197    4588 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I1116 09:52:01.039707    4588 qat.go:35] Failed to init qat-telemtry err: could not get qat status exit status 127
I1116 09:52:01.055616    4588 exporter.go:157] Kepler running on version: bb2b1bb-dirty
I1116 09:52:01.055722    4588 config.go:274] using gCgroup ID in the BPF program: true
I1116 09:52:01.055743    4588 config.go:276] kernel version: 5.15
I1116 09:52:01.055779    4588 exporter.go:169] LibbpfBuilt: false, BccBuilt: true
[... 117 lines cut ...]
</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

</details>

### Container runtime (CRI) and version (if applicable)

<details>
containerd://1.7.1
</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2023-12-05:
@simonarys btw, if you build libbpf image for arm64, the latest Kepler build and base images from @vimalk78  are based on ubi9, they support multiarch. It will make arm image much easier.

vimalk78, 2023-12-06:
> We also tried building it from your image: `quay.io/sustainable_computing_io/kepler_builder:ubi-9-libbpf-1.2.0` but we got the exact **same error**. _Do note that Go wasn’t installed on this image and we had to install it._

```
$ podman run -it --rm  quay.io/sustainable_computing_io/kepler_builder:ubi-9-libbpf-1.2.0 sh
sh-5.1# go version
go version go1.20.10 linux/amd64
```
i can see golang in builder image

vimalk78, 2023-12-06:
I have been able to build `aarch` image for kepler, but that is without `CPUID`. though i have not tested it.

simonarys, 2023-12-06:
Indeed, you're right. Go is installed and the error is the following:
>go: cannot find GOROOT directory: /usr/local/go

Thus re-installing Go into the /usr/local/go folder fixed the error for us, sorry for the confusion.

Since Go is already installed, we now had to replace the GOROOT path from `usr/local/go` to `/lib/golang` on line 11:
`ENV GOPATH=/opt/app-root GO111MODULE=off GOROOT=/lib/golang`
The path is now found when building the Dockerfile. However we are facing a new issue:
```sh
41.67 github.com/sustainable-computing-io/kepler/pkg/manager
42.22 command-line-arguments
44.56 # command-line-arguments
44.56 /lib/golang/pkg/tool/linux_amd64/link: running clang failed: exit status 1
44.56 clang-16: error: no such file or directory: '/usr/lib/x86_64-linux-gnu/libbpf.a'
44.56 clang-16: error: no such file or directory: '/usr/lib/x86_64-linux-gnu/libbpf.a'
44.56 clang-16: error: no such file or directory: '/usr/lib/x86_64-linux-gnu/libbpf.a'
44.56 clang-16: error: no such file or directory: '/usr/lib/x86_64-linux-gnu/libbpf.a'
44.56 clang-16: error: no such file or directory: '/usr/lib/x86_64-linux-gnu/libbpf.a'
44.56 clang-16: error: no such file or directory: '/usr/lib/x86_64-linux-gnu/libbpf.a'
44.56 
44.77 make: *** [Makefile:191: _build_local] Error 1
```

vimalk78, 2023-12-06:
`GOROOT` is already defined in the image
```
sh-5.1# go env | grep ROOT
GOROOT="/usr/lib/golang"
```

SamYuan1990, 2024-02-12:
@vimalk78 , is this issue been fixed?