sustainable-computing-io/kepler #1086: is kepler able to build with a shared library file like .so file instead of a static link as .a ?

SamYuan1990, 2023-11-27:
https://github.com/sustainable-computing-io/kepler/blob/1ae5c47250aad8ecb2ea65ba6197a2325f6660d3/Makefile#L52

it seems that ubuntu version as https://packages.ubuntu.com/source/jammy/libbpf don't provide the status link file as .a and we have to install libbpf from source as https://github.com/sustainable-computing-io/kepler/blob/main/build/Dockerfile.libbpf.base#L23
hence, do we have any chance to install libbpf from distribution package instead of source file?

SamYuan1990, 2023-11-27:
I make some test and it looks strange https://github.com/sustainable-computing-io/kepler/pull/1085#issue-2012209401

SamYuan1990, 2023-11-27:
may related to https://github.com/aquasecurity/libbpfgo/issues/392, @rootfs , @sunya-ch , @marceloamaral if libbpf api is not stable, should we use libbpf for now? as we have to use a latest libbpf c package as our golang code depends on it. but which will not have deb/rpm release.
which means, we need to take care everything for the libbpf for any CVE?

rootfs, 2023-11-27:
here is the libraries needed by kepler libbpf build
```
# ldd _output/bin/linux_amd64/kepler 
        linux-vdso.so.1 (0x00007ffc25ba3000)
        libelf.so.1 => /usr/lib64/libelf.so.1 (0x00007f554d152000)
        libz.so.1 => /usr/lib64/libz.so.1 (0x00007f554d138000)
        libc.so.6 => /usr/lib64/libc.so.6 (0x00007f554cf2f000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f554d173000)
```

SamYuan1990, 2023-11-27:
> here is the libraries needed by kepler libbpf build
>
> ```
> # ldd _output/bin/linux_amd64/kepler
>         linux-vdso.so.1 (0x00007ffc25ba3000)
>         libelf.so.1 => /usr/lib64/libelf.so.1 [...]

@rootfs , if we have to build those files each time when base image build. any idea to trace the CVEs consider with SBOM point of view?

sthaha, 2023-12-01:
> since go binaries are statically linked, we have to use libbpf.a and not a shared library

While I agree using static `.a` serves us better, this isn't true of `go`.  Since we (kepler) uses `cgo` we can always use `CGO_LDFLAGS` to link against dynamic libraries like we do for `-lelf` try: 

```
❯ ldd _output/bin/kepler
        linux-vdso.so.1 (0x00007fffe1c97000)
        libelf.so.1 => /lib64/libelf.so.1 (0x00007feb938ab000)
        libz.so.1 => /lib64/libz.so.1 (0x00007feb93891000)
        libc.so.6 => /lib64/libc.so.6 (0x00007feb936b3000)
        libzstd.so.1 => /lib64/libzstd.so.1 (0x00007feb935f7000)
        /lib64/ld-linux-x86-64.so.2 (0x00007feb938e7000)
```

But I to the original post, I agree with @vimalk78  to use static linking instead of dynamic.

SamYuan1990, 2023-12-02:
@rootfs , can we assume that we do version control by https://github.com/sustainable-computing-io/kepler/blob/main/build/Dockerfile.libbpf.base#L20 and https://github.com/sustainable-computing-io/kepler/blob/main/build/Dockerfile.libbpf.base#L26 for libbpf and xgboost?

closed 2024-01-06 by SamYuan1990