sustainable-computing-io/kepler #1100: contradicting Grafana curves

jochen-schuettler, 2023-12-04:
### What happened?

The consumption curve in kWh shows several large steps between 12 and 1 p.m. The consumption curve in W shows peaks only at on time roughly 12:30.
[image]

### What did you expect to happen?

Either there is one kWh step at the same time as the peak in W, or there must be several peaks in W at the times when there are steps in kWh.

### How can we reproduce it (as minimally and precisely as possible)?

I have no idea.

### Anything else we need to know?

I'll fill in the necessary info below tomorrow.

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

jochen-schuettler, 2023-12-05:
Due to illness, no additional info, but another strange curve:
[image]
It looks as if the consumption is decreasing in the end. How is that possible?

rootfs, 2023-12-05:
not sure about the first issue, but for the 2nd (i.e. decreasing curve), since it is on a 24 hour period basis (i.e. not accumulative) the drop makes sense.

jochen-schuettler, 2023-12-07:
I deleted the 2nd non-issue. Do you have any ideas how to gain more info on the repeating problem?

jochen-schuettler, 2023-12-11:
I updated the Details now

jochen-schuettler, 2023-12-21:
An idea: If the kWh-curve is smoothed by an averaging filter, that could explain the behaviour. Is that the case?

marceloamaral, 2024-01-15:
Yes, KWh extrapolates the average values over a longer interval.

closed 2024-01-16 by jochen-schuettler