sustainable-computing-io/kepler #1130: rpm build failed

rootfs, 2023-12-17:
### What happened?

Per nightly build result, the rpm build flow is broken
https://github.com/sustainable-computing-io/kepler/actions/runs/7237585836/job/19717413710
```
go: cannot find GOROOT directory: /usr/local/go
go: cannot find GOROOT directory: /usr/local/go
rpmbuild packaging/rpm/kepler.spec --build-in-place -bb
make: rpmbuild: No such file or directory
```

### What did you expect to happen?

rpm build should work

### How can we reproduce it (as minimally and precisely as possible)?

seen in nightly build

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2023-12-17:
@vimalk78 PTAL

SamYuan1990, 2023-12-25:
I suppose it got fixed https://github.com/sustainable-computing-io/kepler/actions/runs/7320712888/job/19940182496

vimalk78, 2023-12-25:
@SamYuan1990 yes. @rootfs made a release also.

SamYuan1990, 2023-12-25:
close is it fixed. please reopen if needed.

closed 2023-12-25 by SamYuan1990