sustainable-computing-io/kepler #1134: Hard coded value of source in kepler_node metrics

vprashar2929, 2023-12-19:
### What happened?

When Kepler deployed using Operator on VM for metrics like `kepler_node_core_joules_total` it shows source as `rapl` even if `rapl` is not available. 

Attached screenshot for reference:

[image]

Found source as `rapl` for below queries:
`kepler_node_core_joules_total`
`kepler_node_dram_joules_total`
`kepler_node_package_joules_total`
`kepler_node_uncore_joules_total`

### What did you expect to happen?

The source should not say `rapl` when it is not available i.e Kepler running on VM

### How can we reproduce it (as minimally and precisely as possible)?

Deploy Kepler using Operator on OCP cluster running on VM

### Kepler image tag

<details>
release-0.7.1
</details>

### Kubernetes version

<details>

```console
OCP 4.14

```

</details>

### Cloud provider or bare metal

<details>
VM
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

marceloamaral, 2024-01-15:
@vprashar2929 right, we should change it to "estimated"

marceloamaral, 2024-02-15:
@vprashar2929 I created the PR #1235 to fix this

vprashar2929, 2024-02-26:
@marceloamaral I have verified with the latest image and now it shows source as `trained_power_model` in the case of VM

When Kepler deployed on VM
[image]

When Kepler deployed on BM
[image]