sustainable-computing-io/kepler #1137: cannot attach kprobe, probe entry may not exist

bellalzohir, 2023-12-19:
### What happened?

`` am experiencing an issue where no power data is being reported to Prometheus after installing Kepler using a Helm chart. The problem seems to stem from an inability to attach a kprobe. Below are the logs from the Kepler daemon set.

`# kubectl logs -n kepler daemonset/kepler`
```console
I1219 20:14:56.761312       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I1219 20:14:56.792841       1 qat.go:35] Failed to init qat-telemtry err: could not get qat status exit status 127
I1219 20:14:56.803947       1 exporter.go:157] Kepler running on version: c1cae95
I1219 20:14:56.803958       1 config.go:274] using gCgroup ID in the BPF program: true
I1219 20:14:56.803988       1 config.go:276] kernel version: 5.15
I1219 20:14:56.804058       1 config.go:301] The Idle power will be exposed. Are you running on Baremetal or using single VM per node?
I1219 20:14:56.804066       1 exporter.go:169] LibbpfBuilt: false, BccBuilt: true
I1219 20:14:56.804073       1 exporter.go:188] EnabledBPFBatchDelete: true
I1219 20:14:56.804115       1 power.go:54] use sysfs to obtain power
I1219 20:14:56.804129       1 redfish.go:169] failed to get redfish credential file path
I1219 20:14:56.808629       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I1219 20:14:56.942519       1 exporter.go:203] Initializing the GPU collector
I1219 20:15:02.947615       1 watcher.go:66] Using in cluster k8s config
cannot attach kprobe, probe entry may not exist
I1219 20:15:04.378168       1 bcc_attacher.go:94] attaching kprobe to finish_task_switch failed, trying finish_task_switch.isra.0 instead
I1219 20:15:04.469510       1 bcc_attacher.go:129] Successfully load eBPF module from using bcc
I1219 20:15:04.469534       1 bcc_attacher.go:187] Successfully load eBPF module from bcc with option: [-DMAP_SIZE=10240 -DNUM_CPUS=56 -DSAMPLE_RATE=0 -DSET_GROUP_ID]
I1219 20:15:04.528407       1 container_energy.go:114] Using the Ratio/DynPower Power Model to estimate Container Platform Power
I1219 20:15:04.528421       1 container_energy.go:115] Container feature names: [cpu_instructions]
I1219 20:15:04.528438       1 container_energy.go:124] Using the Ratio/DynPower Power Model to estimate Container Component Power
I1219 20:15:04.528448       1 container_energy.go:125] Container feature names: [cpu_instructions cpu_instructions cache_miss   gpu_sm_util]
I1219 20:15:04.528461       1 process_power.go:113] Using the Ratio/DynPower Power Model to estimate Process Platform Power
I1219 20:15:04.528468       1 process_power.go:114] Container feature names: [cpu_instructions]
I1219 20:15:04.528483       1 process_power.go:123] Using the Ratio/DynPower Power Model to estimate Process Component Power
I1219 20:15:04.528493       1 process_power.go:124] Container feature names: [cpu_instructions cpu_instructions cache_miss   gpu_sm_util]
I1219 20:15:04.528690       1 node_platform_energy.go:53] Using the LinearRegressor/AbsPower Power Model to estimate Node Platform Power
I1219 20:15:04.530947       1 exporter.go:267] Started Kepler in 7.726990945s
```

### What did you expect to happen?

Kepler should successfully report power data to Prometheus.

### How can we reproduce it (as minimally and precisely as possible)?

Install Kepler via helm and see the log of Kepler deamonSet 

### Kepler image tag

<details>
the default one in helm
</details>

### Kubernetes version

<details>

```console
$ kubectl version
Client Version: v1.28.2
Kustomize Version: v5.0.4-0.20230601165947-6ce0bf390ce3
Server Version: v1.28.3
```

</details>

### Cloud provider or bare metal

<details>
bare metal
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
$ uname -a
5.15.0-88-generic 
x86_64 GNU/Linux

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2023-12-19:
@bellalzohir can you try the latest kepler image? It is now based on libbpf. The failure is caused by bcc.

bellalzohir, 2023-12-20:
I did clone the main repo 

```
make build-manifest OPTS="CI_DEPLOY PROMETHEUS_DEPLOY"  
kubectl apply -f _output/generated-manifest/deployment.yaml 
```
the pod can not start 
```
kepler# kubectl logs -n kepler daemonset/kepler-exporter 
Error from server (BadRequest): container "kepler-exporter" in pod "kepler-exporter-wrtl7" is waiting to start: ContainerCreating

```

closed 2023-12-20 by bellalzohir