sustainable-computing-io/kepler #1140: Ensure libbpf version is pinned down to 1.2.0 in unit_test

sthaha, 2023-12-20:
https://github.com/sustainable-computing-io/kepler/blob/365d56fa81c963e9d43dece236b600bad0c00948/.github/workflows/unit_test.yml#L41-L51

Currently unit-test workflow reinstalls (not sure why) libbpf from source and uses latest version of libbpf everytime instead of the one used to build kepler in container builds. This may give us different results depending on the version of libbpf. 
Lets pin down the version of libbpf used like all other dependencies to get reliable and reproducible results everywhere.

rootfs, 2023-12-20:
yes, the libbpf used by kepler image is 1.2.0

vimalk78, 2023-12-21:
may be we should change it to use `kepler-action@v0.0.4` 

https://github.com/sustainable-computing-io/kepler/blob/365d56fa81c963e9d43dece236b600bad0c00948/.github/workflows/unit_test.yml#L35-L38

closed 2024-02-26 by stale[bot]