sustainable-computing-io/kepler #1142: Kepler metrics reporting higher values

vprashar2929, 2023-12-20:
### What happened?

When Kepler deployed using `latest` image on OCP 4.14(VM) it reports higher metric values. 

Attached is a screenshot of a comparison between metric values reported by two Kepler instances on the same OCP cluster

[image]

Here kepler with image `release-0.7.1` is deployed inside `kepler-old` ns and kepler with `latest` is deployed inside `kepler-new`

### What did you expect to happen?

There should be no difference between the metric value produced 

### How can we reproduce it (as minimally and precisely as possible)?

Deploy Kepler with `release-0.7.1` and `latest` on a VM cluster and compare the metric values produced by each.

### Kepler image tag

<details>
latest, release-0.7.1
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>
VM
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2023-12-21:
@marceloamaral can you take a look? I think the delta between 0.7.1 and latest is the new build process and the refactoring

rootfs, 2023-12-21:
@vprashar2929 can you add 0.7.2 in the same cluster too? thanks

vprashar2929, 2023-12-22:
Sure @rootfs Here is the screenshot of when 3 Kepler instances are deployed on OCP 4.14(VM)
[image]

Kepler with `release-0.7.2` is deployed in `kepler-new` ns Kepler with `release-0.7.1` is deployed in `kepler-old` ns and Kepler with `latest` is deployed in `kepler-new` ns

sunya-ch, 2024-02-14:
The local weight has no update. Only the metric name and energy source name changed (cpu_time_ms, intel_rapl).

vprashar2929, 2024-02-26:
@marceloamaral @sunya-ch @rootfs I could still see higher numbers reported by Kepler when `latest` release is deployed on the VM. 
Attached below is the comparison when Kepler with `latest`, `release-0.7.3`, and `release-0.7.2` is deployed on the VM

[image]

vprashar2929, 2024-04-17:
Still could see a higher number reported by Kepler. Comparing `latest` `release-0.7.8` and `release-0.7.2` on VM

[image]

closed 2026-04-10 by sunya-ch