sustainable-computing-io/kepler #1143: kepler_node_other_joules_total metric doesn't get initialized

vprashar2929, 2023-12-20:
### What happened?

When kepler is deployed using `latest` image on the VM it doesn't initialize `kepler_node_other_joules_total` metric. We currently check for this metric whether it is present or not in our e2e tests. 

Also checked on OCP cluster(VM) and got same result

[image]

Here Kepler using `latest` is deployed inside `kepler-new` ns and kepler with `release-0.7.1` deployed inside `kepler-old`ns

In `release-0.7.1` the metric gets initialized 

### What did you expect to happen?

Did we remove the initialization intentionally? If not then expect the metric to be at least initialized. 

### How can we reproduce it (as minimally and precisely as possible)?

Deploy Kepler with `latest` on any cluster (Kind/OpenShift) and check for `kepler_node_other_joules_total` metric

### Kepler image tag

<details>
latest,release-0.7.1
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>
VM
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2023-12-21:
@vprashar2929 i can see the metrics from my own build using the latest code
```
# curl localhost:8888/metrics |grep kepler_node_other_joules_total
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  202k    0  202k    0     0  49.3M      0 --:--:-- --:--:-- --:--:-- 49.3M
# HELP kepler_node_other_joules_total Aggregated value in other value from intel_rapl
# TYPE kepler_node_other_joules_total counter
kepler_node_other_joules_total{instance="rhel",mode="dynamic",package="socket0",source="intel_rapl"} 0.003
kepler_node_other_joules_total{instance="rhel",mode="idle",package="socket0",source="intel_rapl"} 0.669
```

vprashar2929, 2023-12-22:
Strange🤔 I explicitly removed `kepler_node_other_joules_total` check from my e2e [PR](https://github.com/sustainable-computing-io/kepler/pull/1141/files#diff-abd345108aceec20dce0e401c1ececbc4308646e02c3fe4d3bb0e3ff0abe5f1b) as with that CI was [failing](https://github.com/sustainable-computing-io/kepler/actions/runs/7274350458/job/19820195305?pr=1141) after seeing Marcelo's refactor [PR](https://github.com/sustainable-computing-io/kepler/pull/1089/files#diff-abd345108aceec20dce0e401c1ececbc4308646e02c3fe4d3bb0e3ff0abe5f1b). He removed checks for `kepler_node_other_joules_total` and `kepler_container_other_joules_total`

Is the removal of these metrics part of refactoring?

jichenjc, 2024-01-04:
I used latest kepler container image and confim I can't see this as well in my output
@marceloamaral maybe you can share the reason of this removal whether it's `duplicate` to something? 
```
kepler_node_other_joules_total
```

@vprashar2929 at lesat it's not doc here and that might lead to the reason of removal
https://sustainable-computing.io/design/metrics/

closed 2024-03-11 by stale[bot]

thisiskazem, 2024-03-18:
what do "other" metrics exactly refer to in term of technical world?