sustainable-computing-io/kepler #1145: kepler_container_gpu_joules_total metric initalized even when no GPU is available

vprashar2929, 2023-12-21:
### What happened?

When Kepler deployed using `latest` on OCP 4.14 (VM) cluster with no GPU available Kepler still produces `kepler_container_gpu_joules_total` metric. In earlier releases i.e `release-0.7.2` when GPU was unavailable, Kepler didn't initializes the metric.

Attached is a screenshot for reference:
[image]

### What did you expect to happen?

If GPU is not available then Kepler shouldn't initialize the metric

### How can we reproduce it (as minimally and precisely as possible)?

Deploy Kepler with `latest` image on VM and query for `kepler_container_gpu_joules_total` metric

### Kepler image tag

<details>
latest
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>
VM
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

jichenjc, 2024-01-04:
@vprashar2929 will this be helpful for the discussion? seemes the recommendation is to set the ENABLE_GPU=false

https://github.com/sustainable-computing-io/kepler/pull/930

vprashar2929, 2024-01-04:
if we set ENABLE_GPU to false, users who have GPU available won't be able to see the metrics. With `ENABLE_GPU=true` on `release-0.7.2` or older and no GPU available the metrics weren't initializing which is expected. Now with the latest kepler image, the metrics get initialized to 0. I am not sure whether it is expected behavior in post-refactor PR merge or its a miss

closed 2024-03-11 by stale[bot]