sustainable-computing-io/kepler #1152: Cannot get VM power through Kepler process metrics.

jiere, 2024-01-02:
### What happened?

Using Kepler binary to expose metrics, cannot find VM related process metrics.

### What did you expect to happen?

Kepler binary which runs on BM(with process metrics enabled) could expose power metrics of VMs which run on BM. 

### How can we reproduce it (as minimally and precisely as possible)?

1. Use Kepler 0.7.2 tag code base to build kepler binary(run `make build`)
2. Use `qemu` to bring up VMs on BM.
3. Run kepler binary on BM and check process metrics exposed on port 8888.

### Anything else we need to know?

See log snippets as follows:
$ ps -ef |grep qemu |grep -v grep

root       15442       1 13 1222 ?      23:47:17 /usr/bin/qemu-system-x86_64 -name ....
root      312148       1  1 1227 ?      00:38:06 /usr/bin/qemu-system-x86_64 -name ....
root      325968       1  0 1227 ?      00:25:31 /usr/bin/qemu-system-x86_64 -name ....

$ curl http://localhost:8888/metrics |grep kepler_process
....
// Did not find any process metrics belong to pid 15442/312/148/325968.

### Kepler image tag

<details>
0.7.2
</details>

### Kubernetes version

<details>
Not using Kubernetes
</details>

### Cloud provider or bare metal

<details>
BM
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

jichenjc, 2024-01-04:
wondering 15125 is the libvirtd itself .. maybe you can check the output？

https://github.com/sustainable-computing-io/kepler/pull/1028 try to avoid libvirtd but found the code is refactored

```

    * libvirt: find vm id from cgroup path
```

and that's right after 0.7.2 
```
commit 158305ad1e1f21ec0550d395bfb8ca1b4a219a97 (tag: v0.7.2, origin/release-0.7.2)
Author: sustainable-computing-bot <bot@sustainable-computing.io>
Date:   Tue Dec 19 15:18:01 2023 +0000
```

jiere, 2024-01-05:
Thanks @jichenjc for your info! I will have a try.

jiere, 2024-01-07:
Folks, with latest kepler code, when I disabled process metrics(see #1153 ), the kepler stand-alone mode could recognize VM metrics correctly. See detailed logs below:
```
$ virsh list --all
 Id   Name                           State
-----------------------------------------------
 1    td-ccnp-runner                 running
 8    cc-trusted-api-github-runner   running
 11   td-chr-dev                     running
 14   dev-hjh-tdvm                   running
...

$ curl -s http://localhost:8888/metrics |grep kepler_vm
# HELP kepler_vm_bpf_block_irq_total Aggregated value in bpf_block_irq value from bpf
# TYPE kepler_vm_bpf_block_irq_total counter
kepler_vm_bpf_block_irq_total{source="bpf",vm_id="machine-qemu-1-td-ccnp-runner"} 0
kepler_vm_bpf_block_irq_total{source="bpf",vm_id="machine-qemu-11-td-chr-dev"} 0
kepler_vm_bpf_block_irq_total{source="bpf",vm_id="machine-qemu-14-dev-hjh-tdvm"} 0
kepler_vm_bpf_block_irq_total{source="bpf",vm_id="machine-qemu-8-cc-trusted-api-github-runner"} 0
# HELP kepler_vm_bpf_cpu_time_us_total Aggregated value in bpf_cpu_time_us value from bpf
# TYPE kepler_vm_bpf_cpu_time_us_total counter
kepler_vm_bpf_cpu_time_us_total{source="bpf",vm_id="machine-qemu-1-td-ccnp-runner"} 3538
kepler_vm_bpf_cpu_time_us_total{source="bpf",vm_id="machine-qemu-11-td-chr-dev"} 10
kepler_vm_bpf_cpu_time_us_total{source="bpf",vm_id="machine-qemu-14-dev-hjh-tdvm"} 55405
kepler_vm_bpf_cpu_time_us_total{source="bpf",vm_id="machine-qemu-8-cc-trusted-api-github-runner"} 145
# HELP kepler_vm_bpf_net_rx_irq_total Aggregated value in bpf_net_rx_irq value from bpf
# TYPE kepler_vm_bpf_net_rx_irq_total counter
kepler_vm_bpf_net_rx_irq_total{source="bpf",vm_id="machine-qemu-1-td-ccnp-runner"} 51
kepler_vm_bpf_net_rx_irq_total{source="bpf",vm_id="machine-qemu-11-td-chr-dev"} 20
kepler_vm_bpf_net_rx_irq_total{source="bpf",vm_id="machine-qemu-14-dev-hjh-tdvm"} 209
kepler_vm_bpf_net_rx_irq_total{source="bpf",vm_id="machine-qemu-8-cc-trusted-api-github-runner"} 12
# HELP kepler_vm_bpf_net_tx_irq_total Aggregated value in bpf_net_tx_irq value from bpf
# TYPE kepler_vm_bpf_net_tx_irq_total counter
kepler_vm_bpf_net_tx_irq_total{source="bpf",vm_id="machine-qemu-1-td-ccnp-runner"} 0
kepler_vm_bpf_net_tx_irq_total{source="bpf",vm_id="machine-qemu-11-td-chr-dev"} 0
kepler_vm_bpf_net_tx_irq_total{source="bpf",vm_id="machine-qemu-14-dev-hjh-tdvm"} 0
kepler_vm_bpf_net_tx_irq_total{source="bpf",vm_id="machine-qemu-8-cc-trusted-api-github-runner"} 0
# HELP kepler_vm_bpf_page_cache_hit_total Aggregated value in bpf_page_cache_hit value from bpf
# TYPE kepler_vm_bpf_page_cache_hit_total counter
kepler_vm_bpf_page_cache_hit_total{source="bpf",vm_id="machine-qemu-1-td-ccnp-runner"} 0
kepler_vm_bpf_page_cache_hit_total{source="bpf",vm_id="machine-qemu-11-td-chr-dev"} 0
kepler_vm_bpf_page_cache_hit_total{source="bpf",vm_id="machine-qemu-14-dev-hjh-tdvm"} 0
kepler_vm_bpf_page_cache_hit_total{source="bpf",vm_id="machine-qemu-8-cc-trusted-api-github-runner"} 0
# HELP kepler_vm_cache_miss_total Aggregated value in cache_miss value from bpf
# TYPE kepler_vm_cache_miss_total counter
kepler_vm_cache_miss_total{source="bpf",vm_id="machine-qemu-1-td-ccnp-runner"} 3.5855099e+07
kepler_vm_cache_miss_total{source="bpf",vm_id="machine-qemu-11-td-chr-dev"} 1.3004933e+07
kepler_vm_cache_miss_total{source="bpf",vm_id="machine-qemu-14-dev-hjh-tdvm"} 1.65464706e+08
kepler_vm_cache_miss_total{source="bpf",vm_id="machine-qemu-8-cc-trusted-api-github-runner"} 9.023841e+06
# HELP kepler_vm_cgroupfs_cpu_usage_us_total Aggregated value in cgroupfs_cpu_usage_us value from cgroup
# TYPE kepler_vm_cgroupfs_cpu_usage_us_total counter
kepler_vm_cgroupfs_cpu_usage_us_total{source="cgroup",vm_id="machine-qemu-1-td-ccnp-runner"} 0
kepler_vm_cgroupfs_cpu_usage_us_total{source="cgroup",vm_id="machine-qemu-11-td-chr-dev"} 0
kepler_vm_cgroupfs_cpu_usage_us_total{source="cgroup",vm_id="machine-qemu-14-dev-hjh-tdvm"} 0
kepler_vm_cgroupfs_cpu_usage_us_total{source="cgroup",vm_id="machine-qemu-8-cc-trusted-api-github-runner"} 0
# HELP kepler_vm_cgroupfs_memory_usage_bytes_total Aggregated value in cgroupfs_memory_usage_bytes value from cgroup
# TYPE kepler_vm_cgroupfs_memory_usage_bytes_total counter
kepler_vm_cgroupfs_memory_usage_bytes_total{source="cgroup",vm_id="machine-qemu-1-td-ccnp-runner"} 0
kepler_vm_cgroupfs_memory_usage_bytes_total{source="cgroup",vm_id="machine-qemu-11-td-chr-dev"} 0
kepler_vm_cgroupfs_memory_usage_bytes_total{source="cgroup",vm_id="machine-qemu-14-dev-hjh-tdvm"} 0
kepler_vm_cgroupfs_memory_usage_bytes_total{source="cgroup",vm_id="machine-qemu-8-cc-trusted-api-github-runner"} 0
# HELP kepler_vm_cgroupfs_system_cpu_usage_us_total Aggregated value in cgroupfs_system_cpu_usage_us value from cgroup
# TYPE kepler_vm_cgroupfs_system_cpu_usage_us_total counter
kepler_vm_cgroupfs_system_cpu_usage_us_total{source="cgroup",vm_id="machine-qemu-1-td-ccnp-runner"} 0
kepler_vm_cgroupfs_system_cpu_usage_us_total{source="cgroup",vm_id="machine-qemu-11-td-chr-dev"} 0
kepler_vm_cgroupfs_system_cpu_usage_us_total{source="cgroup",vm_id="machine-qemu-14-dev-hjh-tdvm"} 0
kepler_vm_cgroupfs_system_cpu_usage_us_total{source="cgroup",vm_id="machine-qemu-8-cc-trusted-api-github-runner"} 0
# HELP kepler_vm_cgroupfs_user_cpu_usage_us_total Aggregated value in cgroupfs_user_cpu_usage_us value from cgroup
# TYPE kepler_vm_cgroupfs_user_cpu_usage_us_total counter
kepler_vm_cgroupfs_user_cpu_usage_us_total{source="cgroup",vm_id="machine-qemu-1-td-ccnp-runner"} 0
[... cut ...]
kepler_vm_uncore_joules_total{mode="dynamic",source="intel_rapl",vm_id="machine-qemu-8-cc-trusted-api-github-runner"} 0
kepler_vm_uncore_joules_total{mode="idle",source="intel_rapl",vm_id="machine-qemu-1-td-ccnp-runner"} 0
kepler_vm_uncore_joules_total{mode="idle",source="intel_rapl",vm_id="machine-qemu-11-td-chr-dev"} 0
kepler_vm_uncore_joules_total{mode="idle",source="intel_rapl",vm_id="machine-qemu-14-dev-hjh-tdvm"} 0
kepler_vm_uncore_joules_total{mode="idle",source="intel_rapl",vm_id="machine-qemu-8-cc-trusted-api-github-runner"} 0
```
Since #1028 fix was closed yet, I will also close current issue.

closed 2024-01-07 by jiere

jichenjc, 2024-01-08:
>>>with latest kepler code, when I disabled process metrics(see https://github.com/sustainable-computing-io/kepler/issues/1153 )

ok, disable process metrics can have VM metrics seems weird 
is there any Doc about it ?

jiere, 2024-01-08:
> > > > with latest kepler code, when I disabled process metrics(see #1153 )
> 
> ok, disable process metrics can have VM metrics seems weird is there any Doc about it ?

I suppose it is an issue, this is why I kept #1153 open. Since 0.7.2 no such issue on my side.

marceloamaral, 2024-01-15:
> disable process metrics can have VM metrics seems weird

No, the new code after refactoring should not have this problem.
Did you test the code from main?

jichenjc, 2024-01-16:
@marceloamaral #1153 has more detail about the reason and solution ,it's a typo related to refactory and it's fixed already