sustainable-computing-io/kepler #1153: Kepler binary built by latest code runs into panic on BM

jiere, 2024-01-02:
### What happened?

Using Kepler latest code base to build kepler binary and run it with process metrics enabled.
Kepler cannot work normally and runs into panic and exits. 
Kepler 0.7.2 release no such issue.

### What did you expect to happen?

Kepler binary could run normally to check process metrics on BM.

### How can we reproduce it (as minimally and precisely as possible)?

1. Build kepler binary using `make build`
2. Run kepler binary like this: `./kepler -v=5 &`

### Anything else we need to know?

See logs:
# /home/cpio/kepler/bin/kepler-latest -v=5 &
[1] 740471
root@css-emrac-prc10:~# I0102 15:45:00.688114  740471 gpu.go:46] Failed to init nvml, err: failed to init nvml. ERROR_LIBRARY_NOT_FOUND
E0102 15:45:00.692177  740471 utils.go:173] grep cpuid command output failure: exit status 1
I0102 15:45:00.724765  740471 qat.go:35] Failed to init qat-telemtry err: could not get qat status exit status 127
I0102 15:45:00.751977  740471 exporter.go:155] Kepler running on version: b0dcde3-dirty
I0102 15:45:00.752023  740471 config.go:275] using gCgroup ID in the BPF program: true
I0102 15:45:00.752074  740471 config.go:277] kernel version: 6.2
I0102 15:45:00.752439  740471 config.go:302] The Idle power will be exposed. Are you running on Baremetal or using single VM per node?
I0102 15:45:00.752474  740471 exporter.go:167] LibbpfBuilt: true, BccBuilt: false
I0102 15:45:00.752489  740471 exporter.go:186] EnabledBPFBatchDelete: true
I0102 15:45:00.752501  740471 config.go:148] ENABLE_EBPF_CGROUPID: true
I0102 15:45:00.752510  740471 config.go:149] ENABLE_GPU: false
I0102 15:45:00.752519  740471 config.go:150] ENABLE_QAT: false
I0102 15:45:00.752528  740471 config.go:151] ENABLE_PROCESS_METRICS: true
I0102 15:45:00.752538  740471 config.go:152] EXPOSE_HW_COUNTER_METRICS: true
I0102 15:45:00.752547  740471 config.go:153] EXPOSE_CGROUP_METRICS: true
I0102 15:45:00.752556  740471 config.go:154] EXPOSE_IRQ_COUNTER_METRICS: true
I0102 15:45:00.752567  740471 config.go:155] EXPOSE_ESTIMATED_IDLE_POWER_METRICS: true. This only impacts when the power is estimated using pre-prained models. Estimated idle power is meaningful only when Kepler is running on bare-metal or with a single virtual machine (VM) on the node.
I0102 15:45:00.752587  740471 config.go:156] EXPERIMENTAL_BPF_SAMPLE_RATE: %!t(int=0)
I0102 15:45:00.752919  740471 power.go:53] use sysfs to obtain power
I0102 15:45:00.752955  740471 redfish.go:169] failed to get redfish credential file path
I0102 15:45:00.771766  740471 acpi.go:65] Using the ACPI power meter path: /sys/devices/LNXSYSTM:00/LNXSYBUS:00/ACPI000D:00/
I0102 15:45:00.771795  740471 power.go:72] using acpi to obtain power
I0102 15:45:00.771806  740471 bpf_perf.go:85] hardeware counter metrics config true
I0102 15:45:00.771818  740471 bpf_perf.go:103] irq counter metrics config true
I0102 15:45:00.771830  740471 utils.go:86] Available ebpf software counters: [bpf_cpu_time_us bpf_page_cache_hit bpf_net_tx_irq bpf_net_rx_irq bpf_block_irq]
I0102 15:45:00.771856  740471 utils.go:91] Available ebpf hardware counters: [cache_miss cpu_cycles cpu_ref_cycles cpu_instructions]
I0102 15:45:00.771868  740471 utils.go:104] Available cgroup metrics from cgroup: [cgroupfs_memory_usage_bytes cgroupfs_kernel_memory_usage_bytes cgroupfs_tcp_memory_usage_bytes cgroupfs_cpu_usage_us cgroupfs_system_cpu_usage_us cgroupfs_user_cpu_usage_us cgroupfs_ioread_bytes cgroupfs_iowrite_bytes block_devices_used]
I0102 15:45:00.772336  740471 watcher.go:66] Using in cluster k8s config
I0102 15:45:00.772349  740471 watcher.go:73] failed to get config: unable to load in-cluster configuration, KUBERNETES_SERVICE_HOST and KUBERNETES_SERVICE_PORT must be defined
I0102 15:45:00.772364  740471 watcher.go:122] k8s APIserver watcher was not enabled
I0102 15:45:00.772479  740471 prometheus_collector.go:87] Registered Process Prometheus metrics
I0102 15:45:00.772530  740471 prometheus_collector.go:92] Registered Container Prometheus metrics
I0102 15:45:00.772780  740471 exporter.go:82] exit stack:
goroutine 1 [running]:
runtime/debug.Stack()
        /usr/local/go/src/runtime/debug/stack.go:24 +0x65
main.finalizing()
        /home/jie/dev/kepler/cmd/exporter/exporter.go:81 +0x1d
panic({0x1601b20, 0xc0009468c0})
        /usr/local/go/src/runtime/panic.go:884 +0x212
github.com/prometheus/client_golang/prometheus.(*Registry).MustRegister(...)
        /home/jie/dev/kepler/vendor/github.com/prometheus/client_golang/prometheus/registry.go:405
github.com/sustainable-computing-io/kepler/pkg/metrics.(*PrometheusExporter).RegisterMetrics(0xc0005677c0)
        /home/jie/dev/kepler/pkg/metrics/prometheus_collector.go:96 +0x3e5
main.main()
        /home/jie/dev/kepler/cmd/exporter/exporter.go:230 +0x5b9
I0102 15:45:00.772808  740471 exporter.go:84] Exiting...

[1]+  Exit 10                 /home/cpio/kepler/bin/kepler-latest -v=5

### Kepler image tag

<details>
latest
</details>

### Kubernetes version

<details>
No Kubernetes deployed.
</details>

### Cloud provider or bare metal

<details>
BM
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2024-01-03:
@jiere i tested the latest code and it works for me. Can you double check your workspace, e.g. build both the latest kepler and libbpf code and replace the old ones

jiere, 2024-01-05:
@rootfs , I referred to current GHA [here](https://github.com/sustainable-computing-io/kepler-action/blob/main/index.js#L62) for libbpf installation, and used v1.2.0.
Let me try latest libbpf then.

jiere, 2024-01-07:
@rootfs , I had a further try, and found new clues for this issue, it should be no related to libbpf installed versions, but related to kepler config, when enabled process metric, could reproduce this issue, while if process metric was disabled, no such issue, see detailed logs below:
```
root@jie-nuc:~# mkdir /etc/kepler/kepler.config -p
root@jie-nuc:~# echo -n "true" > /etc/kepler/kepler.config/ENABLE_PROCESS_METRICS
root@jie-nuc:~# /home/jie/dev/kepler/_output/bin/kepler -v=5 &
[1] 2678128
root@jie-nuc:~# I0107 03:14:08.888803 2678128 gpu.go:46] Failed to init nvml, err: failed to init nvml. ERROR_LIBRARY_NOT_FOUND
I0107 03:14:08.891195 2678128 qat.go:35] Failed to init qat-telemtry err: could not get qat status exit status 127
I0107 03:14:08.903322 2678128 exporter.go:155] Kepler running on version: b0dcde3-dirty
I0107 03:14:08.903339 2678128 config.go:275] using gCgroup ID in the BPF program: true
I0107 03:14:08.903368 2678128 config.go:277] kernel version: 6.2
I0107 03:14:08.903440 2678128 config.go:302] The Idle power will be exposed. Are you running on Baremetal or using single VM per node?
I0107 03:14:08.903450 2678128 exporter.go:167] LibbpfBuilt: true, BccBuilt: false
I0107 03:14:08.903455 2678128 exporter.go:186] EnabledBPFBatchDelete: true
I0107 03:14:08.903460 2678128 config.go:148] ENABLE_EBPF_CGROUPID: true
I0107 03:14:08.903463 2678128 config.go:149] ENABLE_GPU: false
I0107 03:14:08.903467 2678128 config.go:150] ENABLE_QAT: false
I0107 03:14:08.903470 2678128 config.go:151] ENABLE_PROCESS_METRICS: true
I0107 03:14:08.903474 2678128 config.go:152] EXPOSE_HW_COUNTER_METRICS: true
I0107 03:14:08.903478 2678128 config.go:153] EXPOSE_CGROUP_METRICS: true
I0107 03:14:08.903482 2678128 config.go:154] EXPOSE_IRQ_COUNTER_METRICS: true
I0107 03:14:08.903485 2678128 config.go:155] EXPOSE_ESTIMATED_IDLE_POWER_METRICS: true. This only impacts when the power is estimated using pre-prained models. Estimated idle power is meaningful only when Kepler is running on bare-metal or with a single virtual machine (VM) on the node.
I0107 03:14:08.903491 2678128 config.go:156] EXPERIMENTAL_BPF_SAMPLE_RATE: %!t(int=0)
I0107 03:14:08.903518 2678128 power.go:53] use sysfs to obtain power
I0107 03:14:08.903529 2678128 redfish.go:169] failed to get redfish credential file path
I0107 03:14:08.904141 2678128 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0107 03:14:08.904147 2678128 power.go:72] using none to obtain power
I0107 03:14:08.904151 2678128 bpf_perf.go:85] hardeware counter metrics config true
I0107 03:14:08.904155 2678128 bpf_perf.go:103] irq counter metrics config true
I0107 03:14:08.904161 2678128 utils.go:86] Available ebpf software counters: [bpf_cpu_time_us bpf_page_cache_hit bpf_net_tx_irq bpf_net_rx_irq bpf_block_irq]
I0107 03:14:08.904169 2678128 utils.go:91] Available ebpf hardware counters: [cpu_cycles cpu_ref_cycles cpu_instructions cache_miss]
I0107 03:14:08.904174 2678128 utils.go:104] Available cgroup metrics from cgroup: [cgroupfs_memory_usage_bytes cgroupfs_kernel_memory_usage_bytes cgroupfs_tcp_memory_usage_bytes cgroupfs_cpu_usage_us cgroupfs_system_cpu_usage_us cgroupfs_user_cpu_usage_us cgroupfs_ioread_bytes cgroupfs_iowrite_bytes block_devices_used]
I0107 03:14:08.904363 2678128 watcher.go:66] Using in cluster k8s config
I0107 03:14:08.904369 2678128 watcher.go:73] failed to get config: unable to load in-cluster configuration, KUBERNETES_SERVICE_HOST and KUBERNETES_SERVICE_PORT must be defined
I0107 03:14:08.904375 2678128 watcher.go:122] k8s APIserver watcher was not enabled
I0107 03:14:08.904420 2678128 prometheus_collector.go:87] Registered Process Prometheus metrics
I0107 03:14:08.904440 2678128 prometheus_collector.go:92] Registered Container Prometheus metrics
I0107 03:14:08.904520 2678128 exporter.go:82] exit stack:
goroutine 1 [running]:
runtime/debug.Stack()
        /usr/local/go/src/runtime/debug/stack.go:24 +0x65
main.finalizing()
        /home/jie/dev/kepler/cmd/exporter/exporter.go:81 +0x1d
panic({0x1601b20, 0xc0000405e0})
        /usr/local/go/src/runtime/panic.go:884 +0x212
github.com/prometheus/client_golang/prometheus.(*Registry).MustRegister(...)
        /home/jie/dev/kepler/vendor/github.com/prometheus/client_golang/prometheus/registry.go:405
github.com/sustainable-computing-io/kepler/pkg/metrics.(*PrometheusExporter).RegisterMetrics(0xc0003f9900)
        /home/jie/dev/kepler/pkg/metrics/prometheus_collector.go:96 +0x3e5
main.main()
        /home/jie/dev/kepler/cmd/exporter/exporter.go:230 +0x5b9
I0107 03:14:08.904532 2678128 exporter.go:84] Exiting...

[1]+  Exit 10                 /home/jie/dev/kepler/_output/bin/kepler -v=5
root@jie-nuc:~#
root@jie-nuc:~# echo -n "false" > /etc/kepler/kepler.config/ENABLE_PROCESS_METRICS
root@jie-nuc:~#
root@jie-nuc:~#
root@jie-nuc:~# /home/jie/dev/kepler/_output/bin/kepler -v=5 &
[1] 2678250
root@jie-nuc:~# I0107 03:15:25.696438 2678250 gpu.go:46] Failed to init nvml, err: failed to init nvml. ERROR_LIBRARY_NOT_FOUND
I0107 03:15:25.699599 2678250 qat.go:35] Failed to init qat-telemtry err: could not get qat status exit status 127
I0107 03:15:25.714660 2678250 exporter.go:155] Kepler running on version: b0dcde3-dirty
I0107 03:15:25.714677 2678250 config.go:275] using gCgroup ID in the BPF program: true
I0107 03:15:25.714707 2678250 config.go:277] kernel version: 6.2
I0107 03:15:25.714757 2678250 config.go:302] The Idle power will be exposed. Are you running on Baremetal or using single VM per node?
[... cut ...]
I0107 03:15:25.715706 2678250 prometheus_collector.go:97] Registered VM Prometheus metrics
I0107 03:15:25.715716 2678250 prometheus_collector.go:101] Registered Node Prometheus metrics
I0107 03:15:25.715777 2678250 bpf_perf.go:135] failed to attach bpf with libbpf: failed to load module: stat /bpfassets/libbpf/bpf.o/amd64_kepler.bpf.o: no such file or directory, fall back to bcc attachment
I0107 03:15:25.715786 2678250 exporter.go:237] failed to start : failed to attach bpf assets: no bcc build tag
I0107 03:15:25.715825 2678250 exporter.go:269] Started Kepler in 1.186271ms
```

jiere, 2024-01-07:
BTW, I also double confirmed that v0.7.2 release no such issue, no matter "enabled_process_metrics" or not, kepler stand-alone mode could run well, without running into panic when registering VM Prometheus metrics. @marceloamaral

jichenjc, 2024-01-09:
I am seeing same issue in my `kind` env as well with latest image and also no problem if ENABLE_PROCESS_METRICS is false

```
# kubectl logs kepler-exporter-ztkmb -n kepler
I0109 06:33:03.598266    2026 gpu.go:46] Failed to init nvml, err: failed to init nvml. ERROR_LIBRARY_NOT_FOUND
I0109 06:33:03.603480    2026 qat.go:35] Failed to init qat-telemtry err: could not get qat status exit status 127
I0109 06:33:03.614880    2026 exporter.go:155] Kepler running on version: 1.20.10
I0109 06:33:03.615468    2026 config.go:275] using gCgroup ID in the BPF program: true
I0109 06:33:03.615525    2026 config.go:277] kernel version: 4.18
I0109 06:33:03.615567    2026 exporter.go:167] LibbpfBuilt: true, BccBuilt: false
I0109 06:33:03.615579    2026 exporter.go:186] EnabledBPFBatchDelete: true
I0109 06:33:03.615753    2026 power.go:72] Unable to obtain power, use estimate method
I0109 06:33:03.615854    2026 redfish.go:173] failed to initialize node credential: no supported node credential implementation
I0109 06:33:03.616146    2026 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0109 06:33:03.616162    2026 power.go:72] using none to obtain power
I0109 06:33:03.616176    2026 exporter.go:201] Initializing the GPU collector
I0109 06:33:09.616982    2026 watcher.go:66] Using in cluster k8s config
I0109 06:33:09.717606    2026 watcher.go:137] k8s APIserver watcher was started
I0109 06:33:09.717799    2026 prometheus_collector.go:87] Registered Process Prometheus metrics
I0109 06:33:09.717831    2026 prometheus_collector.go:92] Registered Container Prometheus metrics
I0109 06:33:09.717964    2026 exporter.go:82] exit stack:
goroutine 1 [running]:
runtime/debug.Stack()
        /usr/lib/golang/src/runtime/debug/stack.go:24 +0x65
main.finalizing()
        /workspace/cmd/exporter/exporter.go:81 +0x1d
panic({0x181a8e0, 0xc00007f4e0})
        /usr/lib/golang/src/runtime/panic.go:884 +0x213
github.com/prometheus/client_golang/prometheus.(*Registry).MustRegister(...)
        /workspace/vendor/github.com/prometheus/client_golang/prometheus/registry.go:405
github.com/sustainable-computing-io/kepler/pkg/metrics.(*PrometheusExporter).RegisterMetrics(0xc0004f80f0)
        /workspace/pkg/metrics/prometheus_collector.go:96 +0x375
main.main()
        /workspace/cmd/exporter/exporter.go:230 +0x5b9
I0109 06:33:09.717977    2026 exporter.go:84] Exiting...
```

jichenjc, 2024-01-09:
likely caused by typo so identical prometheus counter want to be registered and caused 2nd failed to do so 

with proposed fix we should be able to register w/ and w/o process enbaled flag set

disable process
```
I0109 07:32:32.310858    5178 prometheus_collector.go:92] Registered Container Prometheus metrics
I0109 07:32:32.310897    5178 prometheus_collector.go:97] Registered VM Prometheus metrics
I0109 07:32:32.310908    5178 prometheus_collector.go:101] Registered Node Prometheus metrics

```

enbale
```
I0109 07:32:59.453688    5406 prometheus_collector.go:87] Registered Process Prometheus metrics
I0109 07:32:59.453719    5406 prometheus_collector.go:92] Registered Container Prometheus metrics
I0109 07:32:59.453744    5406 prometheus_collector.go:97] Registered VM Prometheus metrics
I0109 07:32:59.453782    5406 prometheus_collector.go:101] Registered Node Prometheus metric
```

closed 2024-01-09 by rootfs

jiere, 2024-01-09:
Great finding, thanks Ji!

PR #1176 (2024-01-09): fix context typo
Fixes: #1153