sustainable-computing-io/kepler #1165: No energy usage metrics for isolated CPU cores.

rszmigiel, 2024-01-05:
### What happened?

I'm running PoC with OpenShift 4.13, Kepler 0.9.2 installed with Kepler (Community) Operator. 
One of the use-cases is to visualise energy consumption of DPDK enabled containers. These containers are using isolated CPU cores on a SingleNodeOpenShift installation. 

I got CPUs isolated with the following PerformanceProfile:

```
apiVersion: performance.openshift.io/v2
kind: PerformanceProfile
metadata:
  name: openshift-node-performance-profile
spec:
  additionalKernelArgs:
  - "rcupdate.rcu_normal_after_boot=0"
  - "efi=runtime"
  - "module_blacklist=irdma"
  cpu:
    isolated: "4-31,36-63,68-95,100-127"
    reserved: "0-3,32-35,64-67,96-99"
  hugepages:
    defaultHugepagesSize: 1G
    pages:
      - count: 64
        size: 1G
        node: 0
      - count: 64
        size: 1G
        node: 1
  machineConfigPoolSelector:
    pools.operator.machineconfiguration.openshift.io/master: ""
  nodeSelector:
    node-role.kubernetes.io/master: ''
  numa:
    topologyPolicy: single-numa-node
  realTimeKernel:
    enabled: false
  workloadHints:
    realTime: false
    highPowerConsumption: false
    perPodPowerManagement: false
```

I also got workload partitioning configured (https://docs.openshift.com/container-platform/4.13/scalability_and_performance/enabling-workload-partitioning.html).

While I run a Pod that's configured to use isolated CPU cores, for an instance:

```
        resources:
          limits:
            memory: "24Gi"
            cpu: "50"
            hugepages-1Gi: 24Gi
          requests:
            memory: "24Gi"
            cpu: "50"
            hugepages-1Gi: 24Gi
```

and then run a sample workload to put some load on these cores, for an instance:

```
stress-ng --cpu 50 --io 2 --vm 50 --vm-bytes 1G --timeout 10m --metrics-brief
```

I can observe that assigned CPU cores shows high usage in top output:
[image]

but Kepler's power usage diagrams don't reflect that - they're very flat:
[image]
[... 217 lines cut ...]
      OPERATOR_CONDITION_NAME:      kepler-operator.v0.9.2
    Mounts:                         <none>
  Volumes:                          <none>
Conditions:
  Type           Status  Reason
  ----           ------  ------
  Available      True    MinimumReplicasAvailable
  Progressing    True    NewReplicaSetAvailable
OldReplicaSets:  <none>
NewReplicaSet:   kepler-operator-controller-5d5767d64f (1/1 replicas created)
Events:          <none>

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

vimalk78, 2024-01-08:
AFAIK CPU isolation removes a set of CPUs from scheduling algorithm of kernel. Kepler adds a probe to kernel's `sched_switch` tracepoint to calculate how much cpu time/cpu cycles a process is using, and attributes power usage based on cpu time/cpu cycles of the process.
So if some process is using a CPU which is outside of scheduler, then the probe may not be running for that cpu, Kepler may not be knowing the process's cpu time/cycles to assign any power usage to it, and may not be generating metrics for it.

Cc: @rootfs @marceloamaral

rszmigiel, 2024-01-08:
In such case could we obtain power usage metrics using alternative ways, even they're not as much detailed as in case of the eBPF use? For an instance, to workaround the issue mentioned in this case I used output from ```ipmitool sdr``` command. It provides summarised power usage across all CPUs and memory installed in the system - still better this than nothing ;-)

rootfs, 2024-01-08:
@rszmigiel would you please use the kepler 0.7.2 container image?

rootfs, 2024-01-08:
cc @vprashar2929 @sthaha

rszmigiel, 2024-01-09:
I've used kepler-operator-bundle:0.10.0 and it works! 

[image]

Thank you!

rootfs, 2024-01-09:
great news! thanks for the update @rszmigiel

vimalk78, 2024-01-09:
@rootfs i am really curious to know why it worked with `libbpf` but not with `bcc`.  thats the only difference between two kepler versions.
the approach to calculate the cpu cycles is same in both approaches.

iconeb, 2024-08-27:
It seems it's still happening with latest available version (left side of the graph), compared to 0.7.2 (reinstalled, on the right side)

[image]

sthaha, 2024-08-28:
Reopening issue to continue investigation.

vimalk78, 2024-08-29:
I tried to reproduce this scenario. 
In a machine with 20 cores, i isolated 2 cores and executed stress-ng on these isolated cores. Kepler is able to get energy usage for these processes.

[Screencast from 2024-08-29 19-11-31.webm](https://github.com/user-attachments/assets/bfeb3bee-089f-4ab3-b98e-c4eb3d34de07)

[image]

vimalk78, 2024-08-29:
since the cores are isolated, any task started without cpu pinning will not be allocated to the isolated cores.
in this case the cpu 2 and 12 will not be loaded

[Screencast from 2024-08-29 19-23-31.webm](https://github.com/user-attachments/assets/a28eafb3-91ec-4106-9a93-8801e142e24b)

vimalk78, 2024-09-01:
Cc: @iconeb PTAL

iconeb, 2024-09-04:
I confirm we have a performance profile with reserved and isolated cpus

```
# oc get performanceprofile upf-performance-profile -o json | jq -r .spec.cpu
{
  "isolated": "2-31,34-63,66-95,98-127",
  "reserved": "0-1,32-33,64-65,96-97"
}
```

They are correctly applied at worker node's boot

```
# oc debug node/compute-0.0x4954.openshift.one -- cat /proc/cmdline
[...] intel_iommu=on iommu=pt systemd.unified_cgroup_hierarchy=0 systemd.legacy_systemd_cgroup_controller=1 skew_tick=1 tsc=reliable rcupdate.rcu_normal_after_boot=1 nohz=on rcu_nocbs=2-31,34-63,66-95,98-127 tuned.non_isolcpus=00000003,00000003,00000003,00000003 systemd.cpu_affinity=0,1,32,64,33,65,96,97 intel_iommu=on iommu=pt isolcpus=managed_irq,2-31,34-63,66-95,98-127 nohz_full=2-31,34-63,66-95,98-127 nosoftlockup nmi_watchdog=0 mce=off rcutree.kthread_prio=11 default_hugepagesz=1G hugepagesz=1G hugepages=200 idle=poll rcu_nocb_poll tsc=perfect selinux=0 enforcing=0 noswap clock=pit audit=0 processor.max_cstate=1 intel_idle.max_cstate=0 rcupdate.rcu_normal_after_boot=0 softlockup_panic=0 console=ttyS0,115200n8 pcie_aspm=off pci=noaer firmware_class.path=/var/lib/firmware intel_pstate=disable
```

Pod is running with requests and limits
```
$ oc get pod upf1 -o json | jq .spec.containers[0].resources
{
  "limits": {
    "cpu": "18",
    "hugepages-1Gi": "40Gi",
    "memory": "30Gi",
    "openshift.io/ens785_rn": "3"
  },
  "requests": {
    "cpu": "18",
    "hugepages-1Gi": "40Gi",
    "memory": "30Gi",
    "openshift.io/ens785_rn": "3"
  }
}
```

And on the worker node taskset affinity is assigned as expected
```
taskset -pc 656722      
pid 656722's current affinity list: 3-7,22-25,67-71,86-89
```

The strange thing is that previous graph was created running the *same* pod(s) on the *same* environment, just changing kepler's version in the meantime.

I will try another round of test to provide (if possible) further evidence

rootfs, 2024-09-12:
I have tested Kepler on RHEL that started with isolated CPUs. The isolated CPUs were assigned to a VM. Kepler can capture the VM and report metrics. We have added [this configuration](https://github.com/sustainable-computing-io/kepler-metal-ci/pull/240) in our CI.

sthaha, 2025-08-11:
@rszmigiel I am closing this as the new version of kepler (0.10.0 and above should contain a fix for this. If this isn't the case, could you please open a new bug ?

closed 2025-08-11 by sthaha