sustainable-computing-io/kepler #1172: steps in base image as yumdownloader is not stable

SamYuan1990, 2024-01-06:
### What happened?

```
Dockerfile.builder:19
--------------------
  17 |     
  18 |     WORKDIR /libbpf-source
  19 | >>> RUN yumdownloader --source libbpf
  20 |     RUN rpm2cpio libbpf-1.2.0-1.el9.src.rpm | cpio -iv
  21 |     RUN tar xf ./linux-*el9.tar.xz
--------------------
ERROR: failed to solve: process "/bin/sh -c yumdownloader --source libbpf" did not complete successfully: exit code: 1
```
https://github.com/sustainable-computing-io/kepler/actions/runs/7420068859/job/20190743132

### What did you expect to happen?

enhance base image steps to make it stable in build process.

### How can we reproduce it (as minimally and precisely as possible)?

waiting for schedule build or trigger it manually.

### Kepler image tag

n/a

### Kubernetes version

n/a

### Cloud provider or bare metal

n/a

### OS version

n/a

### Install tools

n/a

### Kepler deployment config

n/a

### Container runtime (CRI) and version (if applicable)

n/a

### Related plugins (CNI, CSI, ...) and versions (if applicable)

n/a

SamYuan1990, 2024-01-06:
```
 > [13/24] RUN yumdownloader --source libbpf:
0.303 
0.303 This system is not registered with an entitlement server. You can use subscription-manager to register.
0.303 
0.307 enabling ubi-9-codeready-builder-source repository
0.365 Last metadata expiration check: 0:01:39 ago on Fri Jan  5 09:19:10 2024.
1.098 [MIRROR] libbpf-1.2.0-1.el9.src.rpm: Curl error (52): Server returned nothing (no headers, no data) for https://cdn-ubi.redhat.com/content/public/ubi/dist/ubi9/9/x86_64/baseos/source/SRPMS/Packages/l/libbpf-1.2.0-1.el9.src.rpm [Empty reply from server]
2.698 [MIRROR] libbpf-1.2.0-1.el9.src.rpm: Curl error (18): Transferred a partial file for https://cdn-ubi.redhat.com/content/public/ubi/dist/ubi9/9/x86_64/baseos/source/SRPMS/Packages/l/libbpf-1.2.0-1.el9.src.rpm [transfer closed with 114447363 bytes remaining to read]
3.942 [MIRROR] libbpf-1.2.0-1.el9.src.rpm: Curl error (18): Transferred a partial file for https://cdn-ubi.redhat.com/content/public/ubi/dist/ubi9/9/x86_64/baseos/source/SRPMS/Packages/l/libbpf-1.2.0-1.el9.src.rpm [transfer closed with 114447363 bytes remaining to read]
4.408 [MIRROR] libbpf-1.2.0-1.el9.src.rpm: Curl error (18): Transferred a partial file for https://cdn-ubi.redhat.com/content/public/ubi/dist/ubi9/9/x86_64/baseos/source/SRPMS/Packages/l/libbpf-1.2.0-1.el9.src.rpm [transfer closed with 114447363 bytes remaining to read]
4.424 [FAILED] libbpf-1.2.0-1.el9.src.rpm: No more mirrors to try - All mirrors were already tried without success
------
```
should we must have register to build base image? @rootfs , @vimalk78

vimalk78, 2024-01-08:
@SamYuan1990  i don't think we need to register. this could be due to some temporary issues in CDN

SamYuan1990, 2024-01-08:
> @SamYuan1990 i don't think we need to register. this could be due to some temporary issues in CDN

let's keep this issue open for days. if cdn issue, it's still block our build and we may consider to fix.

closed 2024-03-15 by stale[bot]