sustainable-computing-io/kepler #1177: Enable process metrics in CI

jichenjc, 2024-01-10:
### What happened?

https://github.com/sustainable-computing-io/kepler/issues/1153
caused some issue , we should enable this in the CI in order to make sure it's not break

### What did you expect to happen?

enable the CI with ENABLE_PROCESS and ENABLE_VM

### How can we reproduce it (as minimally and precisely as possible)?

check https://github.com/sustainable-computing-io/kepler/issues/1153

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

jichenjc, 2024-01-10:
@rootfs @SamYuan1990 

seems current CI didn't enable this , is it reasonable to enbale those e.g VM, PROCESS etc and check them in the CI as well (whether the metrics reported) ? I think it won't lead to too much overhead but guarantee we don't have regression with certain PR
```
build_manifest() {
        $NO_BUILDS && {
                ok "Skipping building manifests"
                return 0
        }
        header "Build Kepler Manifest"
        run make build-manifest \
                OPTS="$OPTS" \
                IMAGE_REPO="$IMAGE_REPO" \
                IMAGE_TAG="$IMAGE_TAG"
}

```

SamYuan1990, 2024-01-10:
which means we need to https://github.com/sustainable-computing-io/kepler/blob/26f6f4925fa34bce1a5851fdb7013912efa872f8/manifests/config/exporter/exporter.yaml#L28 make it overwrite by CI?

rootfs, 2024-01-10:
@SamYuan1990 yes thats the config to enable it

SamYuan1990, 2024-01-10:
where is ENABLE_VM?

SamYuan1990, 2024-03-11:
> where is ENABLE_VM?

@jichenjc / @rootfs , any idea for ENABLE_VM configuration?