sustainable-computing-io/kepler #1191: Support to recognize the VM that is not launched by Libvirt

LeiZhou-97, 2024-01-21:
### What would you like to be added?

Currently, Kepler can only recognize the power of VM when it is launched by Libvirt. (The limitations of code implementation)

Find an unified method to recognize all running VMs on the bare metal.

### Why is this needed?

Improve the accuracy of vm power.

rootfs, 2024-01-23:
@LeiZhou-97 can you provide a list of hypervisors that you think we can support? thanks

jichenjc, 2024-01-23:
I am interested in this topic and I have question with @rootfs as well on hypervisor

or you are thinking about the VM created w/o libvirt/virsh , directly through qemu so we need isntead of check libvirt, check qemu process directly?

LeiZhou-97, 2024-01-24:
Hi @rootfs  @jichenjc 

What I mean is that it would be nice to find a generic way to collect metrics about all surviving VMs regardless of what hypervisor they are using. Directly counting all available hypervisors and then determining them one by one in the code is not a good approach IMO.

Same as @jichenjc , checking out the qemu process was my initial thought, but I'm curious if there's another, better way to do it.

jichenjc, 2024-01-24:
>>>checking out the qemu process was my initial thought, but I'm curious if there's another, better way to do it.

usually different hyperivosr has their own way to manage VM and represent it , I am not sure how we can achieve this 
adding qemu might be a thing we can do , other than that you might want to add VMWare hypervisor support ?

LeiZhou-97, 2024-01-24:
> usually different hyperivosr has their own way to manage VM and represent it , I am not sure how we can achieve this adding qemu might be a thing we can do , other than that you might want to add [...]

Yes, different hypervisors may use different ways to manage VMs, so I opened the issue here to see if the community had any ideas.

My daily use scenario is mainly raw qemu-launched VMs not VMWare.

aharivel, 2024-01-25:
Hi @LeiZhou-97 
My daily use scenario is also launching QEMU VMs by hand. 
But do you think Kepler is a tool for people that launch QEMU VMs by hand ? 
All the effort made in Kepler for the e2e experience shows that it looks mainly for people that use automation. 
And when you talk about QEMU and automation, libvirt is the tool. 
And this is why I choose the libvirt has the main implementation for VM in Kepler. 
Other hypervisor could potentially be implemented but I've got no experience with them. 
This is just my 2 cents on the subject.

LeiZhou-97, 2024-01-25:
@aharivel 
Thanks for your reply.

You may have misunderstood my answer. I'm just saying that I use qemu to boot VM on a daily basis, I'm not saying that kepler needs to specifically recognize this way, after all, end-users are more likely to boot VMs with libvirt/vmware or something like that.  I'm in agreement with you. 

Like I said above, I opened this issue to ask if anyone in the community knows of some low-level and efficient way to identify all the VMs that are running. IMO, it's a bit impossible to adapt all hypervisors by judging the cgroup name the way it is right now. It can be seen as a feature enhancement, not a bug.

jichenjc, 2024-01-29:
>>>It can be seen as a feature enhancement, not a bug.

agree this is an enhancement, and I think VMWare might be best fit here thought someone might add vSphere enablement...
seems libvirt enablement is good enough for most case already