sustainable-computing-io/kepler #1198: GPU NVIDIA H100 PCIe not supported

sbiaudet, 2024-01-25:
### What happened?

I've deployed Kepler on a Kubernetes to monitor a cluster with a GPU node with a NIVIDIA H100 PCIe.

In the kepler logs from this node, I've this error. In parallel I'm monitoring this GPU with a dgcm-exporter instance and it can collect gpu energy consumption metrics correctly.

I0125 07:04:48.972351 1 power.go:86] Failed to collect GPU metrics, trying to initizalize again: failed to get processes' utilization on device {0x7f639b40bdf8}: Not Supported
I0125 07:04:48.972407 1 gpu_nvml.go:62] found 1 gpu devices
I0125 07:04:48.972416 1 gpu_nvml.go:73] GPU 0 NVIDIA H100 PCIe

Do you have an idea ?

### What did you expect to happen?

Collect metrics from NVIDIA H100 PCIe

### How can we reproduce it (as minimally and precisely as possible)?

Run a Kepler instance on a device with a NVIDIA H100 PCIe

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
Client Version: version.Info{Major:"1", Minor:"27", GitVersion:"v1.27.2", GitCommit:"7f6f68fdabc4df88cfea2dcf9a19b2b830f1e647", GitTreeState:"clean", BuildDate:"2023-05-17T14:20:07Z", GoVersion:"go1.20.4", Compiler:"gc", Platform:"darwin/arm64"}
Kustomize Version: v5.0.1
Server Version: version.Info{Major:"1", Minor:"28", GitVersion:"v1.28.2", GitCommit:"89a4ea3e1e4ddd7f7572286090359983e0387b2f", GitTreeState:"clean", BuildDate:"2023-09-13T09:29:07Z", GoVersion:"go1.20.8", Compiler:"gc", Platform:"linux/amd64"}
```

</details>

### Cloud provider or bare metal

<details>
Scaleway Kapsule cluster
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2024-01-25:
I suspect [this comment](https://forums.developer.nvidia.com/t/slurm-gpu-autodetection-with-nvml-and-h100/263866/2) could be related

We need to upgrade the nvml library

rootfs, 2024-01-25:
@sbiaudet can you share your `nvidia-sml` output and dcgm-exporter image tag?

sbiaudet, 2024-01-25:
@rootfs the dgcm-exporter image is nvcr.io/nvidia/k8s/dcgm-exporter:3.3.0-3.2.0-ubuntu22.04

and here my `nvidia-smi `

```
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.129.03             Driver Version: 535.129.03   CUDA Version: 12.3     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA H100 PCIe               On  | 00000000:01:00.0 Off |                   On |
| N/A   55C    P0              97W / 350W |                  N/A |     N/A      Default |
|                                         |                      |              Enabled |
+-----------------------------------------+----------------------+----------------------+

+---------------------------------------------------------------------------------------+
| MIG devices:                                                                          |
+------------------+--------------------------------+-----------+-----------------------+
| GPU  GI  CI  MIG |                   Memory-Usage |        Vol|      Shared           |
|      ID  ID  Dev |                     BAR1-Usage | SM     Unc| CE ENC DEC OFA JPG    |
|                  |                                |        ECC|                       |
|==================+================================+===========+=======================|
|  0    2   0   0  |              37MiB / 40320MiB  | 46      0 |  3   0    3    0    3 |
|                  |               0MiB / 65535MiB  |           |                       |
+------------------+--------------------------------+-----------+-----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|  No running processes found                                                           |
+---------------------------------------------------------------------------------------+
```

rootfs, 2024-01-26:
@sbiaudet thanks for the info. Kepler itself doesn't include nvidia nvml library, let me see what is missing in kepler image

rootfs, 2024-01-29:
@sbiaudet do you use MIG? 

kepler uses process utilization (as you can see in the log) to ratio the power usage. The [nvidia doc](https://docs.nvidia.com/deploy/nvml-api/group__nvmlGridQueries.html#group__nvmlGridQueries_1gb0ea5236f5e69e63bf53684a11c233bd) has a note:
```
Note:
On MIG-enabled GPUs, querying process utilization is not currently supported.
```

sbiaudet, 2024-01-29:
Yes I’m using MIG but with another dashboard via dcgm-exporter installed with gpu-operator I’ve this metrics on consumption. In Kepler the GPU reported consumption is 250W.

￼

sbiaudet, 2024-01-29:
Yes I’m using MIG but with another dashboard via dcgm-exporter installed with gpu-operator I’ve this metrics on consumption. In Kepler the GPU reported consumption is 250W.

[image]

rootfs, 2024-01-29:
@sbiaudet thanks for the additional info. So it looks there are two issues you found: MIG support, and power consumption reporting mismatch. 

For the 2nd issue, was it also on a MIG GPU? What is the name of the kepler metrics that reported 250W?

sbiaudet, 2024-01-29:
@rootfs the Kepler metrics that reported 250W is kepler_container_gpu_joules_total.

rootfs, 2024-01-30:
@sbiaudet thanks, looking into these two issues now.

rootfs, 2024-01-30:
@sbiaudet I found a bug and have a fix in #1210  but not sure if this is the fix to your 2nd issue. I tested it on my own local GPU (no MIG) and it can report `kepler_container_gpu_joules_total` correctly:

I used [this](https://github.com/sustainable-computing-io/kepler/blob/main/manifests/tests/gpu/cuda.yaml) gpu test pod. The `kepler_container_gpu_joules_total` i observed rose from 0 to 56W then back to 0 after pod completed, while the `nvidia-smi dmon -s p` output was 67W during the workload was active.

rootfs, 2024-01-30:
Once I get a MIG GPU, will check the first issue

rootfs, 2024-02-01:
@sbiaudet can you try the kepler latest container image and check if the 2nd issue is resolved? For the MIG issue, let's track it in #1215

marceloamaral, 2024-02-02:
@rootfs it might be possible to get the process utilization in MIG devices 
https://github.com/NVIDIA/nvidia-settings/blob/main/src/nvml.h#L5629
I will try to investigate it a lille bit more.

sbiaudet, 2024-02-05:
Hi @rootfs, I'm using helm chart to deploy Kepler on my cluster. Does docker nightly build or branch build exists ?

rootfs, 2024-02-05:
@sbiaudet I see, the fix is not in a release yet. I'll let you know when the new chart is available.

rootfs, 2024-02-14:
I can reproduce this issue on a MIG instance. Working on a fix now.

PR #1238 (2024-02-21): NVIDIA Mig support
fix #1198 

# Test status
I am running a text generation inference pod and the measured power
## Load generation
`kubectl exec` to the `text-generation-inference` pod, and run the command to generate text and pause 5 seconds after each run:
```bash
# while true; do curl localhost:8080/generate -X POST -H "Content-Type: application/json" -d '{"inputs": "<s>[INST] Write a K8s YAML file to create a pod that deploys nginx[/INST]", "parameters": {"max_new_tokens": 400}}';sleep 5;done
```
## Test result
```console
# nvidia-smi
+----------------------------------------------------- [... cut]
closed 2024-02-21 by rootfs