sustainable-computing-io/kepler #1215: Improve GPU power attribution

rootfs, 2024-02-01:
### What would you like to be added?

The current GPU power attribution is based on SM utilization. It doesn't account for memory usage. We should consider both factors.

### Why is this needed?

Build a GPU power model that has both SM and memory utilization, so that workloads of compute and memory intensive natures can be attributed with fair power usage.

@marceloamaral @sunya-ch wdyt?

rootfs, 2024-02-01:
SM/Memory utilization models can also be extended to model MIG devices power consumption, if such devices cannot read power usage directly.

rootfs, 2024-02-01:
related issue #1198

marceloamaral, 2024-02-06:
@rootfs The problem with using memory utilization in the power model is that it only tells us about how memory is allocated, not how much it's actually being used for tasks like reading and writing data. These operations are what really affect how much power the GPU's memory is consuming.

We can get how much memory the GPU is using for reading and writing data, but currently, we can't do this for each process running on the GPU. So, it's probably best to focus only on the Ratio Power model using SM utilization for now. Another advantage of the simple Ratio power model is that it is not workload-specific compared to a trained model.

marceloamaral, 2024-02-06:
For the MIG, we can find which process is using the MIG device, so that Kepler will discovery the process/containers.
However, I am still investigating how to associate the GPU utilization to each MIG/Process.
NVIDIA dcgm has some logic for that, I am currently trying to understand that.
https://docs.nvidia.com/datacenter/dcgm/latest/user-guide/feature-overview.html