sustainable-computing-io/kepler #1221: No VM detected in the Kepler exporter

aharivel, 2024-02-08:
### What happened?

Hi all,

I've installed the latest RPM (release-0.7.2) available on my machine and tried to get the VM metrics. 
I only get : 
kepler_vm_package_joules_total{mode="dynamic",name="driver",pid="1533376"} 0
kepler_vm_package_joules_total{mode="dynamic",name="driver",pid="1533377"} 0
kepler_vm_package_joules_total{mode="dynamic",name="driver",pid="1533378"} 0
kepler_vm_package_joules_total{mode="dynamic",name="driver",pid="1533379"} 0
kepler_vm_package_joules_total{mode="dynamic",name="driver",pid="1533380"} 0
kepler_vm_package_joules_total{mode="dynamic",name="driver",pid="1533381"} 0
kepler_vm_package_joules_total{mode="dynamic",name="driver",pid="1533382"} 0
...

I've got 3 VM running via Libvirt:
[root@fedora aharivel]# cat /run/libvirt/qemu/centos-stream9.pid 
1673350 
[root@fedora aharivel]# cat /run/libvirt/qemu/vnf1.pid 
3546233
[root@fedora aharivel]# cat /run/libvirt/qemu/vnf2.pid 
3490657

Even looking at kepler_process_* those PID doesn't exist in the list.

But, I can see that the first PID 1533376 is avalaible:
kepler_process_uncore_joules_total{command="virtqemud",mode="dynamic",pid="1533376"} 7.143
kepler_process_uncore_joules_total{command="virtqemud",mode="idle",pid="1533376"} 0

### What did you expect to happen?

- "driver" was supposed to be removed from the kepler VM metrics #1028
- I should see 3 VM in the metrics exporter: centos-stream9, vnf1, vnf2

### How can we reproduce it (as minimally and precisely as possible)?

- Launch 1 or multiple VM with libvirt
- install RPM (release-0.7.2) 
- web browser: 0.0.0.0:8888/metrics
- Check the "name=" of kepler_vm_* metrics

### Anything else we need to know?

// Run as root
$ ps -aux | grep kepler
root     3604842  0.3  0.1 1863472 41088 ?       Ssl  13:47   0:17 kepler

[root@fedora aharivel]# cat /proc/1673350/cgroup 
0::/machine.slice/machine-qemu\x2d3\x2dcentos\x2dstream9.scope/libvirt/emulator

[root@fedora aharivel]# cat /proc/3546233/cgroup 
0::/machine.slice/machine-qemu\x2d22\x2dvnf1.scope/libvirt/emulator

[root@fedora aharivel]# cat /proc/3490657/cgroup 
0::/machine.slice/machine-qemu\x2d18\x2dvnf2.scope/libvirt/emulator

### Kepler image tag

<details>
release-0.7.2
</details>

### Kubernetes version

<details>

</details>

### Cloud provider or bare metal

<details>
baremetal
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
NAME="Fedora Linux"
VERSION="39 (Workstation Edition)"
ID=fedora
VERSION_ID=39
VERSION_CODENAME=""
PLATFORM_ID="platform:f39"
PRETTY_NAME="Fedora Linux 39 (Workstation Edition)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:39"
DEFAULT_HOSTNAME="fedora"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f39/system-administrators-guide/"
SUPPORT_URL="https://ask.fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=39
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=39
SUPPORT_END=2024-11-12
VARIANT="Workstation Edition"
VARIANT_ID=workstation
$ uname -a
Linux fedora 6.6.11-200.fc39.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Jan 10 19:25:59 UTC 2024 x86_64 GNU/Linux
```

</details>

### Install tools

<details>
dnf localinstall
</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2024-02-09:
can you try the latest image? i saw the VM power on my end built from source

marceloamaral, 2024-02-09:
Is the BPF working? Could you please share the logs with verbosity enabled?
As @rootfs said, can you try the latest image?

aharivel, 2024-02-09:
Hi @rootfs @marceloamaral 
Thanks for your message.
I woule like to give you more output but when I compile Kepler with:
$make kepler
There is no more outpout directory "_output/bin" where I can find the binary. 
Is there any modification in the makefile that erase that directory ? 
Is there a new command to build the binary ?

rootfs, 2024-02-09:
@aharivel i built kepler using `make _build_local`

aharivel, 2024-02-12:
Hi @rootfs @marceloamaral ,
Indeed the last version is wortking properly  :+1: 

Now my question is: when will be the next release ? Because the current one is 2month old and when I encourage people to test Kepler I point them to use the RPM available in the Release of the Github. 
That would be great if the last version available is actually working like the main branch !

rootfs, 2024-02-12:
@aharivel just created it [here](https://github.com/sustainable-computing-io/kepler/releases/tag/v0.7.3)

closed 2024-02-12 by rootfs

aharivel, 2024-02-12:
@rootfs thanks a lot Huamin ! +1