sustainable-computing-io/kepler #1224: rpmautospec is missing

SamYuan1990, 2024-02-11:
### What happened?

rpmautospec hence the daily build for RPM failed.

### What did you expect to happen?

rpmautospec hence the daily build for RPM failed, #1206 don't fix the issue as the dependency for rpmautospec is not meet on ubi yum repo. 

### How can we reproduce it (as minimally and precisely as possible)?

```
bash-5.1# yum install rpmautospec 
Updating Subscription Management repositories.
Unable to read consumer identity

This system is not registered with an entitlement server. You can use subscription-manager to register.

Last metadata expiration check: 0:39:38 ago on Sun Feb 11 13:46:39 2024.
Error: 
 Problem: package python3-rpmautospec-0.6.0-2.el9.noarch from epel requires (python3.9dist(pygit2) < 2 with python3.9dist(pygit2) >= 1.4), but none of the providers can be installed
  - package rpmautospec-0.6.0-2.el9.noarch from epel requires python3-rpmautospec = 0.6.0-2.el9, but none of the providers can be installed
  - package python3-pygit2-1.12.2-1.el9.x86_64 from epel requires libgit2.so.1.6()(64bit), but none of the providers can be installed
  - conflicting requests
  - nothing provides libhttp_parser.so.2()(64bit) needed by libgit2-1.6.4-1.el9.x86_64 from epel
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)
```

### Kepler image tag

n/a

### Kubernetes version

n/a

### Cloud provider or bare metal

n/a

### OS version

n/a

### Install tools

n/a

### Kepler deployment config

n/a

### Container runtime (CRI) and version (if applicable)

n/a

### Related plugins (CNI, CSI, ...) and versions (if applicable)

n/a

SamYuan1990, 2024-02-11:
https://www.rpmfind.net/linux/rpm2html/search.php?query=libhttp_parser.so.2()(64bit)
there is no provider found for ubi?

rootfs, 2024-02-12:
new builder image is pushed and [rpm just got build](https://github.com/sustainable-computing-io/kepler/releases/tag/v0.7.3), the issue should be resolved now

closed 2024-02-12 by rootfs