sustainable-computing-io/kepler #1233: Support multiple GPUs

rootfs, 2024-02-14:
### What would you like to be added?

Resource usage `ResourceUsage` can only capture one GPU usage. The NVML API also reports usage per GPU. If there are multiple GPUs in the system, then it is unclear how to translate usage of single GPU usage to multiple GPUs, especially when these GPUs are heterogenous. 

### Why is this needed?

Make `ResourceUsage` a 2D slice so it can capture usage and power on all of them

@marceloamaral @sunya-ch

marceloamaral, 2024-02-16:
@rootfs,

The `ResourceUsage` struct is defined as a map of string keys to` UInt64StatCollection` values.

The `UInt64StatCollection` struct, in turn, is a collection of` UInt64Stat` values, with each UInt64Stat representing a specific metric.

Here's the definition of UInt64StatCollection:
```go
// UInt64StatCollection keeps a collection of UInt64Stat
type UInt64StatCollection struct {
    Stat map[string]*UInt64Stat
}
```
This structure allows for the storage of metrics for multiple GPUs, with each GPU identified by its ID in the map key.

However, seeing of the code, it appears that there is a mistake. The current implementation does not store the metrics per GPU as intended, it is using a generic key instead...

To address this issue, I have submitted the PR #1236 that fixes the bug and ensures that the ResourceUsage stores the metrics per GPU. As the processes metrics are aggregated later to create the node metrics, the node metric will also have the metrics per GPU.

rootfs, 2024-02-20:
this is fixed

closed 2024-02-20 by rootfs