sustainable-computing-io/kepler #1279: Kernel consumption too high

jochen-schuettler, 2024-02-29:
### What happened?

The measured consumption for the kernel is much too high, beyond what the machine is physically capable of.
[image]

### What did you expect to happen?

Correct measurement for kernel.

### How can we reproduce it (as minimally and precisely as possible)?

Install kepler 0.7.7 into AWS ECS. Observe measured kernel consumption over some days.

### Kepler image tag

<details>
0.7.7
</details>

### Kubernetes version

<details>
1.24
</details>

### Cloud provider or bare metal

<details>
AWS EKS / ECS
</details>

### OS version

<details>
Amazon Linux 2
</details>

### Install tools

<details>
Helm
</details>

### Kepler deployment config

<details>
Unchanged from Helm chart.
</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>
CNI Version: v1.12.0-eksbuild.2 CSI Version: v1.14.0-eksbuild.1 
</details>

rootfs, 2024-03-04:
I suspect this is a overflow issue. Do you have this issue before or only during this period?

jochen-schuettler, 2024-03-05:
it started with 0.7.3 -> 0.7.7, while 0.7.3 had other issues. ongoing problem
[image]

jochen-schuettler, 2024-03-05:
namespace "tools" is also affected
[image]

jochen-schuettler, 2024-03-05:
compare namespace "kepler" in the same frame
[image]

marceloamaral, 2024-03-05:
@jochen-schuettler thanks for investigating this.

- Are you running on bare-metal or VM?
- Can you check the metrics in prometheus:
    - `rate(kepler_container_bpf_cpu_time_ms_total[1m])`
    - `rate(kepler_container_task_clock_ms_total[1m])`
    - `rate(kepler_container_package_joules_total[1m])`
    - `rate(kepler_container_dram_joules_total[1m])`
    - `rate(kepler_container_other_joules_total[1m])`
    - `rate(kepler_node_package_joules_total[1m])`
    - `rate(kepler_node_dram_joules_total[1m])`
    - `rate(kepler_node_other_joules_total[1m])`

jochen-schuettler, 2024-03-05:
It's a VM on AWS ECS. I don't have access to Prometheus, but to Grafana. This is the first metric, `rate(kepler_container_bpf_cpu_time_ms_total[1m])`
[image]

jochen-schuettler, 2024-03-05:
`rate(kepler_container_task_clock_ms_total[1m])`
[image]

jochen-schuettler, 2024-03-05:
`rate(kepler_container_package_joules_total[1m])`
[image]

jochen-schuettler, 2024-03-05:
`rate(kepler_container_dram_joules_total[1m])`
[image]

jochen-schuettler, 2024-03-05:
`rate(kepler_container_other_joules_total[1m])`
[image]

jochen-schuettler, 2024-03-05:
`rate(kepler_node_dram_joules_total[1m])`
[image]

jochen-schuettler, 2024-03-05:
`rate(kepler_node_other_joules_total[1m])`
[image]

jochen-schuettler, 2024-03-08:
(no change by v0.7.8)

jochen-schuettler, 2024-03-11:
@sunya-ch : see above, any idea ?

jochen-schuettler, 2024-04-04:
The issue is still open. @marceloamaral , @rootfs : Can I help you in any way to investigate ?

TWpgo, 2024-04-04:
[image]
I also get some really unrealistic measurements from time to time

jochen-schuettler, 2024-04-08:
Additional Info: The VM is running on a AMD EPYC 7R13 Processor (stepping 1). This is not included in [cpus.yaml](https://github.com/sustainable-computing-io/kepler/blob/main/data/cpus.yaml). So maybe the problem is related to that?

rootfs, 2024-04-09:
Maybe there are some overflow during extended tests? @vprashar2929 have you seen this in any tests?

rootfs, 2024-04-09:
@jochen-schuettler can you get prometheus stats `kepler_container_joules_total` over the period of spikes? This stats is used to calculate the power consumption.

jochen-schuettler, 2024-04-10:
[Total Power Consumption (W) in Namespace All-data-2024-04-10 09 15 31.csv](https://github.com/sustainable-computing-io/kepler/files/14928804/Total.Power.Consumption.W.in.Namespace.All-data-2024-04-10.09.15.31.csv)

vprashar2929, 2024-04-17:
@rootfs I have seen this before and in the case of VM's Kepler does report higher energy usage. I think post `0.7.2` only it started. https://github.com/sustainable-computing-io/kepler/issues/1142

jochen-schuettler, 2024-04-19:
We have occurences regardless of the kernel namespace in a local setup in both BM and VM. Looking at bpf_cpu_time_ms there reveals rapid rises over some minutes. kepler 0.7.2 seems ok, yes.