sustainable-computing-io/kepler #1283: Integrate Kepler with Graphite ?

annis-souames, 2024-03-04:
Hello

I have been using Kepler for some time and it has worked fine, however, due to some external reasons, I had to move from Prometheus + Grafana stack into Grafana Cloud + Graphite for storing metrics, I was wondering what would be the easiest way to get Kepler data into graphite (even if I have to write a small script for this). 

Thanks

annis-souames, 2024-03-06:
The suggested poject seems to require `prom_url`, I assume it needs Prometheus right ?

rootfs, 2024-03-06:
It doesn't need prometheus, @jesuino can you help? thanks

jesuino, 2024-03-06:
Hello @rootfs I didn't send the metrics dashboards at the time because I was not able to enable CORS on Kepler. However, here's the dashboard that reads directly from Kepler metrics (no Prometheus):

https://github.com/jesuino/dashbuilder-yaml-samples/blob/main/kepler/kepler_metrics.dash.yaml

You can see it running with an unreleased Dashbuilder version here:

https://jesuino.github.io/dashboards?import=https://raw.githubusercontent.com/jesuino/dashbuilder-yaml-samples/main/kepler/kepler_metrics.dash.yaml

Or you can edit with the latest released dashbuilder version here: 

https://start.kubesmarts.org/#/import?url=https://github.com/jesuino/dashbuilder-yaml-samples/blob/main/kepler/kepler_metrics.dash.yaml

rootfs, 2024-03-07:
@jesuino thank you for the info

@annis-souames if dashboard builder works for your use case, please let's know.

closed 2026-05-10 by github-actions[bot]