sustainable-computing-io/kepler #1302: Defects and vulnerabilities reported by Synk scan

vprashar2929, 2024-03-17:
Recently we ran a Synk scan on the [openshift-power-monitoring/kepler](https://github.com/openshift-power-monitoring/kepler) which is a fork of this repository. Upon running the scan following issues in the code were reported:
```
Testing /go/src/github.com/openshift-power-monitoring/kepler ...
 ✗ [Low] Path Traversal
   ID: 06f12b18-f109-4efb-8ecb-464b66443e4f 
   Path: vendor/github.com/onsi/ginkgo/v2/ginkgo/internal/test_suite.go, line 226 
   Info: Unsanitized input from file name flows into os.ReadDir, where it is used as a path. This may result in a Path Traversal vulnerability and allow an attacker to get a list of arbitrary files.
 ✗ [Low] Path Traversal
   ID: 6c4ab052-e7bc-4754-bb25-b71ce1b9873c 
   Path: vendor/github.com/onsi/ginkgo/v2/ginkgo/internal/test_suite.go, line 245 
   Info: Unsanitized input from file name flows into os.ReadDir, where it is used as a path. This may result in a Path Traversal vulnerability and allow an attacker to get a list of arbitrary files.
 ✗ [Low] Path Traversal
   ID: 7ffdd1a1-99a3-446d-9cb0-a4b3dc173594 
   Path: vendor/github.com/onsi/ginkgo/v2/ginkgo/internal/test_suite.go, line 233 
   Info: Unsanitized input from file name flows into os.ReadFile, where it is used as a path. This may result in a Path Traversal vulnerability and allow an attacker to read arbitrary files.
 ✗ [Low] Path Traversal
   ID: 0c4438a8-2164-4517-8123-2ba2fa0eed65 
   Path: vendor/github.com/onsi/ginkgo/v2/ginkgo/internal/test_suite.go, line 276 
   Info: Unsanitized input from file name flows into os.ReadFile, where it is used as a path. This may result in a Path Traversal vulnerability and allow an attacker to read arbitrary files.
 ✗ [Low] Use of Password Hash With Insufficient Computational Effort
   ID: 558631a9-97b4-4947-9e36-8b039863ef5f 
   Path: vendor/github.com/google/uuid/hash.go, line 44 
   Info: The MD5 hash (used in crypto.md5.New) is insecure. Consider changing it to a secure hash algorithm
 ✗ [Low] Use of Password Hash With Insufficient Computational Effort
   ID: 7adf07d8-0f84-4eb4-a79b-78ab4c876231 
   Path: vendor/github.com/google/uuid/hash.go, line 52 
   Info: The SHA1 hash (used in crypto.sha1.New) is insecure. Consider changing it to a secure hash algorithm
 ✗ [Low] Use of Password Hash With Insufficient Computational Effort
   ID: 278f74a8-94a8-42ee-b41e-a6e1276e9a80 
   Path: vendor/github.com/godbus/dbus/v5/auth_sha1.go, line 50 
   Info: The SHA1 hash (used in crypto.sha1.New) is insecure. Consider changing it to a secure hash algorithm
 ✗ [Low] Use of Password Hash With Insufficient Computational Effort
   ID: d6e21ffc-64ec-4eaa-8d2b-b42a75e1acaf 
   Path: vendor/github.com/cilium/ebpf/asm/instruction.go, line 703 
   Info: The SHA1 hash (used in crypto.sha1.New) is insecure. Consider changing it to a secure hash algorithm
 ✗ [Low] Use of Password Hash With Insufficient Computational Effort
   ID: 9be44835-52d0-479b-96d7-b4d50f132538 
   Path: vendor/github.com/go-task/slim-sprig/crypto.go, line 17 
   Info: The SHA1 hash (used in crypto.sha1.Sum) is insecure. Consider changing it to a secure hash algorithm
 ✗ [Low] Use of Hardcoded Credentials
   ID: 1ad0f9f3-e4b6-4755-b54a-5e01584df7e8 
   Path: vendor/k8s.io/klog/v2/klog_file.go, line 48 
   Info: Do not hardcode credentials in code. Found hardcoded credential used in userName.
 ✗ [Low] Use of Hardcoded Credentials
   ID: 489145bf-b263-430e-b2c2-689f25f34728 
   Path: pkg/sensors/platform/source/redfish_test.go, line 70 
   Info: Do not hardcode passwords in code. Found hardcoded saved in Password.
 ✗ [Low] Use of Hardcoded Credentials
   ID: 79877ddf-7b51-4d76-9fb8-408cfa94a68a 
   Path: pkg/sensors/platform/source/redfish_test.go, line 69 
   Info: Do not hardcode credentials in code. Found hardcoded credential used in Username.
 ✗ [Medium] Division By Zero
   ID: e832019e-616e-4d56-80c7-ddf2fa7f4217 
   Path: bpfassets/bcc/bcc.c, line 188 
   Info: The divisor in the division operator is possibly zero, which may cause a division by zero to occur. Divisors should be checked to be non-zero before use.
 ✗ [Medium] Division By Zero
   ID: 07115c73-80b5-45d7-8016-aa6b711e23ab 
   Path: bpfassets/bcc/bcc.c, line 192 
   Info: The divisor in the division operator is possibly zero, which may cause a division by zero to occur. Divisors should be checked to be non-zero before use.
 ✗ [Medium] Use of Hardcoded Credentials
   ID: 24ab3a28-7175-4073-960d-f7d16e4d823e 
   Path: vendor/github.com/onsi/ginkgo/v2/ginkgo/internal/run.go, line 336 
   Info: Do not hardcode passwords in code. Found hardcoded saved in passed.
 ✗ [Medium] Use of Hardcoded Credentials
   ID: 5ff31cd8-c65e-4f2b-9931-333d5607fe64 
   Path: vendor/github.com/onsi/ginkgo/v2/ginkgo/internal/run.go, line 338 
   Info: Do not hardcode passwords in code. Found hardcoded saved in passed.
 ✗ [Medium] Use of Hardcoded Credentials
   ID: 3604a7d4-f0a1-41b4-b6cb-d2f469f62d81 
   Path: vendor/github.com/opencontainers/runc/libcontainer/user/lookup_unix.go, line 16 
[... 53 lines cut ...]
 ✗ [Medium] Path Traversal
   ID: da7b5e73-bf14-48e0-b342-79cd063844b0 
   Path: vendor/github.com/opencontainers/runc/libcontainer/cgroups/utils.go, line 247 
   Info: Unsanitized input from file name flows into os.ReadDir, where it is used as a path. This may result in a Path Traversal vulnerability and allow an attacker to get a list of arbitrary files.
 ✗ [Medium] Path Traversal
   ID: 9da35a00-7d67-4d83-bcd7-a8168315fefe 
   Path: vendor/github.com/opencontainers/runc/libcontainer/cgroups/utils.go, line 257 
   Info: Unsanitized input from file name flows into os.ReadDir, where it is used as a path. This may result in a Path Traversal vulnerability and allow an attacker to get a list of arbitrary files.
 ✗ [Medium] Cross-site Scripting (XSS)
   ID: f314eae2-1d76-41d2-a0fe-80ecbdebf23c 
   Path: cmd/exporter/exporter.go, line 251 
   Info: Unsanitized input from a CLI argument flows into Write, where it is used to render an HTML page returned to the user. This may result in a stored/second-order Cross-Site Scripting (XSS) vulnerability.
 ✗ [High] Generation of Error Message Containing Sensitive Information
   ID: 8caf48f2-b306-402e-af2c-60729ac92822 
   Path: vendor/github.com/onsi/ginkgo/v2/internal/suite.go, line 384 
   Info: Information exposure through error stack trace in fmt.Printf.
✔ Test completed
Organization:      openshift-ci-internal
Test type:         Static code analysis
Project path:      /go/src/github.com/openshift-power-monitoring/kepler
Summary:
  34 Code issues found
  1 [High]   21 [Medium]   12 [Low] 
Code Report Complete
```

SamYuan1990, 2024-03-17:
github.com/onsi/ginkgo which is just for testing, hence ... @vprashar2929 , do we have any good approach to show a real vul?
btw, is it possible to make a xss for exporter?

SamYuan1990, 2024-03-17:
is Snyk open for integration with GHA?

SamYuan1990, 2024-03-17:
btw, should we use template as https://github.com/sustainable-computing-io/kepler/security/advisories/new for report?

SamYuan1990, 2024-03-17:
after a quick review, IMO:
- vulnerabilities from vendors, please open issue to the vendors specific.
- our TLS check logic to get either k8s cluster info or redfish info.
- the exporter thing... as env injection.
- test code or bcc code vulnerabilities which is no impact to product.

@vprashar2929 , I suggest we following https://www.first.org/cvss/v4.0/specification-document for our CVE report stand and reuse https://github.com/sustainable-computing-io/kepler/security/advisories/new template.