sustainable-computing-io/kepler #1310: Instability when running in AWS EKS on some instance families

qa-florian-wende, 2024-03-19:
### What happened?

We were testing different deployments in an AWS EKS cluster to monitor which uses how much energy. Although we could see individual pods and containers in the Kepler Dashboard, all values were zero except for those of the system processes.

[image]

We also checked the metrics endpoint of the Kepler exporters directly, so it isn't just an issue with the Kepler Dashboard.
We ran some loadtests on our deployments, so it isn't just a problem with values rounded down to zero.
We repeated our tests with various AWS EC2 instance families, instance sizes and Kubernetes Versions (using eksctl to deploy the cluster) and whenever one combination of the parameters worked, when we re-deployed the cluster with the same parameters it didn't work anymore (once it even stopped working while the cluster continued to run).

### What did you expect to happen?

I expected to see measurements all the time.

### How can we reproduce it (as minimally and precisely as possible)?

Unfortunately, at this moment, we are not able to reproduce the bug consistently.
The setup which ceased to work while the cluster was running was with the Bottlerocket ami-family (which seems to be the only Linux distribution which is supported by eksctl and supports cgroup v2) on t3.large instances in eu-north-1 with Kubernetes Version 1.28.

### Kepler image tag

<details>
`latest`
We witnessed the behaviour first on Nov 20 2023 and last on Feb 22 2024.
</details>

### Kubernetes version

We witnessed the behaviour on 1.24, 1.28 and 1.29.

### Cloud provider or bare metal

AWS

### OS version

<details>
We tried Amazon Linux 2, Ubuntu 20.04 and Bottlerocket, but we didn't specify an AMI, eksctl chose one for us.
</details>

### Install tools

eksctl and helm

### Kepler deployment config

<details>
```yaml
apiVersion: v1
kind: Namespace
metadata:
  labels:
    pod-security.kubernetes.io/audit: privileged
    pod-security.kubernetes.io/enforce: privileged
    pod-security.kubernetes.io/warn: privileged
    security.openshift.io/scc.podSecurityLabelSync: "false"
    sustainable-computing.io/app: kepler
  name: kepler
---
apiVersion: v1
kind: ServiceAccount
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: kepler-sa
  namespace: kepler
---
apiVersion: rbac.authorization.k8s.io/v1
kind: Role
metadata:
  labels:
    app.kubernetes.io/component: prometheus
    app.kubernetes.io/instance: k8s
    app.kubernetes.io/name: prometheus
    sustainable-computing.io/app: kepler
  name: prometheus-k8s
  namespace: kepler
rules:
- apiGroups:
  - ""
  resources:
  - services
  - endpoints
  - pods
  verbs:
  - get
  - list
  - watch
- apiGroups:
  - extensions
  resources:
  - ingresses
  verbs:
  - get
  - list
  - watch
- apiGroups:
  - networking.k8s.io
  resources:
  - ingresses
  verbs:
  - get
  - list
  - watch
---
apiVersion: rbac.authorization.k8s.io/v1
kind: ClusterRole
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: kepler-clusterrole
rules:
- apiGroups:
  - ""
  resources:
  - nodes/metrics
  - nodes/proxy
  - nodes/stats
  - pods
  verbs:
  - get
  - watch
  - list
---
apiVersion: rbac.authorization.k8s.io/v1
kind: RoleBinding
metadata:
  labels:
    app.kubernetes.io/component: prometheus
    app.kubernetes.io/instance: k8s
    app.kubernetes.io/name: prometheus
    sustainable-computing.io/app: kepler
  name: prometheus-k8s
  namespace: kepler
roleRef:
  apiGroup: rbac.authorization.k8s.io
  kind: Role
  name: prometheus-k8s
subjects:
- kind: ServiceAccount
  name: prometheus-k8s
  namespace: monitoring
---
apiVersion: rbac.authorization.k8s.io/v1
kind: ClusterRoleBinding
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: kepler-clusterrole-binding
roleRef:
  apiGroup: rbac.authorization.k8s.io
  kind: ClusterRole
  name: kepler-clusterrole
subjects:
- kind: ServiceAccount
  name: kepler-sa
  namespace: kepler
---
apiVersion: v1
data:
  BIND_ADDRESS: 0.0.0.0:9102
  CGROUP_METRICS: '*'
  CPU_ARCH_OVERRIDE: ""
  ENABLE_EBPF_CGROUPID: "true"
  ENABLE_GPU: "true"
  ENABLE_PROCESS_METRICS: "false"
  EXPOSE_CGROUP_METRICS: "true"
  EXPOSE_HW_COUNTER_METRICS: "true"
  EXPOSE_IRQ_COUNTER_METRICS: "true"
  EXPOSE_KUBELET_METRICS: "true"
  KEPLER_LOG_LEVEL: "5"
  KEPLER_NAMESPACE: kepler
  METRIC_PATH: /metrics
  MODEL_CONFIG: |
    CONTAINER_COMPONENTS_ESTIMATOR=false
    # by default we use buildin weight file
    # CONTAINER_COMPONENTS_INIT_URL=https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentModelWeight/CgroupOnly/ScikitMixed/ScikitMixed.json
  REDFISH_PROBE_INTERVAL_IN_SECONDS: "60"
  REDFISH_SKIP_SSL_VERIFY: "true"
kind: ConfigMap
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: kepler-cfm
  namespace: kepler
---
apiVersion: v1
data:
  redfish.csv: |
    eW91cl9rdWJlbGV0X25vZGVfbmFtZSxyZWRmaXNoX3VzZXJuYW1lLHJlZGZpc2hfcGFzc3
    dvcmQsaHR0cHM6Ly9yZWRmaXNoX2lwX29yX2hvc3RuYW1lCg==
kind: Secret
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: redfish-4kh9d7bc7m
  namespace: kepler
type: Opaque
---
apiVersion: v1
kind: Service
metadata:
  labels:
    app.kubernetes.io/component: exporter
    app.kubernetes.io/name: kepler-exporter
    sustainable-computing.io/app: kepler
  name: kepler-exporter
  namespace: kepler
spec:
  clusterIP: None
  ports:
  - name: http
    port: 9102
    targetPort: http
  selector:
    app.kubernetes.io/component: exporter
    app.kubernetes.io/name: kepler-exporter
    sustainable-computing.io/app: kepler
---
apiVersion: apps/v1
kind: DaemonSet
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: kepler-exporter
  namespace: kepler
spec:
  selector:
    matchLabels:
      app.kubernetes.io/component: exporter
      app.kubernetes.io/name: kepler-exporter
      sustainable-computing.io/app: kepler
  template:
    metadata:
      labels:
        app.kubernetes.io/component: exporter
        app.kubernetes.io/name: kepler-exporter
        sustainable-computing.io/app: kepler
    spec:
      containers:
      - args:
        - /usr/bin/kepler -v=5 -kernel-source-dir=/usr/share/kepler/kernel_sources
          -redfish-cred-file-path=/etc/redfish/redfish.csv
        command:
        - /bin/sh
        - -c
        env:
        - name: NODE_IP
          valueFrom:
            fieldRef:
              fieldPath: status.hostIP
        - name: NODE_NAME
          valueFrom:
            fieldRef:
              fieldPath: spec.nodeName
        image: quay.io/sustainable_computing_io/kepler:latest
        imagePullPolicy: Always
        livenessProbe:
          failureThreshold: 5
          httpGet:
            path: /healthz
            port: 9102
            scheme: HTTP
          initialDelaySeconds: 10
          periodSeconds: 60
          successThreshold: 1
          timeoutSeconds: 10
        name: kepler-exporter
        ports:
        - containerPort: 9102
          name: http
        resources:
          requests:
            cpu: 100m
            memory: 400Mi
        securityContext:
          privileged: true
        volumeMounts:
        - mountPath: /lib/modules
          name: lib-modules
        - mountPath: /sys
          name: tracing
        - mountPath: /proc
          name: proc
        - mountPath: /etc/kepler/kepler.config
          name: cfm
          readOnly: true
        - mountPath: /etc/redfish
          name: redfish
          readOnly: true
      dnsPolicy: ClusterFirstWithHostNet
      serviceAccountName: kepler-sa
      tolerations:
      - effect: NoSchedule
        key: node-role.kubernetes.io/master
      volumes:
      - hostPath:
          path: /lib/modules
          type: Directory
        name: lib-modules
      - hostPath:
          path: /sys
          type: Directory
        name: tracing
      - hostPath:
          path: /proc
          type: Directory
        name: proc
      - configMap:
          name: kepler-cfm
        name: cfm
      - name: redfish
        secret:
          secretName: redfish-4kh9d7bc7m
---
apiVersion: monitoring.coreos.com/v1
kind: ServiceMonitor
metadata:
  labels:
    app.kubernetes.io/component: exporter
    app.kubernetes.io/name: kepler-exporter
    sustainable-computing.io/app: kepler
    release: kube-prometheus-stack
  name: kepler-exporter
  namespace: monitoring
spec:
  endpoints:
  - interval: 3s
    port: http
    relabelings:
    - action: replace
      regex: (.*)
      replacement: $1
      sourceLabels:
      - __meta_kubernetes_pod_node_name
      targetLabel: instance
    scheme: http
  jobLabel: app.kubernetes.io/name
  namespaceSelector:
    matchNames:
    - kepler
  selector:
    matchLabels:
      app.kubernetes.io/component: exporter
      app.kubernetes.io/name: kepler-exporter
```
</details>

rootfs, 2024-04-02:
I think EKS is still in cgroup v1. In this case, can you add `-enable-cgroup-id=false` in kepler? You can run `kubectl edit daemonset/kepler-exporter -n kepler` and change the args [here](https://github.com/sustainable-computing-io/kepler/blob/main/manifests/config/exporter/exporter.yaml#L72)

If this works on your end, a kepler-doc PR is welcome!

rootfs, 2024-04-06:
The dashboard is not well tested. There is another similar case here https://github.com/sustainable-computing-io/kepler/issues/1321

Can you check if the metrics e.g. `sum(rate(kepler_container_joules_total[1m]))` on the prometheus UI? It could be somewhere in the grafana and prometheus connection (e.g. token expiration)

qa-florian-wende, 2024-04-15:
It's definitely not just the dashboard, the graph in Prometheus shows exactly the same. What's interesting is that the metric seems to fail one node after the other:
[image]
[image]

So it appears unlikely that it's just a problem with Prometheus either because there is just one Prometheus instance in my setup.
But the Kepler Exporters never crashed, when I look at the /metrics endpoint on any of the Kepler-Exporter pods, I still see the value where the metrics froze, e.g.
```
# HELP kepler_container_package_joules_total Aggregated value in package value from trained_power_model
# TYPE kepler_container_package_joules_total counter
kepler_container_package_joules_total{container_id="system_processes",container_name="kernel_processes",container_namespace="kernel",mode="dynamic",pod_name="kernel_processes",source="trained_power_model"} 19443.159
```
When I look at the log of the exporters, I can watch the totals increase every few seconds:
<details><summary>Details</summary>
<p>
I0415 15:16:09.688498       1 metric_collector.go:116] Collector Update elapsed time: 16.01673ms
I0415 15:16:12.674285       1 libbpf_attacher.go:384] successfully get data with batch get and delete with 106 pids in 353.221µs
I0415 15:16:12.687665       1 stats.go:251] Unknown node feature: gpu_compute_util, adding 0 value
I0415 15:16:12.688033       1 stats.go:251] Unknown node feature: gpu_compute_util, adding 0 value
I0415 15:16:12.688142       1 stats.go:251] Unknown node feature: gpu_compute_util, adding 0 value
I0415 15:16:12.688244       1 stats.go:251] Unknown node feature: gpu_compute_util, adding 0 value
I0415 15:16:12.688327       1 stats.go:251] Unknown node feature: gpu_compute_util, adding 0 value
I0415 15:16:12.688433       1 stats.go:251] Unknown node feature: gpu_compute_util, adding 0 value
I0415 15:16:12.688513       1 stats.go:251] Unknown node feature: gpu_compute_util, adding 0 value
I0415 15:16:12.688648       1 stats.go:251] Unknown node feature: gpu_compute_util, adding 0 value
I0415 15:16:12.689356       1 metric_collector.go:337] energy from pod/container: name: karpenter-866b96f7c4-8p8p4/controller namespace: karpenter containerid:202ac0e53a5941a875d812e138ff1cb01bee8d9ed4490359ff99a49a74cf3cb9
 cgroupMetrics: map[block_devices_used:0 (0) cgroupfs_cpu_usage_us:0 (0) cgroupfs_ioread_bytes:0 (0) cgroupfs_iowrite_bytes:0 (0) cgroupfs_kernel_memory_usage_bytes:0 (0) cgroupfs_memory_usage_bytes:0 (0) cgroupfs_system_cpu_usage_us:0 (0) cgroupfs_tcp_memory_usage_bytes:0 (0) cgroupfs_user_cpu_usage_us:0 (0)]
    Dyn ePkg (mJ): 0 (0) (eCore: 0 (0) eDram: 0 (0) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0) platform (mJ): 0 (0) 
    Idle ePkg (mJ): 0 (0) (eCore: 0 (0) eDram: 0 (0) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0) platform (mJ): 0 (0) 
    ResUsage: map[block_devices_used:0 (0) bpf_block_irq:0 (0) bpf_cpu_time_ms:0 (0) bpf_net_rx_irq:0 (0) bpf_net_tx_irq:0 (0) bpf_page_cache_hit:0 (0) cache_miss:0 (0) cgroupfs_cpu_usage_us:0 (0) cgroupfs_ioread_bytes:0 (0) cgroupfs_iowrite_bytes:0 (0) cgroupfs_kernel_memory_usage_bytes:0 (0) cgroupfs_memory_usage_bytes:0 (0) cgroupfs_system_cpu_usage_us:0 (0) cgroupfs_tcp_memory_usage_bytes:0 (0) cgroupfs_user_cpu_usage_us:0 (0) cpu_cycles:0 (0) cpu_instructions:0 (0) cpu_ref_cycles:0 (0) task_clock_ms:0 (0)]

I0415 15:16:12.689628       1 metric_collector.go:337] energy from pod/container: name: kube-prometheus-stack-operator-85f86ff6f6-lddhx/kube-prometheus-stack namespace: monitoring containerid:82102415adffb1fa15f1971f09e3dac0abda82ef84dc34dc78c10176ce350079
 cgroupMetrics: map[block_devices_used:0 (0) cgroupfs_cpu_usage_us:8874 (87181948) cgroupfs_ioread_bytes:0 (0) cgroupfs_iowrite_bytes:0 (0) cgroupfs_kernel_memory_usage_bytes:0 (0) cgroupfs_memory_usage_bytes:4096 (24403968) cgroupfs_system_cpu_usage_us:6944 (35541810) cgroupfs_tcp_memory_usage_bytes:0 (0) cgroupfs_user_cpu_usage_us:1929 (51640137)]
    Dyn ePkg (mJ): 0 (702) (eCore: 0 (702) eDram: 0 (198) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0) platform (mJ): 0 (150) 
    Idle ePkg (mJ): 21281310 (21281310) (eCore: 21281310 (21281310) eDram: 14349432 (14349432) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0) platform (mJ): 65031 (65031) 
    ResUsage: map[block_devices_used:0 (0) bpf_block_irq:0 (0) bpf_cpu_time_ms:0 (165) bpf_net_rx_irq:0 (179) bpf_net_tx_irq:0 (0) bpf_page_cache_hit:0 (0) cache_miss:0 (0) cgroupfs_cpu_usage_us:0 (0) cgroupfs_ioread_bytes:0 (0) cgroupfs_iowrite_bytes:0 (0) cgroupfs_kernel_memory_usage_bytes:0 (0) cgroupfs_memory_usage_bytes:0 (0) cgroupfs_system_cpu_usage_us:0 (0) cgroupfs_tcp_memory_usage_bytes:0 (0) cgroupfs_user_cpu_usage_us:0 (0) cpu_cycles:0 (0) cpu_instructions:0 (0) cpu_ref_cycles:0 (0) task_clock_ms:0 (10214)]

I0415 15:16:12.689861       1 metric_collector.go:337] energy from pod/container: name: kepler-exporter-v6wk2/kepler-exporter namespace: kepler containerid:e9002a5a368e50995dc05ef0bcfdf02539532861e745a1092fda41686c4fc6a9
 cgroupMetrics: map[block_devices_used:0 (0) cgroupfs_cpu_usage_us:18761 (166135607) cgroupfs_ioread_bytes:0 (0) cgroupfs_iowrite_bytes:0 (0) cgroupfs_kernel_memory_usage_bytes:0 (0) cgroupfs_memory_usage_bytes:12288 (26558464) cgroupfs_system_cpu_usage_us:4041 (33699465) cgroupfs_tcp_memory_usage_bytes:0 (0) cgroupfs_user_cpu_usage_us:14719 (132436141)]
    Dyn ePkg (mJ): 0 (2430) (eCore: 0 (2430) eDram: 0 (699) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0) platform (mJ): 0 (558) 
[... cut ...]
    ResUsage: map[block_devices_used:0 (0) bpf_block_irq:0 (0) bpf_cpu_time_ms:0 (5012624) bpf_net_rx_irq:157 (1422283) bpf_net_tx_irq:0 (0) bpf_page_cache_hit:0 (0) cache_miss:0 (0) cgroupfs_cpu_usage_us:0 (0) cgroupfs_ioread_bytes:0 (0) cgroupfs_iowrite_bytes:0 (0) cgroupfs_kernel_memory_usage_bytes:0 (0) cgroupfs_memory_usage_bytes:0 (0) cgroupfs_system_cpu_usage_us:0 (0) cgroupfs_tcp_memory_usage_bytes:0 (0) cgroupfs_user_cpu_usage_us:0 (0) cpu_cycles:0 (0) cpu_instructions:0 (0) cpu_ref_cycles:0 (0) task_clock_ms:5756 (44579154)]

I0415 15:16:15.707477       1 metric_collector.go:116] Collector Update elapsed time: 34.273537ms
</p>
</details>

peiniliu, 2025-03-04:
Any news regarding issue #1310 and #1321 ?
The same happens with v0.7.8 on OpenStack VMs.  Kepler works fine within the first hour or two, then stops reporting.

- kepler-exporter no restart.
- /metrics, Prometheus, and Grafana have data 0 or counter stop. 
- kepler internal reports data

qa-florian-wende, 2025-03-04:
With Amazon Linux 2023 (which is now supported by eksctl) and new Kubernetes versions (1.31), we have not encountered the stability issues again.

closed 2026-05-10 by github-actions[bot]